open-power/skiboot #10: Latest skiboot does not IPL in op-build

geissonator, 2015-03-04:
I merged this today.

https://github.com/open-power/op-build/commit/a050b4c7b8446cfd6f1f480138b41f6cdd4f6ae0

We get this failure on habanero and palmetto when booting.  Guess I'll revert it op-build.

[893214895,5] BT: Interface intialized, IO 0x00e4
[1354480573,4] SLW: HB-provided idle states property found
[1354583151,5] NVRAM: Size is 576 KB
[1594800654,3] NVRAM: Partition at offset 0xffff0 extends beyond end of nvram !
[1594947972,3] NVRAM: Re-initializing
[1596889455,5] Chip 0 Found PBCQ2 at /xscom@3fc0000000000/pbcq@2012800
[1597059022,5] Chip 0 Found PBCQ1 at /xscom@3fc0000000000/pbcq@2012400
[1598002147,5] Chip 0 Found PBCQ0 at /xscom@3fc0000000000/pbcq@2012000
[1599125400,3] PHB0: Base location code not found !
LOCK ERROR: Invalid recursive lock @0x30101a48 (state: 0x0000006000000001)
[1660968927,0] Aborting!
CPU 0060 Backtrace:
 S: 0000000031b83870 R: 000000003001351c   mambo_write+0xe94
 S: 0000000031b83900 R: 00000000300172a0   mambo_write+0x4c18
 S: 0000000031b83970 R: 0000000030015ea8   mambo_write+0x3820
 S: 0000000031b839f0 R: 0000000030015f64   mambo_write+0x38dc
 S: 0000000031b83a70 R: 0000000030025228   mambo_write+0x12ba0
 S: 0000000031b83b50 R: 0000000030021da0   mambo_write+0xf718
 S: 0000000031b83bc0 R: 000000003003b898   mambo_write+0x29210
 S: 0000000031b83c90 R: 00000000300413c8   mambo_write+0x2ed40
 S: 0000000031b83e30 R: 0000000030014214   mambo_write+0x1b8c
 S: 0000000031b83f00 R: 0000000030002524   boot_entry+0x18c

ozbenh, 2015-03-04:
Those symbols look completely wrong, can you send me the skiboot.elf generated during the build ?

geissonator, 2015-03-04:
Ack, I verified the revert worked in the same op-build dir so don't have it anymore.  Should be able to recreate with a " git clone --recursive git@github.com:jk-ozlabs/op-build.git".  Had 4 people hit this independently so something is up with it.

stewart-ibm, 2015-04-02:
I'll assume you're no longer having this issue, as we're not seeing it on our machines... please reopen if still seeing with latest op-build and skiboot.

closed 2015-04-02 by stewart-ibm