open-power/skiboot #12: max args to an opal call

andreiw, 2015-05-12:
opal_entry (and opal-spec.txt) claim 9 arguments may be passed in r3..r11, yet the OpenPower ABI mentions (2.2.1.1 Register Roles) that r11 is not used for parameter passing, and again in 2.2.3 (Parameter Passing in Registers) that only 8 registers are available, with the rest spilling to stack.

A simple test confirms:
    2001021c:   09 00 20 39     li      r9,9
    20010220:   60 00 21 f9     std     r9,96(r1)
    20010224:   0a 00 20 39     li      r9,10
    20010228:   68 00 21 f9     std     r9,104(r1)
    2001022c:   0b 00 20 39     li      r9,11
    20010230:   70 00 21 f9     std     r9,112(r1)
    20010234:   0c 00 20 39     li      r9,12
    20010238:   78 00 21 f9     std     r9,120(r1)
    2001023c:   01 00 60 38     li      r3,1
    20010240:   02 00 80 38     li      r4,2
    20010244:   03 00 a0 38     li      r5,3
    20010248:   04 00 c0 38     li      r6,4
    2001024c:   05 00 e0 38     li      r7,5
    20010250:   06 00 00 39     li      r8,6
    20010254:   07 00 20 39     li      r9,7
    20010258:   08 00 40 39     li      r10,8
    2001025c:   9d fe ff 4b     bl      200100f8 <foo>

If OPAL really does support 9 arguments, the opal_entry logic should be reflected (fetching arg 9 from client stack, yuck)... To me it seems mostly a typo (i.e. OPAL really supports only 8 args, not counting the token in r0), but then r11 manipulation for OPAL_TRACE_ENTRY should be avoided.

ozbenh, 2015-05-12:
> opal_entry (and opal-spec.txt) claim 9 arguments may be passed in
> r3..r11, yet the OpenPower ABI mentions (2.2.1.1 Register Roles) that
> r11 is not used for parameter passing, and again in 2.2.3 (Parameter
> Passing in Registers) that only 8 registers are available, with the
> rest spilling to stack.

It's a stupid thinko that I'm to blame for, it's 8 you are correct.

We should probably enforce it since we know the number of args in
the register macro/functions... we might even be able to do a build
fail.

stewart-ibm, 2015-05-12:
Cool, just sent a patch out to the list - It'd be great if you one/both of you could ack it. It just fixes docs and the OPAL_TRACE_ENTRY part.

andreiw, 2015-05-17:
This is resolved now.

closed 2015-05-17 by andreiw