open-power/skiboot #13: OPAL spec, constrains on OPAL calls

andreiw, 2015-05-12:
opal-spec.txt should probably describe the expected machine state across an OPAL call a bit better, in that (from reading Linux kernel code and skiboot):
- r1 is preserved, r12 is scratch, r13-r31 preserved
- 64-bit HV real mode
- big-endian (this is already mentioned)
- external interrupts disabled

ozbenh, 2015-05-12:
> opal-spec.txt should probably describe the expected machine state
> across an OPAL call a bit better, in that (from reading Linux kernel
> code and skiboot):
>
>       * r1 is preserved, r12 is scratch, [...]

Good point.

Ben.

stewart-ibm, 2015-05-25:
Thanks for suggestion, I've made an addition to opal-spec.txt mentioning these things. It's not great docs, but it's there for when we need to turn it into something better.

closed 2015-05-25 by stewart-ibm