open-power/skiboot #16: Fix OPAL_REINIT_CPUS with HILE_LE on Mambo systemsim

andreiw, 2015-05-17:
```PVR_VERS_MAJ(mfspr(SPR_PVR))``` for Mambo systemsim with processor/type P8
has major version 1, yet ```core/cpu.c``` sets ```hile_supported``` is set for version 2.

However, HID0.HILE does work correctly in P8 systemsim. I've verified that setting the bit causes exception code with MSR.HV=1 to be treated as little-endian. Thus, the following patch introduces another chip quirk (```QUIRK_HAVE_HILE```).

Before:
```
[19244603,3] OPAL: Trying a CPU re-init with flags: 0x2
[19250136,3] SLW: Not found on chip 0
[19257911,3] ELOG: Error getting buffer to log error
```
(and LE exception handlers don't work)

After:
```
[19244603,3] OPAL: Trying a CPU re-init with flags: 0x2
```
(and LE exception handlers *do* work)

Patch:
```
From 3996b2f40efadfd53d9bdf9a77638ed36aaed40b Mon Sep 17 00:00:00 2001
From: Andrei Warkentin <andrey.warkentin@gmail.com>
Date: Sun, 17 May 2015 01:49:53 -0400
Subject: [PATCH] OPAL: enable opal_reinit_cpus(OPAL_REINIT_CPUS_HILE_LE) to
 work on Mambo systemsim

PVR_VERS_MAJ(mfspr(SPR_PVR)) for Mambo systemsim with processor/type P8
has major version 1, however HID0.HILE does seem to work correctly,
so add a chip quirk for Mambo.

---
 core/chip.c    | 2 +-
 core/cpu.c     | 2 +-
 include/chip.h | 1 +
 3 files changed, 3 insertions(+), 2 deletions(-)

diff --git a/core/chip.c b/core/chip.c
index 2ba7b6e..09ea825 100644
--- a/core/chip.c
+++ b/core/chip.c
@@ -71,7 +71,7 @@ void init_chips(void)
 	if (dt_find_by_path(dt_root, "/mambo")) {
 		proc_chip_quirks |= QUIRK_NO_CHIPTOD | QUIRK_MAMBO_CALLOUTS
 			| QUIRK_NO_F000F | QUIRK_NO_PBA | QUIRK_NO_OCC_IRQ
-			| QUIRK_DISABLE_NAP;
+			| QUIRK_DISABLE_NAP | QUIRK_HAVE_HILE;
 		prlog(PR_NOTICE, "CHIP: Detected Mambo simulator\n");
 	}
 	if (dt_node_is_compatible(dt_root, "qemu,powernv")) {
diff --git a/core/cpu.c b/core/cpu.c
index 4775883..670d7e9 100644
--- a/core/cpu.c
+++ b/core/cpu.c
@@ -742,7 +742,7 @@ static int64_t opal_reinit_cpus(uint64_t flags)
 	 * use the HID bit. We use the PVR (we could use the EC level in
 	 * the chip but the PVR is more readily available).
 	 */
-	if (hile_supported &&
+	if ((hile_supported || chip_quirk(QUIRK_HAVE_HILE)) &&
 	    (flags & (OPAL_REINIT_CPUS_HILE_BE | OPAL_REINIT_CPUS_HILE_LE))) {
 		bool hile = !!(flags & OPAL_REINIT_CPUS_HILE_LE);
 
diff --git a/include/chip.h b/include/chip.h
index 0547902..2288d29 100644
--- a/include/chip.h
+++ b/include/chip.h
@@ -98,6 +98,7 @@ enum proc_chip_quirks {
 	QUIRK_NO_PBA		= 0x00000008,
 	QUIRK_NO_OCC_IRQ       	= 0x00000010,
 	QUIRK_DISABLE_NAP	= 0x00000020,
+	QUIRK_HAVE_HILE		= 0x00000040,
 } proc_chip_quirks;
 
 static inline bool chip_quirk(unsigned int q)
-- 
1.9.1
```

ozbenh, 2015-05-17:
On Sun, 2015-05-17 at 00:41 -0700, Andrei Warkentin wrote:
> PVR_VERS_MAJ(mfspr(SPR_PVR)) for Mambo systemsim with processor/type
> P8
> has major version 1, yet core/cpu.c sets hile_supported is set for
> version 2.
>
> However, HID0.HILE does work correctly in P8 [...]

I'd rather we fix Mambo :-) I'll ask internally.

Ben.

> Before:
> 
> [19244603,3] OPAL: Trying a CPU re-init with flags: 0x2
> [19250136,3] SLW: Not found on chip 0
> [19257911,3] ELOG: Error getting buffer to log error
> 
> (and LE exception handlers don't work)
> 
> After:
> 
> [19244603,3] OPAL: Trying a CPU re-init with flags: 0x2
> 
> (and LE exception handlers do work)
> 
> Patch:
> 
> From 3996b2f40efadfd53d9bdf9a77638ed36aaed40b Mon Sep 17 00:00:00 2001
> From: Andrei Warkentin <andrey.warkentin@gmail.com>
> Date: Sun, 17 May 2015 01:49:53 -0400
> Subject: [PATCH] OPAL: enable opal_reinit_cpus(OPAL_REINIT_CPUS_HILE_LE) to
>  work on Mambo systemsim
> 
> PVR_VERS_MAJ(mfspr(SPR_PVR)) for Mambo systemsim with processor/type P8
> has major version 1, however HID0.HILE does seem to work correctly,
> so add a chip quirk for Mambo.
> 
> Signed-off-by: Andrei Warkentin <andrey.warkentin@gmail.com>
> ---
>  core/chip.c    | 2 +-
>  core/cpu.c     | 2 +-
>  include/chip.h | 1 +
>  3 files changed, 3 insertions(+), 2 deletions(-)
> 
> diff --git a/core/chip.c b/core/chip.c
> index 2ba7b6e..09ea825 100644
> --- a/core/chip.c
> +++ b/core/chip.c
> @@ -71,7 +71,7 @@ void init_chips(void)
>     if (dt_find_by_path(dt_root, "/mambo")) {
>         proc_chip_quirks |= QUIRK_NO_CHIPTOD | QUIRK_MAMBO_CALLOUTS
>             | QUIRK_NO_F000F | QUIRK_NO_PBA | QUIRK_NO_OCC_IRQ
> -           | QUIRK_DISABLE_NAP;
> +           | QUIRK_DISABLE_NAP | QUIRK_HAVE_HILE;
>         prlog(PR_NOTICE, "CHIP: Detected Mambo simulator\n");
>     }
>     if (dt_node_is_compatible(dt_root, "qemu,powernv")) {
> diff --git a/core/cpu.c b/core/cpu.c
> index 4775883..670d7e9 100644
> --- a/core/cpu.c
> +++ b/core/cpu.c
> @@ -742,7 +742,7 @@ static int64_t opal_reinit_cpus(uint64_t flags)
>      * use the HID bit. We use the PVR (we could use the EC level in
>      * the chip but the PVR is more readily available).
>      */
> -   if (hile_supported &&
> +   if ((hile_supported || chip_quirk(QUIRK_HAVE_HILE)) &&
>         (flags & (OPAL_REINIT_CPUS_HILE_BE | OPAL_REINIT_CPUS_HILE_LE))) {
>         bool hile = !!(flags & OPAL_REINIT_CPUS_HILE_LE);
> 
> diff --git a/include/chip.h b/include/chip.h
> index 0547902..2288d29 100644
> --- a/include/chip.h
> +++ b/include/chip.h
> @@ -98,6 +98,7 @@ enum proc_chip_quirks {
>     QUIRK_NO_PBA        = 0x00000008,
>     QUIRK_NO_OCC_IRQ        = 0x00000010,
>     QUIRK_DISABLE_NAP   = 0x00000020,
> +   QUIRK_HAVE_HILE     = 0x00000040,
>  } proc_chip_quirks;
> 
>  static inline bool chip_quirk(unsigned int q)

stewart-ibm, 2015-09-09:
Should be fixed as of dd3c3ec045ff47adb5531a49bda50e4c86cd38fa which changes the PVR we give Mambo to 0x4b0201 (in external/mambo/skiboot.tcl) - so hopefully this should be all fixed now

closed 2015-09-09 by stewart-ibm