open-power/skiboot #17: systemsim perf question

andreiw, 2015-05-21:
Sorry if this is a bit OT, but in the systemsim config file for skiboot, the CPU frequency is set to 512, and the TB is also  512.
https://github.com/open-power/skiboot/blob/master/external/mambo/skiboot.tcl#L70

Yet when I run skiboot and a payload (I'm playing around with LE 64-bit code), the TB seems to be 1000-times slower. The TB just doesn't increment fast enough.

E.g. The 1000 should not be necessary, perhaps it's a factor of my environment (a not so new laptop). Are you guys aware of systemsim configuration options to make the timebase at least be approximately correct? Not atomic clock quality, just not 1000 or 10 or whatever times slower. Sorry again if this is OT.
```
                uint64_t now = mftb();
                while ((mftb() - now) * 1000 < (secs_to_tb(3))) {
                        printk("Now = %u.%u s\n",
                               ((mftb() - now) * 1000) / tb_hz,
                               ((mftb() - now) * 1000) % tb_hz);
                }
```

ozbenh, 2015-05-22:
> Sorry if this is a bit OT, but in the systemsim config file for
> skiboot, the CPU frequency is set to 512, and the TB is also 512. [...]

The way mambo (aka systemsim) works, there is no expectation that the
simulated timebase has any relationship whatsoever to wall clock outside
of sim. All you can do really is play with the ratio emulated CPU
cycles / TB cycles.

With qemu you get a TB that is adjusted to wall clock, however my qemu
"native" model isn't public yet (working on it...)

Cheers,
Ben.

andreiw, 2015-05-22:
Thanks, that makes sense.

closed 2015-05-22 by andreiw

stewart-ibm, 2015-09-09:
btw - @ozbenh has pushed his qemu up to https://github.com/ozbenh/qemu and you can boot skiboot and linux in it - may be useful to you

andreiw, 2015-09-14:
Whoa! Very cool. Thanks for the heads up!