open-power/skiboot #19: skiboot fails to build on RHEL6.

williamspatrick, 2015-08-04:
>     [ HOSTCC ] extract-gcov.c
> cc1: error: unrecognized command line option "-Wjump-misses-init"
> cc1: error: unrecognized command line option
> "-Wsuggest-attribute=const"
> cc1: error: unrecognized command line option
> "-Wsuggest-attribute=noreturn"
> cc1: error: unrecognized command line option "-Wstack-usage=1024"

We get the following failure on RHEL6 when doing an op-build because of using compiler warning options that do not exist.  The compiler options are fine for TARGET-CC but not HOST-CC.  In op-build(buildroot), the original /usr/bin/gcc is used for HOST-CC while the cross-compiler is used for TARGET-CC.

williamspatrick, 2015-08-04:
@jk-ozlabs , @ozbenh , @stewart-ibm - I attempted to make a fix with open-power/op-build@2b4464b05008919aa337bd13a3e4b153226bb222 .  We need this fixed in mainline skiboot though and when the fix shows up in op-build we can remove the patch file.

stewart-ibm, 2015-09-09:
Should be fixed as of 8ec792b604481fb02ffc5c41bd02e74cffda9b1d and skiboot-5.1.1

closed 2015-09-09 by stewart-ibm