open-power/skiboot #21: FSPCT: No centaur memory detected after cold reset

saqibkh, 2015-12-14:
No centaur memory was detected after doing a cold reset of the BMC:

System Info:
BMC ip address= 9.41.164.80  name=habcap10.aus.stglabs.ibm.com
OS ip address=9.41.164.12 name=habcap10p1.aus.stglabs.ibm.com
---------------------------------------------------------------------------------------------------
Following is a list of commands that were executed and their outcome:

HOST OS: $ sudo /home/tyan/skiboot/external/xscom-utils/getscom -l
[sudo] password for tyan:
Chip ID  | Rev   | Chip type
---------|-------|--------
80000005 | DD2.0 | Centaur memory buffer
80000004 | DD2.0 | Centaur memory buffer
80000001 | DD2.0 | Centaur memory buffer
80000000 | DD2.0 | Centaur memory buffer
00000000 | DD2.0 | P8 (Venice) processor

# ipmitool -H habcap10 -I lanplus -U ${Userid} -P ${Password} mc cold reset

HOST OS: $ sudo /home/tyan/skiboot/external/xscom-utils/getscom -l
[sudo] password for tyan:
Chip ID  | Rev   | Chip type
---------|-------|--------
00000000 | DD2.0 | P8 (Venice) processor   <===========  No Centaur Memory

Also attached is an output obtained from the serial port, which shows that the BMC was unable to write/detect these memory.
[image]

ozbenh, 2015-12-14:
> No centaur memory was detected after doing a cold reset of the BMC:
> System Info:
> BMC ip address= 94116480 name=habcap10ausstglabsibmcom
> OS ip address=94116412 name=habcap10p1ausstglabsibmcom [...]

This isn't a skiboot problem. HostBoot didn't forward the information
about the Centaurs to us.

gery-toulouse, 2018-09-04:
Should be closed as not updated for nearly 18 months. Please reopen if needed.

closed 2018-09-04 by sammj