open-power/skiboot #22: Support N25Qx256Ax NOR flash

anoo1, 2015-12-18:
Commit 71c2d6b72be7213914e2708c884b45fa74f525e6 added support for N25Q256Ax which is what IBM is using, but Foxconn is using N25Qx256Ax for their Barreleye systems. So need to add N25Qx256Ax to pflash.

ozbenh, 2015-12-20:
On Fri, 2015-12-18 at 13:22 -0800, anoo1 wrote:
> Commit 71c2d6b added support for N25Q256Ax which is what IBM is
> using, but Foxconn is using N25Qx256Ax for their Barreleye systems.
> So need to add N25Qx256Ax to pflash.

Is there any difference ?

Cheers,
Ben.

anoo1, 2015-12-23:
They don't appear to be different. Just need the name added to the pflash library because currently pflash returns "unsupported chip". Norm has verified that adding the new name flashes the chip successfully by adding this chip name to the local pflash copy of openbmc.

ozbenh, 2015-12-24:
> They don't appear to be different. Just need the name added to the
> pflash library because currently pflash returns "unsupported chip".
> Norm has verified that adding the new name flashes the chip
> successfully by adding this chip name to the local pflash copy of
> openbmc.

Wait, you mean that it has a different flash ID than the non-x variant
? Ok, in that case, we do need to add it. It's weird though, usually
they keep the IDs the same for minor differences (there are other
commands to query more details).

Cheers,
Ben.

williamspatrick, 2016-01-09:
@stewart-ibm - Please merge if appropriate.

williamspatrick, 2016-01-09:
Err... I guess this is an issue and not a pull request.  @stewart-ibm can you code this up quick and put it in?  Or has it already been taken care of?

anoo1, 2016-01-11:
A barreleye that just arrived has a "Winbond W25Q256BV" chip, so this chip needs to be added as well.

{ 0xef4019, 0x02000000, FL_ERASE_ALL | FL_CAN_4B | FL_ERASE_64K|FL_ERASE_BULK,    "Winbond W25Q256BV"},

stewart-ibm, 2016-01-12:
@anoo1 since you have access to such a system and can test, can you please post a patch to skiboot@lists.ozlabs.org ? Or, I can post one and you can test it :)

@williamspatrick - for N25Qx256Ax, do we know the details for them? Again, I don't have direct access to such a system, so will need someone to at least test.

anoo1, 2016-01-13:
Patch sent with the changes for both chips, N25Qx256Ax and W25Q256BV.

anoo1, 2016-02-23:
Per Stewart, this has been merged to master as of 539720ca9b1543a35411511a4154d3c9e5e4dc84.

closed 2016-02-23 by anoo1