open-power/skiboot #23: Failure to inject system checkstop error on all cores

saqibkh, 2016-01-10:
System returned an error when trying to inject a system checkstop error on some of the cores.
Note: Not all cores had this issue

System info:
bmc --> habcap10.aus.stglabs.ibm.com
lpar ---> habcap10p1.aus.stglabs.ibm.com

from BMC:
cat /proc/ractrends/Helper/FwInfo
FW_VERSION=2.16.85627.65366
FW_DATE=Dec 7 2015
FW_BUILDTIME=13:40:23 CDT
FW_DESC=8348 P2 SRC RR9 12072015
FW_PRODUCTID=1
FW_RELEASEID=RR9
FW_CODEBASEVERSION=2.X

khansa@habcap11p1:~$ ipmitool -H habcap10 -I lanplus -U ADMIN -P admin fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : IBM-habanero-ibm-OP8_v1.7_1.17
 Product Extra         :        hostboot-bc98d0b-5463ef4
 Product Extra         :        occ-0362706-9e016d2
 Product Extra         :        skiboot-5.1.9-ada75ff
 Product Extra         :        hostboot-binaries-43d5a59
 Product Extra         :        habanero-xml-a71550e-85cdcb7
 Product Extra         :        capp-ucode-105cb8f

------------------------------------------------------------------------------------------------------------------
Following is a list of step that were executed on the partition and the output received.
tyan@habcap10p1:~$ cat /sys/firmware/opal/msglog | grep -i chip | grep -i core
[7261039858,7] OCC: Chip 0 Core 1 PPMSR d1d1ca0008000000
[7261557077,7] OCC: Chip 0 Core 4 PPMSR dadaca0008000000
[7262073389,7] OCC: Chip 0 Core 5 PPMSR e3e3ca0008000000
[7262589628,7] OCC: Chip 0 Core 6 PPMSR ebebca0008000000
[7263107075,7] OCC: Chip 0 Core a PPMSR f0f0ca0008000000
[7263622298,7] OCC: Chip 0 Core c PPMSR f0f0ca0008000000
[7264137165,7] OCC: Chip 0 Core d PPMSR f0f0ca0008000000
[7264651994,7] OCC: Chip 0 Core e PPMSR f0f0ca0008000000

tyan@habcap10p1:~$ sudo /home/tyan/skiboot/external/xscom-utils/getscom -l
Chip ID  | Rev   | Chip type
---------|-------|--------
80000005 | DD2.0 | Centaur memory buffer
80000004 | DD2.0 | Centaur memory buffer
80000001 | DD2.0 | Centaur memory buffer
80000000 | DD2.0 | Centaur memory buffer
00000000 | DD2.0 | P8 (Venice) processor

tyan@habcap10p1:~$ sudo /home/tyan/skiboot/external/xscom-utils//putscom -c 0x00000000 0x1e012800 0x0100000000000000     <----------  Checkstop detected at core # e
Error -5 writing XSCOM
tyan@habcap10p1:~$ sudo /home/tyan/skiboot/external/xscom-utils//putscom -c 0x00000000 0x1d012800 0x0100000000000000      <----------  Checkstop not detected at core # d
Error -5 writing XSCOM
tyan@habcap10p1:~$ sudo /home/tyan/skiboot/external/xscom-utils//putscom -c 0x00000000 0x1c012800 0x0100000000000000       <----------  Checkstop detected at core # c

saqibkh, 2016-01-10:
[msglog.txt](https://github.com/open-power/skiboot/files/85139/msglog.txt)

pridhiviraj, 2016-02-04:
Can you disable CPU states 1 and 2 , then try injecting check stop errors.
for i in /sys/devices/system/cpu/cpu*/cpuidle/state1/disable; do echo 1 > $i; done
for i in /sys/devices/system/cpu/cpu*/cpuidle/state2/disable; do echo 1 > $i; done

saqibkh, 2016-02-04:
No such file exists.

tyan@habcap10p1:/sys/devices/system/cpu$ ls /sys/devices/system/cpu
cpu0   cpu18  cpu27  cpu36  cpu45  cpu54  cpu63         possible
cpu1   cpu19  cpu28  cpu37  cpu46  cpu55  cpu7          power
cpu10  cpu2   cpu29  cpu38  cpu47  cpu56  cpu8          present
cpu11  cpu20  cpu3   cpu39  cpu48  cpu57  cpu9          probe
cpu12  cpu21  cpu30  cpu4   cpu49  cpu58  cpufreq       release
cpu13  cpu22  cpu31  cpu40  cpu5   cpu59  cpuidle       subcores_per_core
cpu14  cpu23  cpu32  cpu41  cpu50  cpu6   dscr_default  uevent
cpu15  cpu24  cpu33  cpu42  cpu51  cpu60  kernel_max
cpu16  cpu25  cpu34  cpu43  cpu52  cpu61  offline
cpu17  cpu26  cpu35  cpu44  cpu53  cpu62  online

pridhiviraj, 2016-02-04:
[root@ite037 ~]# ls /sys/devices/system/cpu/cpu0/cpuidle/state
state0/ state1/ state2/ 

files will be there, in the above comment, the shell script loop through the all cpu's, just execute the script for state1 and state2 and disable the states

saqibkh, 2016-02-04:
for loop logic isn't correct

root@habcap10p1:/home/tyan# for i in /sys/devices/system/cpu/cpu/cpuidle/state1/disable; do echo 1 > $i; done
bash: /sys/devices/system/cpu/cpu/cpuidle/state1/disable: No such file or directory

pridhiviraj, 2016-02-04:
for i in /sys/devices/system/cpu/cpu*/cpuidle/state1/disable; do echo 1 > $i; done
for i in /sys/devices/system/cpu/cpu*/cpuidle/state2/disable; do echo 1 > $i; done

sorry, here * is missed. try these now.

saqibkh, 2016-02-04:
these are the same commands as above.

root@habcap10p1:/home/tyan# for i in /sys/devices/system/cpu/cpu/cpuidle/state1/disable; do echo 1 > $i; done
bash: /sys/devices/system/cpu/cpu/cpuidle/state1/disable: No such file or directory

pridhiviraj, 2016-02-04:
for i in /sys/devices/system/cpu/cpu*/cpuidle/state1/disable; do echo 1 > $i; done
for i in /sys/devices/system/cpu/cpu*/cpuidle/state1/disable; do echo 1 > $i; done
system/cpu/cpu*/cpuidle. in comment * is not getting printed

saqibkh, 2016-02-04:
Yes I'm able to inject these errors.

Can you tell me what these states represent and why we are disabling them.
Do I need to disable them every time before injecting these error?
Should I re-enable them afterwards?

pridhiviraj, 2016-02-04:
Yes, we need to disable before injection and re-enable afterwards. These states are CPU sleep states.
In order to inject core check stop error, cpu should be running, that is state 0. So we need to disble remaining sleep states. these states will save power if we enable them. 
https://www.ibm.com/developerworks/community/wikis/home?lang=en#!/wiki/Linux%20Kernel%20Power%20Management%20on%20IBM%20POWER%20Servers
Here you can see about cpu power management, and to know more about these cpu sleep states.

gery-toulouse, 2018-09-04:
Should be closed as not updated for nearly 18 months. Please reopen if needed.

closed 2018-09-04 by sammj