open-power/skiboot #24: Failed to get chip id using getscom tool

saqibkh, 2016-01-10:
getscom tool failed to identify the Chip ID on garrrison system, with the latest code

System Info:
bmc --> g68.aus.stglabs.ibm.com
lpar --> g68L.aus.stglabs.ibm.com
Build Level: 20151211n

from BMC:
cat /proc/ractrends/Helper/FwInfo 
FW_VERSION=2.13.85625
FW_DATE=Dec 10 2015
FW_BUILDTIME=15:32:56 CDT
FW_DESC=8335-GTB SRC BUILD RR9 12102015
FW_PRODUCTID=1
FW_RELEASEID=RR9
FW_CODEBASEVERSION=2.X

tyan@g68L:~$ sudo /home/tyan/skiboot/external/xscom-utils/getscom -l
Chip ID  | Rev   | Chip type
---------|-------|--------
80000015 | DD2.0 | Centaur memory buffer
80000014 | DD2.0 | Centaur memory buffer
80000005 | DD2.0 | Centaur memory buffer
80000004 | DD2.0 | Centaur memory buffer
00000001 | DD1.0 | Unknown ID 0xd3                         <--- Failed to identify chip type
00000000 | DD1.0 | Unknown ID 0xd3                         <--- Failed to identify chip type

saqibkh, 2016-01-10:
[msglog.txt](https://github.com/open-power/skiboot/files/85179/msglog.txt)

gery-toulouse, 2018-09-04:
Should be closed as not updated for nearly 18 months. Please reopen if needed.

closed 2018-09-04 by sammj