open-power/skiboot #30: ipmitool fru PNOR version info missing capp-ucode- version.

pridhiviraj, 2016-05-07:
ipmi fru info for pnor version, missing capp-ucode version.

How to reproduce:
Build a habanero_update.pnor image using op-build master-next branch, Flash this image on OpenPower habanero machine.
Read ipmitool fru print 43 for PNOR version info.
ipmitool fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-6ec4e33
 Product Extra         : 	buildroot-d78ccc1
 Product Extra         : 	skiboot-5.2.1
 Product Extra         : 	hostboot-e189edd-9824b9f
 Product Extra         : 	linux-4.4.8-openpower1-bd9161a
 Product Extra         : 	petitboot-v1.0.1-8f13d5f
 Product Extra         : 	habanero-xml-5a2a541-cb4e251
 Product Extra         : 	occ-19a3ebc
 Product Extra         : 	hostboot-binaries-7f593a3
 Product Extra         : 	c

Try reading PNOR VERSION partition using ipmitool.
ipmitool raw 0x3a 0x0b 0x56 0x45 0x52 0x53 0x49 0x4f 0x4e 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 | xxd -r -p
open-power-habanero-6ec4e33
	buildroot-d78ccc1
	skiboot-5.2.1
	hostboot-e189edd-9824b9f
	linux-4.4.8-openpower1-bd9161a
	petitboot-v1.0.1-8f13d5f
	habanero-xml-5a2a541-cb4e251
	occ-19a3ebc
	hostboot-binaries-7f593a3
	capp-ucode-105cb8f

We can observe from above two outputs, ipmitool fru print 43 is missing app-ucode-105cb8f which is a version of capp-ucode.

contact: ppaidipe@linux.vnet.ibm.com

williamspatrick, 2016-05-07:
Now that the Linux and buildroot version was added, we do not have the space in the ipmi sensor to hold all of the version information, so it gets truncated.  We made a decision on the order of the packages so that higher priority packages appeared first and capp-ucode was considered lower priority.  We need to either revert to the old version behavior, which did not have Linux, buildroot, and Petitboot, or send this to the BMC to see if the sensor size can be increased.  This is not a skiboot defect. 

Patrick

pridhiviraj, 2016-05-09:
Thanks patrick, closing the issue here, will raise against op-build

closed 2016-05-09 by pridhiviraj