open-power/skiboot #31: Switching from golden side pnor boot to primary side pnor boot is not happening for habanero machine.

pridhiviraj, 2016-05-07:
Problem statement:
Install a habanero system with OpenPower Firmware, and do a golden side pnor boot. After further reboots system still booting always from golden side pnor firmware, and it is not booting from primary side pnor firmware.

How to Reproduce:
1.Install a Habanero system with latest OpenPower Firmware.

ipmitool sensor list | grep -i "boot count"
Boot Count       | 0x0        | discrete   | 0x0280| na        | na        | na        | na        | na        | na   
initially boot count =0x02

2. Do a Power ON followed by immediate power off(To make sure boot fail early)
check for boot count sensor value, it should be decremented by 1
ipmitool sensor list | grep -i "boot count"
Boot Count       | 0x0        | discrete   | 0x0180| na        | na        | na        | na        | na        | na

3. Do a Power ON followed by immediate power off for 2nd time(To make sure boot fail early)
check for boot count sensor value, it should be decremented by 1
ipmitool sensor list | grep -i "boot count"
Boot Count       | 0x0        | discrete   | 0x0080| na        | na        | na        | na        | na        | na        

3. Do a Power ON Operation to boot the system from golden side PNOR(Because boot count =0x00)

ipmitool sensor list | grep -i golden
BIOS Golden Side | 0x0        | discrete   | 0x0180| na        | na        | na        | na        | na        | na        
BMC Golden Side  | 0x0        | discrete   | 0x0080| na        | na        | na        | na        | na        | na 

cd /sys/firmware/devicetree/base/xscom@3fc0000000000/isa@b0020/pnor@fc000000
lsprop *-tocs
active-image-tocs
		 02000000 03ff8000
alternate-image-tocs
		 00000000 01ff8000
golden-image-tocs
		 02000000 03ff8000
Here we can observe at this stage system booted from golden side

4. Reboot the sytem again.
reboot
[  OK  ] Started Show Plymouth Reboot Screen.
[  OK  ] Stopped Virtualization daemon.
         Stopping D-Bus System Message Bus...

After completion of reboot check again system boot side still it is in golden side of pnor.
cd /sys/firmware/devicetree/base/xscom@3fc0000000000/isa@b0020/pnor@fc000000
lsprop *-tocs
active-image-tocs
		 02000000 03ff8000
alternate-image-tocs
		 00000000 01ff8000
golden-image-tocs
		 02000000 03ff8000

ipmitool sensor list | grep -i golden
BIOS Golden Side | 0x0        | discrete   | 0x0180| na        | na        | na        | na        | na        | na        
BMC Golden Side  | 0x0        | discrete   | 0x0080| na        | na        | na        | na        | na        | na        

Further subsequent reboots also system is not booting from primary side of pnor, and it is always booting from golden side of pnor.

How to switch from golden side pnor to primary side pnor manually:

1. On BMC Busy box execute below command.
/etc/init.d/boot_into_primary

2. Do a ipmi hard power reset to boot the system into primary.
ipmitool power reset.

Firmware levels:
When system is in primary side:
open-power-habanero-6ec4e33
	buildroot-d78ccc1
	skiboot-5.2.1
	hostboot-e189edd-9824b9f
	linux-4.4.8-openpower1-bd9161a
	petitboot-v1.0.1-8f13d5f
	habanero-xml-5a2a541-cb4e251
	occ-19a3ebc
	hostboot-binaries-7f593a3
	capp-ucode-105cb8f

When system is in golden side:
 Product Name          : OpenPOWER Firmware
 Product Version       : TN71-BP012-810.1611.20160406a
 Product Extra         : 	hostboot-bc98d0b-3f66050
 Product Extra         : 	occ-0362706-805fe5a
 Product Extra         : 	skiboot-5.1.13
 Product Extra         : 	hostboot-binaries-43d5a59
 Product Extra         : 	habanero-xml-5a2a541-228a675
 Product Extra         : 	capp-ucode-105cb8f

williamspatrick, 2016-05-07:
How is this a skiboot issue and not a BMC issue?

Patrick

pridhiviraj, 2016-05-09:
Thanks Patrick, will raise issues against op-build.

closed 2016-05-09 by pridhiviraj