open-power/skiboot #32: xscom-utils and gard tool build failure on OpenPower Machine.

pridhiviraj, 2016-05-13:
xscom-utils and gard tool are both failed to build in this environment
habanero Hardware, PowerKVM 3.1.0 and below gcc level.
rpm -qf /bin/gcc
gcc-4.8.3-9.el7_1.ibm.3.ppc64le
cd skiboot/external/xscom-utils/
[root@localhost xscom-utils]# make 
    CC  xscom.o
    CC  version.o
  LINK  getscom
    CC  putscom.o
  LINK  putscom
    CC  getsram.o
    CC  sram.o
  LINK  getsram
[root@localhost xscom-utils]# ls
Makefile   getsram.c  sram.c  version.c  xscom.h
getscom.c  putscom.c  sram.h  xscom.c
[root@localhost xscom-utils]# cd ../gard/
[root@localhost gard]# make
cc -O2 -Wall -Werror -I. -c version.c -o version.o
cc -O2 -Wall -Werror -I. -c gard.c -o gard.o
cc -O2 -Wall -Werror -I. -c libflash/file.c -o libflash-file.o
cc -O2 -Wall -Werror -I. -c libflash/libflash.c -o libflash-libflash.o
cc -O2 -Wall -Werror -I. -c libflash/libffs.c -o libflash-libffs.o
cc -O2 -Wall -Werror -I. -c libflash/ecc.c -o libflash-ecc.o
cc -O2 -Wall -Werror -I. -c libflash/blocklevel.c -o libflash-blocklevel.o
    CC  common-arch_flash_common.o
    CC  common-arch_flash_powerpc.o
ld -r common-arch_flash_common.o common-arch_flash_powerpc.o -o common-arch_flash.o
ld: cannot find common-arch_flash_common.o: No such file or directory
ld: cannot find common-arch_flash_powerpc.o: No such file or directory
make: *** [common-arch_flash.o] Error 1 

And tried with make clean also, hitting the same issue.
cd skiboot/external/xscom-utils/
 make clean
rm -rf *.[od] getscom putscom getsram
make
    CC  xscom.o
    CC  version.o
  LINK  getscom
    CC  putscom.o
  LINK  putscom
    CC  getsram.o
    CC  sram.o
  LINK  getsram
ls
getscom.c  getsram.c  Makefile  putscom.c  sram.c  sram.h  version.c  xscom.c  xscom.h

Object and executable files are not getting created.
FW Level
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-3b7310d
 Product Extra         : 	buildroot-bc248fa
 Product Extra         : 	skiboot-5.2.2
 Product Extra         : 	hostboot-15bb028-9824b9f
 Product Extra         : 	linux-4.4.9-openpower1-cbbfd90
 Product Extra         : 	petitboot-v1.0.1-8f13d5f
 Product Extra         : 	habanero-xml-5a2a541-cb4e251
 Product Extra         : 	occ-19a3ebc
 Product Extra         : 	hostboot-binaries-7f593a3
 Product Extra         : 	c

cyrilbur-ibm, 2016-05-16:
Hi, @pridhiviraj 

Could you please tell me what branch/tag/commit you're at when trying this.

I have tried the gard tool with skiboot-5.2.2 tag and it appears to work fine, I get all the object files and the binary gard file.

```
0 cyrilb@p84:~ $ git clone https://github.com/open-power/skiboot.git
Cloning into 'skiboot'...
remote: Counting objects: 17317, done.
remote: Compressing objects: 100% (3220/3220), done.
remote: Total 17317 (delta 14154), reused 17098 (delta 14014)
Receiving objects: 100% (17317/17317), 11.39 MiB | 0 bytes/s, done.
Resolving deltas: 100% (14154/14154), done.
Checking connectivity... done.
0 cyrilb@p84:~ $ cd skiboot/
0 cyrilb@p84:~/skiboot $ git checkout skiboot-5.2.2
Note: checking out 'skiboot-5.2.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at a49d21c... Add skiboot 5.2.2 release notes
0 cyrilb@p84:~/skiboot $ cd external/gard/
0 cyrilb@p84:~/skiboot/external/gard $ ls
Makefile  Makefile.dist  config.h  gard.c  gard.h  opal-gard.1  rules.mk  test
0 cyrilb@p84:~/skiboot/external/gard $ make
ln -sf ../../make_version.sh
ln -sf ../../libflash .
ln -sf ../../ccan .
ln -sf ../common .
ln -sf ../../hw/ast-bmc/ast-sf-ctrl.c common/ast-sf-ctrl.c
ln -sf ../../include/ast.h common/ast.h
ln -sf arch_flash_powerpc_io.h common/io.h
cc -O2 -Wall -Werror -I. -c version.c -o version.o
cc -O2 -Wall -Werror -I. -c gard.c -o gard.o
cc -O2 -Wall -Werror -I. -c libflash/file.c -o libflash-file.o
cc -O2 -Wall -Werror -I. -c libflash/libflash.c -o libflash-libflash.o
cc -O2 -Wall -Werror -I. -c libflash/libffs.c -o libflash-libffs.o
cc -O2 -Wall -Werror -I. -c libflash/ecc.c -o libflash-ecc.o
cc -O2 -Wall -Werror -I. -c libflash/blocklevel.c -o libflash-blocklevel.o
cc -O2 -Wall -Werror -I.  -c common/arch_flash_common.c -o common-arch_flash_common.o
cc -O2 -Wall -Werror -I.  -c common/arch_flash_powerpc.c -o common-arch_flash_powerpc.o
ld -r common-arch_flash_common.o common-arch_flash_powerpc.o -o common-arch_flash.o
cc -O2 -Wall -Werror -I. version.o gard.o libflash-file.o libflash-libflash.o libflash-libffs.o libflash-ecc.o libflash-blocklevel.o common-arch_flash.o -o gard
0 cyrilb@p84:~/skiboot/external/gard $ gcc --version
gcc (Ubuntu 5.2.1-22ubuntu2) 5.2.1 20151010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

0 cyrilb@p84:~/skiboot/external/gard $ ls
Makefile       common                      common-arch_flash_powerpc.o  gard.c  libflash               libflash-file.o      make_version.sh  test
Makefile.dist  common-arch_flash.o         config.h                     gard.h  libflash-blocklevel.o  libflash-libffs.o    opal-gard.1      version.c
ccan           common-arch_flash_common.o  gard                         gard.o  libflash-ecc.o         libflash-libflash.o  rules.mk         version.o
```

Same result at (master)
commit 94c7f9a21d35c4237a8e5ee82199e69935b02e28
Author: Michael Neuling <mikey@neuling.org>
Date:   Wed May 11 20:53:10 2016 +1000

shenki, 2016-05-16:
@pridhiviraj can you please paste the output of the `version.c` file in your skiboot build tree. It is generated at build time, so you need to do this after typing `make` at the top level.

```
$ cd skiboot
$ make
$ cat version.c
$ cat version.c 
const char version[] = "skiboot-5.2.0-1"
```

pridhiviraj, 2016-05-16:
@cyrilbur-ibm Hi it is same issue with below tag and commit levels.
git checkout -b skiboot-5.2.2 
Switched to a new branch 'skiboot-5.2.2'
commit  94c7f9a21d35c4237a8e5ee82199e69935b02e28
make
    LN  libflash
    LN  ccan
    LN  common
    LN  common/ast-sf-ctrl.c
    LN  common/ast.h
    LN  common/io.h
    LN  make_version.sh
cc -O2 -Wall -Werror -I. -c version.c -o version.o
cc -O2 -Wall -Werror -I. -c gard.c -o gard.o
cc -O2 -Wall -Werror -I. -c libflash/file.c -o libflash-file.o
cc -O2 -Wall -Werror -I. -c libflash/libflash.c -o libflash-libflash.o
cc -O2 -Wall -Werror -I. -c libflash/libffs.c -o libflash-libffs.o
cc -O2 -Wall -Werror -I. -c libflash/ecc.c -o libflash-ecc.o
cc -O2 -Wall -Werror -I. -c libflash/blocklevel.c -o libflash-blocklevel.o
    CC  common-arch_flash_common.o
    CC  common-arch_flash_powerpc.o
ld -r common-arch_flash_common.o common-arch_flash_powerpc.o -o common-arch_flash.o
ld: cannot find common-arch_flash_common.o: No such file or directory
ld: cannot find common-arch_flash_powerpc.o: No such file or directory
make: *** [common-arch_flash.o] Error 1

Even at master branch also hitting the same issue.
94c7f9a21d35c4237a8e5ee82199e69935b02e28

shenki, 2016-05-16:
If you type "make" a second or third time does it then succeed?

pridhiviraj, 2016-05-16:
i tried more than 5 times, still it is same issue.

cyrilbur-ibm, 2016-05-16:
Could you paste the output of `make --version` and `make V=1` please

pridhiviraj, 2016-05-17:
Hi
make V=1
gcc -O2 -g -Wall -m64   -c xscom.c -o xscom.o
gcc -O2 -g -Wall -m64   -c version.c -o version.o
gcc   -o getscom getscom.c xscom.o version.o
gcc -O2 -g -Wall -m64   -c putscom.c -o putscom.o
gcc   -o putscom putscom.o xscom.o version.o
gcc -O2 -g -Wall -m64   -c getsram.c -o getsram.o
gcc -O2 -g -Wall -m64   -c sram.c -o sram.o
gcc   -o getsram getsram.o xscom.o sram.o version.o

 make --version
GNU Make 3.82
Built for powerpc64le-koji-linux-gnu
Copyright (C) 2010  Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

cooktail, 2016-05-31:
you must use powerpc gcc.  make  CC=powerpc-linux-gcc

cyrilbur-ibm, 2016-05-31:
Sorry for the delay @pridhiviraj, I must have missed a notification somewhere. It would have been easier if you had reproduced all your previous output using make V=1. 

Did you get a binary, I don't see any errors?

Also I know the gard tool better so the make V=1 output of that might help.

Thanks @cooktail, this is running on a powerpc machine so `gcc` is a powerpc gcc. Either way, the gard tool at least should also compile for x86.

pridhiviraj, 2016-06-03:
Hi @cyrilbur-ibm 
This issue seen on only one system and on only one of the os's . It is working fine on remaining all the systems.
Currently i don't have the system , where the problem is observed.

cyrilbur-ibm, 2016-06-06:
Hi @pridhiviraj,
Thanks for getting back to me. I think I'll close this issue for now just so we don't end up with this hanging around.

When you replicate the issue, please open another issue and reference this one.

Thanks!

closed 2016-06-06 by shenki