open-power/skiboot #33: No failure report during the RAS- VPD not updated in RAS function test.

kyewonh, 2016-06-17:
DIMMs during IPL( RAS: Pulling out one dimm at IPL time brings all of 8 dimms under CPU2 offline #76 ).
Summary of issues :
1. Powered on the system with 2 Bad drams.
2. training failed and issued reconfig loop.
3. memdiag failed and created logs saying both spare and chip mark used. 

| PRD Signature : 0x000D0002 0xFFFF0043 |
| Signature Description : mba(n0p1c0) () RDR: Both spare and chip mark used |
|------------------------------------------------------------------------------|

| mba(n0p1c0) (0x000d0002) : ********************************************* |
| DRAM_REPAIRS_DATA : 2 |
| : Rank: 0 CM: 68 SM: -- EccSp: 70 |
| : Rank: 1 CM: 68 SM: -- EccSp: 70 |
| DRAM_REPAIRS_VPD : 2 |
| : R:0 ff0000000000000000ff 000000000000000000ff |
| : R:1 ff0000000000000000ff 000000000000000000ff |
| |

4. Memdiag issues reconfig loop and failed again at 12.4 mss_eff_config complaining about DIMM plugging rules
| Created by : hwpf |
| HwpReturnCode : RC_MSS_EFF_GROUPING_UNABLE_TO_GROUP_MCS |
| HWP Error description : mss_eff_grouping found an MCS that could not be gr|
| : ouped. This is a memory plugging error. Refer to m|
| : emory plugging rules. The attached Memory Buffer C|
| : hip is deconfigured as a result |
|------------------------------------------------------------------------------|
| User Defined Data |

This issues reconfig loop and taking out total of 8 DIMMs 
5. powered off the system, removed tape on the DIMM and moved to new slot . Removed gard and powered on the system again.
There is no training fail and no memdiag fail. Its a clean IPL, which proves VPD is not updated 
  --- Procedure and output files --------------------
I added all results for taped 8DQs and followed this procedure.
1. Power off
2. populated with all good DIMM. 
3. Clear vpd.
4. Power on
5. Power off
6. test mode
7. Power on --> 1'st IPL log : file name --> 1stIPL 
8. Power off
9. Taped 8DQ pin on the DIMM
10. Power on --> 2'nd IPL : file name --> 2ndIPL_taped_8DQs
11. get sel list --> file name --> taped_8DQs_g_sel_list
12. get eSEL log --> file name --> taped_8DQs_eSEL_log_p05_NBA0p0.txt
13.Power off the system
14. Downlaod two logs before power off to Ubuntu hard disk
15. Remove 8 DQ TAPE on the DIMM
16. Move next slot
17. Power on and save tee istep -> file name --> 3IPL_removed_8DQ_p5_NBA0_p1
18. Sel list and copy output -> file name --> removed_8DQ_sel_list.txt
19. Run esel .pl and get log -> file name --> removed_8DQ_eSEL_log.txt

Also Added 2 log files gard list, gard show data.

[1stIPL.txt](https://github.com/open-power/skiboot/files/320943/1stIPL.txt)
[2ndIPL_taped_8DQs.txt](https://github.com/open-power/skiboot/files/320941/2ndIPL_taped_8DQs.txt)
[3IPL_removed_8DQ_p5_NBA0_p1.txt](https://github.com/open-power/skiboot/files/320946/3IPL_removed_8DQ_p5_NBA0_p1.txt)
[removed_8DQ_eSEL_log.txt](https://github.com/open-power/skiboot/files/320945/removed_8DQ_eSEL_log.txt)
[removed_8DQ_sel_list.txt](https://github.com/open-power/skiboot/files/320944/removed_8DQ_sel_list.txt)
[syslog1_taped_8DQs.txt](https://github.com/open-power/skiboot/files/320942/syslog1_taped_8DQs.txt)
[taped_8DQs_eSEL_log_p05_NBA0p0.txt](https://github.com/open-power/skiboot/files/320947/taped_8DQs_eSEL_log_p05_NBA0p0.txt)
[taped_8DQs_g_sel_list.txt](https://github.com/open-power/skiboot/files/320948/taped_8DQs_g_sel_list.txt)
[msglog1_taped_8DQs.txt](https://github.com/open-power/skiboot/files/320950/msglog1_taped_8DQs.txt)
[syslog1_taped_8DQs.txt](https://github.com/open-power/skiboot/files/320949/syslog1_taped_8DQs.txt)

williamspatrick, 2016-06-17:
Skiboot is not involved with this. Please move issue to either the op-build or hostboot repository. 

Also, we need to know what system and code level you were running on. If you are not running vanilla open source code, we need to know that. It is possible this is not the right support path for your problems.

closed 2016-06-17 by williamspatrick

nguyendtatibm, 2016-06-17:
Hi Patrick,

This issue was mistakenly posted here.  I've ask the submitter to move it to his correct private area.

Regards,

Trinh