open-power/skiboot #34: assertion in mem_region_release_unused-gcov test

shenki, 2016-07-18:
Running `skiboot-5.2.4` (65e436ad726aac81fe8acca55868afdf9d5f3161), built with`gcc (Ubuntu 5.4.0-6ubuntu1) 5.4.0 20160609`:

```
$ gdb core/test/run-mem_region_release_unused-gcov
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Reading symbols from core/test/run-mem_region_release_unused-gcov...done.
(gdb) r
Starting program: /home/joel/dev/skiboot/core/test/run-mem_region_release_unused-gcov 
MEM: parsing reserved memory from reserved-names/-ranges properties
Releasing unused memory:
    ibm,firmware-heap: 0/4096 used
    ibm,firmware-allocs-memory@7ffff7ecc010: 1040/1048576 used
run-mem_region_release_unused-gcov: core/test/run-mem_region_release_unused.c:166: main: Assertion `r->start == other->start + other->len' failed.

Program received signal SIGABRT, Aborted.
0x00007ffff7a43418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt full
#0  0x00007ffff7a43418 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
        resultvar = 0
        pid = 10725
        selftid = 10725
#1  0x00007ffff7a4501a in __GI_abort () at abort.c:89
        save_stage = 2
        act = {__sigaction_handler = {sa_handler = 0x4, sa_sigaction = 0x4}, sa_mask = {
            __val = {93824992245216, 140737488346800, 140737488346192, 0, 140737354092544, 
              93824992298416, 166, 93824992299189, 0, 0, 140737348442812, 140737349540392, 
              140737349554032, 93824994436000, 140737349540392, 93824992298416}}, 
          sa_flags = -134262784, sa_restorer = 0x5555555645b0}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007ffff7a3bbd7 in __assert_fail_base (fmt=<optimised out>, 
    assertion=assertion@entry=0x5555555645b0 "r->start == other->start + other->len", 
    file=file@entry=0x555555564460 "core/test/run-mem_region_release_unused.c", 
    line=line@entry=166, 
    function=function@entry=0x5555555648b5 <__PRETTY_FUNCTION__.5091> "main") at assert.c:92
        str = 0x55555576e9c0 "run-mem_region_release_unused-gcov: core/test/run-mem_region_release_unused.c:166: main: Assertion `r->start == other->start + other->len' failed.\n"
        total = 4096
#3  0x00007ffff7a3bc82 in __GI___assert_fail (
    assertion=0x5555555645b0 "r->start == other->start + other->len", 
    file=0x555555564460 "core/test/run-mem_region_release_unused.c", line=166, 
    function=0x5555555648b5 <__PRETTY_FUNCTION__.5091> "main") at assert.c:101
No locals.
#4  0x00005555555612c3 in main () at core/test/run-mem_region_release_unused.c:166
        i = 8
        r = 0x55555576e340
        other = 0x55555576e280
        other_mem = 0x7ffff7ecc010
        last = 0x7fffffffdeb0 "\001"
        __PRETTY_FUNCTION__ = "main"
```

shenki, 2016-07-18:
```(gdb) p r->len 
$3 = 13631488
(gdb) p r->start 
$4 = 808452096
(gdb) p other->start 
$5 = 140737352876048
(gdb) p other->len 
$6 = 1040
```

shenki, 2016-07-18:
```
(gdb) p r
$7 = (struct mem_region *) 0x55555576e340
(gdb) p other
$8 = (struct mem_region *) 0x55555576e280
```

gery-toulouse, 2018-09-04:
Closing as not commented for 2 years. Please retest with a recent version and reopen if needed

closed 2018-09-04 by sammj