open-power/skiboot #35: pflash build has a parallel build race condition.

williamspatrick, 2016-07-22:
```
| NOTE: make -j 32 CROSS_COMPILE=arm-openbmc-linux-gnueabi- SKIBOOT_VERSION=5.2.4 PFLASH_VERSION=5.2.4 V=1 -C external/pflash all LINKAGE=dynamic
| make: Entering directory `/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.2.4-r0/git/external/pflash'
| ln -sf ../../libflash ./libflash
| ln -sf ../../ccan ./ccan
| ln -sf ../common ./common
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/sysroots/barreleye  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.2.4-r0=/usr/src/debug/pflash/5.2.4-r0 -fdebug-prefix-map=/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/sysroots/barreleye=  -O2 -Wall -I. -c pflash.c -o pflash.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/sysroots/barreleye  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.2.4-r0=/usr/src/debug/pflash/5.2.4-r0 -fdebug-prefix-map=/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/tmp/openbmc.612a6908196e60ee1251a072ba621e167614170f.66UDBxA3/build/tmp/sysroots/barreleye=  -O2 -Wall -I. -c progress.c -o progress.o
| make: *** No rule to make target `../shared/libflash.so.skiboot-5.2.4', needed by `pflash'.  Stop.
| make: *** Waiting for unfinished jobs....
```

Sometimes we are failing with this.

shenki, 2016-07-22:
I could reproduce locally by starting with a completely clean directory and building with the same arguments as the bitbake recipe: 

```
git clean -x -f
make -j 32 CROSS_COMPILE=arm-linux-gnueabi- SKIBOOT_VERSION=5.2.4 PFLASH_VERSION=5.2.4 V=1 -C external/pflash all LINKAGE=dynamic
```

It fails between 10% and 20% of the time.

I made some cleanups that appear to resolve it for me:

https://lists.ozlabs.org/pipermail/skiboot/2016-July/004128.html

I suggest waiting until @stewart-ibm applies this and pulling in a new tag.

stewart-ibm, 2016-07-27:
Joel Stanley <notifications@github.com> writes:
> I suggest waiting until @stewart-ibm applies this and pulling in a new
> tag.

Expect skiboot 5.2.5 either tomorrow or Monday.

-- 
Stewart Smith
OPAL Architect, IBM.

stewart-ibm, 2017-02-23:
I'm going to assume this is properly fixed now (that patch was merged a while ago).

I also assume it's fixed because @williamspatrick isn't bugging me about it :)

closed 2017-02-23 by stewart-ibm