open-power/skiboot #36: Observed xscom write errors during an HMI Error injection tests.

pridhiviraj, 2016-08-23:
Install a OpenPower Firestone system with firestone_update.pnor which is built from op-build master-next branch. HMI error injection tests results in xscom write errors.

Timer facility error injection test:
 ./putscom -c 00000008 14013281 0003080000000000;echo $?
22:38:48 [911178433037,3] XSCOM: write error gcid=0x0 pcb_addr=0x14013281 stat=0x3
22:38:48 [911178448677,6] IPMI: dropping non severe PEL event
22:38:48 [911178451611,3] XSCOM: Write failed, ret =  -6
22:38:48 Error -5 writing XSCOM

HMI Malfunction alert(Core checkstop) error injection:
./putscom -c 00000008 1b013100 1000000000000000
22:38:48 [3048108888610,3] XSCOM: write error gcid=0x0 pcb_addr=0x1b013100 stat=0x3
22:38:48 [3048108904555,6] IPMI: dropping non severe PEL event
22:38:48 [3048108907750,3] XSCOM: Write failed, ret =  -6
22:38:48 Error -5 writing XSCOM

Firmware version: PNOR level
System Firmware (ID 47)
Product Name : OpenPOWER Firmware
Product Version : open-power-firestone-23bdc73
Product Extra : buildroot-81b8d98
Product Extra : skiboot-5.3.1
Product Extra : hostboot-d9ee1a1-be2f697
Product Extra : linux-4.4.18-openpower1-5906f7a
Product Extra : petitboot-v1.2.2-7fc085b
Product Extra : firestone-xml-783df1f
Product Extra : occ-bb092e9
Product Extra : hostboot-binaries-7f593a3
Product Extra : capp-u

BMC FW level:
cat /proc/ractrends/Helper/FwInfo
FW_VERSION=2.13.98117
FW_DATE=Jun 15 2016
FW_BUILDTIME=11:26:40 CDT
FW_DESC=8335 SRC BUILD RR9 06152016
FW_PRODUCTID=1
FW_RELEASEID=RR9
FW_CODEBASEVERSION=2.X

maheshsal, 2016-08-23:
Did you disable CPU idle states ? However disabling CPU idle state does not mean that it would wake up CPUs that are already in idle state. After disabling CPU idle states you may have to wait for some time before you start injection so that all COREs come out of idle OR run 'ls' command on each CPU so that they come out of idle state.

pridhiviraj, 2016-08-24:
Hi Mahesh
As starting of test itself we are disabling cpu idle states. And also Error injection happened before for other cores, it is failing for some of cores with xscom write errors.

22:38:48 [pexpect]#./putscom -c 00000008 11013100 0000000000100000; echo $?
22:38:48 ./putscom -c 00000008 11013100 0000000000100000; echo $?
22:38:48 0000000000100008
22:38:48 0
22:38:48 [pexpect]#Injected thread hang recoverable error
22:38:48 dmesg
22:38:48 dmesg
22:38:48 [ 4214.569767] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4214.569774]  Error detail: Processor Recovery done
22:38:48 [ 4214.569779] 	HMER: 2040400000000000
22:38:48 [ 4214.569784] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4214.569788]  Error detail: Processor Recovery done
22:38:48 [ 4214.569792] 	HMER: 2040400000000000
22:38:48 [ 4214.569797] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4214.569801]  Error detail: Processor Recovery done
22:38:48 [ 4214.569805] 	HMER: 2040400000000000
22:38:48 [ 4214.569810] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4214.569814]  Error detail: Processor Recovery done
22:38:48 [ 4214.569819] 	HMER: 2040400000000000
22:38:48 [ 4214.569823] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4214.569827]  Error detail: Processor Recovery done
22:38:48 [ 4214.569831] 	HMER: 2040400000000000
22:38:48 [ 4214.569836] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4214.569840]  Error detail: Processor Recovery done
22:38:48 [ 4214.569844] 	HMER: 2040400000000000
22:38:48 [ 4214.569848] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4214.569852]  Error detail: Processor Recovery done
22:38:48 [ 4214.569856] 	HMER: 2040400000000000
22:38:48 [ 4214.569861] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4214.569865]  Error detail: Processor Recovery done
22:38:48 [ 4214.569869] 	HMER: 2040400000000000
22:38:48 [pexpect]#Processor recovery done
22:38:48 dmesg -C
22:38:48 dmesg -C
22:38:48 [pexpect]#./putscom -c 00000008 12013100 0000000000100000; echo $?
22:38:48 ./putscom -c 00000008 12013100 0000000000100000; echo $?
22:38:48 0000000000000008
22:38:48 0
22:38:48 [pexpect]#Injected thread hang recoverable error
22:38:48 dmesg
22:38:48 dmesg
22:38:48 [ 4300.039215] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4300.039223]  Error detail: Processor Recovery done
22:38:48 [ 4300.039228] 	HMER: 2040400000000000
22:38:48 [ 4300.039233] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4300.039237]  Error detail: Processor Recovery done
22:38:48 [ 4300.039241] 	HMER: 2040400000000000
22:38:48 [ 4300.039246] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4300.039249]  Error detail: Processor Recovery done
22:38:48 [ 4300.039253] 	HMER: 2040400000000000
22:38:48 [ 4300.039258] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4300.039262]  Error detail: Processor Recovery done
22:38:48 [ 4300.039265] 	HMER: 2040400000000000
22:38:48 [ 4300.039270] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4300.039274]  Error detail: Processor Recovery done
22:38:48 [ 4300.039278] 	HMER: 2040400000000000
22:38:48 [ 4300.039282] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4300.039286]  Error detail: Processor Recovery done
22:38:48 [ 4300.039290] 	HMER: 2040400000000000
22:38:48 [ 4300.039294] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4300.039298]  Error detail: Processor Recovery done
22:38:48 [ 4300.039302] 	HMER: 2040400000000000
22:38:48 [ 4300.039306] Harmless Hypervisor Maintenance interrupt [Recovered]
22:38:48 [ 4300.039310]  Error detail: Processor Recovery done
22:38:48 [ 4300.039314] 	HMER: 2040400000000000
22:38:48 [pexpect]#Processor recovery done
22:38:48 dmesg -C
22:38:48 dmesg -C
22:38:48 [pexpect]#./putscom -c 00000008 14013100 0000000000100000; echo $?
22:38:48 ./putscom -c 00000008 14013100 0000000000100000; echo $?
22:38:48 [2267967594591,3] XSCOM: write error gcid=0x0 pcb_addr=0x14013100 stat=0x3
22:38:48 [2267967610719,6] IPMI: dropping non severe PEL event
22:38:48 [2267967613867,3] XSCOM: Write failed, ret =  -6
22:38:48 Error -5 writing XSCOM

And the same test case passed for earlier runs.

maheshsal, 2016-08-24:
What I meant is, apart from disabling CPU idle states you need run 'ls' command on each CPU so that they come out of idle state before you start the injections. Disabling CPU idle state does not mean that it would wake up CPUs that are already in idle state. This action prevent any CPU to go into idle mode in future.

e.g.
$ Disable idle states
$ for cpu in $(cat /proc/cpuinfo | grep -i processor| awk '{ print $3}')
do
	taskset -c $cpu ls
done

Above for loop should wake up all the CPUs from idle states. And then all putscom should run fine.

pridhiviraj, 2016-08-24:
Thanks mahesh, i will test with above check and will let you know the results.

pridhiviraj, 2016-08-24:
Hi Mahesh,
Executed with above steps after disabling CPU idle states, seen same XSCOM write errors exactly for same chip and core.

[pexpect]#./putscom -c 00000008 12013281 0003080000000000;echo $?
./putscom -c 00000008 12013281 0003080000000000;echo $?
0000080000000000
0
[pexpect]#Injected TFMR error 0003080000000000
dmesg
dmesg
[ 1588.426281] Severe Hypervisor Maintenance interrupt [Recovered]
[ 1588.426288]  Error detail: Timer facility experienced an error
[ 1588.426294] 	HMER: 0840000000000000
[ 1588.426299] 	TFMR: 3612000980a44000
[pexpect]#Timer facility experienced an error and got recovered
dmesg -C
dmesg -C
[pexpect]#./putscom -c 00000008 14013281 0003080000000000;echo $?
./putscom -c 00000008 14013281 0003080000000000;echo $?
[877177176114,3] XSCOM: write error gcid=0x0 pcb_addr=0x14013281 stat=0x3
[877177189895,6] IPMI: dropping non severe PEL event
[877177192722,3] XSCOM: Write failed, ret =  -6
Error -5 writing XSCOM
1
[pexpect]#dmesg -C
dmesg -C
[pexpect]#./putscom -c 00000008 15013281 0003080000000000;echo $?
./putscom -c 00000008 15013281 0003080000000000;echo $?
0000080000000000
0
[pexpect]#Injected TFMR error 0003080000000000
dmesg
dmesg
[ 1697.536591] Severe Hypervisor Maintenance interrupt [Recovered]
[ 1697.536600]  Error detail: Timer facility experienced an error
[ 1697.536605] 	HMER: 0840000000000000
[ 1697.536609] 	TFMR: 3612000980a44000
[pexpect]#Timer facility experienced an error and got recovered

gery-toulouse, 2018-09-04:
Should be closed as not updated for nearly 18 months. Please reopen if needed.

closed 2018-09-04 by sammj