open-power/skiboot #37: Use or get rid of systemd socket

r4f4, 2016-08-25:
Looking at the opal-prd code, I noticed the systemd-created socket is not being used. The code just handles its own socket and ignores systemd's one. There are 2 ways to solve that:

1) don't use systemd sockets. opal-prd already does all the socket handling. That means the file opal-prd.socket can be removed and opal-prd.service depency on it, dropped.

2) ask systemd to hand over the socket it created with opal-prd.socket. If there is no need to support non-systemd systems, the whole socket handling part can be removed from the code and the function 'sd_listen_fds' from systemd can be used to retrieve the socket's fd. A code snippet for that can be seen in [1]. In case there is any interest in this, I can work on a patch.

[1] http://0pointer.de/blog/projects/socket-activation.html

jk-ozlabs, 2016-09-23:
I think just removing it would be fine. The original `.socket` unit definition was based on other boilerplate code, so there's no real reason we need it.

r4f4, 2016-09-23:
I just sent a pull request fixing the issue.

stewart-ibm, 2016-11-02:
fixed in 5.4.0-rc3

THnaks all!

closed 2016-11-02 by stewart-ibm