open-power/skiboot #38: Support for AST2500 flash registers needed.

williamspatrick, 2016-09-02:
We discovered that libflash / pflash do not currently support the new register layout for the AST2500.  Is this work planned to be done in the near future?  We have systems coming on-line soon that are going to require this support and we need to integrate it back into openbmc.

Please see openbmc/openbmc#538 and openbmc/openbmc#541 for additional background.

cyrilbur-ibm, 2016-09-06:
I've started writing a driver with the new layout. Shouldn't take too long I suppose, would it be best if I was assigned to it?

shenki, 2016-09-07:
Oops. @williamspatrick I think @cyrilbur-ibm needs to be added to a group so I can assign the issue to him?

williamspatrick, 2016-10-05:
@cyrilbur-ibm - Any update on this?  Work seems to have ceased lately.

shenki, 2016-10-06:
@cyrilbur-ibm has been working on this. The ast2500 involves the strapping registers in the mix, making it harder than the ast2400 to do from userspace

GHF, 2016-10-11:
Can folks who work on this please drop me some comments re: https://lists.ozlabs.org/pipermail/openbmc/2016-September/004670.html ? Specifically, does this missing support only affect the "updating PNOR from host" case?

shenki, 2016-10-11:
@GHF Sorry, I missed that email when you sent it. I will reply to it today.

stewart-ibm, 2017-08-15:
We seem to have solved our problems with MTD devices and related fun. Going to close this out.

closed 2017-08-15 by stewart-ibm