open-power/skiboot #41: With latest skiboot code on a firenze system failed to boot during p8_i2c_init.

pridhiviraj, 2016-09-27:
Not sure what is exactly the cause for this failure, But tried same skiboot lid one time it is worked and later on boot is aborting in skiboot code.
[   83.344371236,5] SkiBoot skiboot-5.3.2-191-g69a9337 starting...
[   83.344378129,5] initial console log level: memory 7, driver 5
[   83.344381077,6] CPU: P8 generation processor(max 8 threads/core)
[   83.344383798,7] CPU: Boot CPU PIR is 0x0870 PVR is 0x004b0201
[   83.344386757,7] CPU: Initial max PIR set to 0x1fff
[   83.345046793,5] OPAL table: 0x300b6e40 .. 0x300b7310, branch table: 0x30002000
[   83.345053589,7] Parsing HDAT...
[   83.345055592,5] SPIRA-S found.
[   83.345061239,6] Got PCIA !
[   83.345064476,6] CORE[0]: HW_PROC_ID=2 PROC_CHIP_ID=0 EC=0x21 OK
[   83.345068323,6] CORE[0]: PIR=48 RES=48 OK (8 threads)
[   83.345071816,6]     Cache: I=32 D=64/512/8192/0
[   83.345124160,6] CORE[1]: HW_PROC_ID=3 PROC_CHIP_ID=0 EC=0x21 OK
[   83.345127666,6] CORE[1]: PIR=96 RES=96 OK (8 threads)
[   83.345130847,6]     Cache: I=32 D=64/512/8192/0
[   83.345183008,6] CORE[2]: HW_PROC_ID=4 PROC_CHIP_ID=0 EC=0x21 OK
[   83.345186506,6] CORE[2]: PIR=104 RES=104 OK (8 threads)
[   83.345189800,6]     Cache: I=32 D=64/512/8192/0
[   83.345243971,6] CORE[3]: HW_PROC_ID=5 PROC_CHIP_ID=0 EC=0x21 OK
[   83.345247410,6] CORE[3]: PIR=112 RES=112 OK (8 threads)
[   83.345250727,6]     Cache: I=32 D=64/512/8192/0
[   83.345306459,6] CORE[4]: HW_PROC_ID=6 PROC_CHIP_ID=1 EC=0x21 OK
[   83.345309943,6] CORE[4]: PIR=160 RES=160 OK (8 threads)
[   83.345313210,6]     Cache: I=32 D=64/512/8192/0
[   83.345370978,6] CORE[5]: HW_PROC_ID=8 PROC_CHIP_ID=1 EC=0x21 OK
[   83.345374363,6] CORE[5]: PIR=176 RES=176 OK (8 threads)
[   83.345377532,6]     Cache: I=32 D=64/512/8192/0
[   83.345437230,6] CORE[6]: HW_PROC_ID=9 PROC_CHIP_ID=1 EC=0x21 OK
[   83.345440672,6] CORE[6]: PIR=224 RES=224 OK (8 threads)
[   83.345443952,6]     Cache: I=32 D=64/512/8192/0
[   83.345506142,6] CORE[7]: HW_PROC_ID=11 PROC_CHIP_ID=1 EC=0x21 OK
[   83.345509699,6] CORE[7]: PIR=240 RES=240 OK (8 threads)
[   83.345512998,6]     Cache: I=32 D=64/512/8192/0
[   83.345577243,6] CORE[8]: HW_PROC_ID=12 PROC_CHIP_ID=2 EC=0x21 OK
[   83.345580866,6] CORE[8]: PIR=2080 RES=2080 OK (8 threads)
[   83.345584245,6]     Cache: I=32 D=64/512/8192/0
[   83.345650462,6] CORE[9]: HW_PROC_ID=13 PROC_CHIP_ID=2 EC=0x21 OK
[   83.345653998,6] CORE[9]: PIR=2088 RES=2088 OK (8 threads)
[   83.345657241,6]     Cache: I=32 D=64/512/8192/0
[   83.345725430,6] CORE[10]: HW_PROC_ID=14 PROC_CHIP_ID=2 EC=0x21 OK
[   83.345728954,6] CORE[10]: PIR=2096 RES=2096 OK (8 threads)
[   83.345732321,6]     Cache: I=32 D=64/512/8192/0
[   83.345802194,6] CORE[11]: HW_PROC_ID=17 PROC_CHIP_ID=2 EC=0x21 OK
[   83.345805742,6] CORE[11]: PIR=2160 RES=2160 OK [boot] (8 threads)
[   83.345809567,6]     Cache: I=32 D=64/512/8192/0
[   83.345881043,6] CORE[12]: HW_PROC_ID=18 PROC_CHIP_ID=3 EC=0x21 OK
[   83.345884583,6] CORE[12]: PIR=2208 RES=2208 OK (8 threads)
[   83.345888000,6]     Cache: I=32 D=64/512/8192/0
[   83.345962527,6] CORE[13]: HW_PROC_ID=19 PROC_CHIP_ID=3 EC=0x21 OK
[   83.345966109,6] CORE[13]: PIR=2216 RES=2216 OK (8 threads)
[   83.345969454,6]     Cache: I=32 D=64/512/8192/0
[   83.346045787,6] CORE[14]: HW_PROC_ID=20 PROC_CHIP_ID=3 EC=0x21 OK
[   83.346049357,6] CORE[14]: PIR=2224 RES=2224 OK (8 threads)
[   83.346052787,6]     Cache: I=32 D=64/512/8192/0
[   83.346131208,6] CORE[15]: HW_PROC_ID=21 PROC_CHIP_ID=3 EC=0x21 OK
[   83.346134797,6] CORE[15]: PIR=2272 RES=2272 OK (8 threads)
[   83.346138123,6]     Cache: I=32 D=64/512/8192/0
[   83.346230153,6] IPLPARAMS: 1 serial ports in array
[   83.346233762,6] IPLPARAMS: Serial 0 rsrc: 2a00 loc: U78CB.001.WZS002B-P1-C1-T1
[   83.346246288,7] MS VPD: is at 0x31202ca0
[   83.346248727,7] MS VPD: MSAC is at 0x31202d10
[   83.346251619,7] MS VPD: TCMS is at 0x31202d2c
[   83.346254049,7] MS VPD: Maximum configured address: 0x8000002fffffffff
[   83.346257237,7] MS VPD: Maximum possible address: 0x8000002fffffffff
[   83.346260669,7] MS VPD: 0x31202ca0, area 0: installed functional shared
[   83.346477059,7]   Range: 0x0000002000000000..0x0000002800000000 on Chip 0x0 mattr: 0xa0000
[   83.346498710,7] MS VPD: 0x31202ca0, area 1: installed functional shared
[   83.346521324,7]   Range: 0x0000002000000000..0x0000002800000000 on Chip 0x0 mattr: 0xa0000
[   83.346534671,7] MS VPD: 0x31202ca0, area 2: installed functional shared
[... 328 lines cut ...]
[    1.181598874,7] CHIPTOD: PIR 0x0828 TB=29577a91
[    1.186720072,7] CHIPTOD: PIR 0x0830 TB=29a59f3d
[    1.191841664,7] CHIPTOD: PIR 0x0870 TB=29f3c576
[    1.191845273,7] CHIPTOD: PIR 0x08a0 TB=29f3d390
[    1.196966356,7] CHIPTOD: PIR 0x08a8 TB=2a41f7cb
[    1.202087738,7] CHIPTOD: PIR 0x08b0 TB=2a901d30
[    1.207209751,7] CHIPTOD: PIR 0x08e0 TB=2ade450e
[    1.212343017,7] CHIPTOD: TOD Topology in Use: Primary
[    1.212344512,7] CHIPTOD:   Primary configuration:
[    1.212345636,7] CHIPTOD:   Chip id: 0, Role: MDMT, Status: Active Master
[    1.212347195,7] CHIPTOD:   Secondary configuration:
[    1.212348252,7] CHIPTOD:   Chip id: 16, Role: MDMT, Status: Backup Master
[    1.212366090,3] DT: Unexpected property length /xscom@3fc0000000000/i2cm@a0020/clock-frequency
[    1.212369108,3] DT: Expected len: 4 got len: 8
[    1.212370117,0] Assert fail: core/device.c:603:0
[    1.212371550,0] Aborting!
CPU 0870 Backtrace:
 S: 0000000033dc39e0 R: 0000000030013758   .backtrace+0x24
 S: 0000000033dc3a60 R: 0000000030018e0c   ._abort+0x4c
 S: 0000000033dc3ae0 R: 0000000030018e88   .assert_fail+0x34
 S: 0000000033dc3b60 R: 0000000030023da4   .dt_require_property+0xb4
 S: 0000000033dc3bf0 R: 000000003002403c   .dt_prop_get_u32+0x14
 S: 0000000033dc3c60 R: 000000003004e884   .p8_i2c_init+0x12c
 S: 0000000033dc3e30 R: 0000000030014684   .main_cpu_entry+0x4a8
 S: 0000000033dc3f00 R: 00000000300025a0   boot_entry+0x198

stewart-ibm, 2016-09-28:
What FSP firmware version?

pridhiviraj, 2016-09-28:
system is in latest FW860.10 (TV860_047) driver b0922a_1640.861

stewart-ibm, 2016-09-28:
@hegdevasant has an idea and should post a patch shortly.

pridhiviraj, 2016-09-28:
cool, thanks

hegdevasant, 2016-09-28:
Please test with https://patchwork.ozlabs.org/patch/675988/

I've tested on my box and its working fine.

-Vasant

pridhiviraj, 2016-09-28:
Hi vasant
Tested the patch, system booted fine to petitboot without any abort in skiboot phase.
Thanks

stewart-ibm, 2016-09-28:
and merged. Thanks!

closed 2016-09-28 by stewart-ibm