open-power/skiboot #42: extract-gcov.c not 64-bit clean

andreiw, 2016-10-17:
The way this tool is written (i.e. struct gcov_info with pointers) assumes it can only be compiled on 64-bit hosts, while the build system neither enforces a 64-bit build nor ensures the host system is sane. Building on hosts with default 32-bit targets causes bunch of errors around padding/casting.

Seen on: Debian 8 on a PowerMac G5 (user space is mostly 32-bit, kernel is 64, gcc takes -m64 to create 64-bit bins).

The simplest fix I could see is passing -m64 to HOSTCC.

diff --git a/Makefile.main b/Makefile.main
index 86a54be..78804ec 100644
--- a/Makefile.main
+++ b/Makefile.main
@@ -33,7 +33,7 @@ CWARNS := -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs \
 # Host tools and options
 HOSTCC=gcc
 HOSTEND=$(shell uname -m | sed -e 's/^i.*86$$/LITTLE/' -e 's/^x86.*/LITTLE/' -e 's/^ppc64le/LITTLE/' -e 's/^ppc.*/BIG/')
-HOSTCFLAGS=-O1 $(CWARNS) -DHAVE_$(HOSTEND)_ENDIAN -MMD
+HOSTCFLAGS=-O1 $(CWARNS) -DHAVE_$(HOSTEND)_ENDIAN -MMD -m64
 HOSTCFLAGS += $(call try-cflag,$(HOSTCC),-std=gnu11)
 HOSTCFLAGS += $(call try-cflag,$(HOSTCC),-Wjump-misses-init) \
          $(call try-cflag,$(HOSTCC),-Wsuggest-attribute=const) \

andreiw, 2016-10-18:
Okay I filed this in a rush and the title doesn't make sense... I really meant that extract-gcov only correctly builds on 64-bit hosts, whereas usually "not 64-bit clean" means the exact opposite. I apologize for the confusion.

stewart-ibm, 2016-10-25:
Hah, that doesn't surprise me at all. That tool is possible the most omg-let-me-see-if-i-can-hack-this-together-to-barely-work-well-enough thing I've written in a *loooong* time.

In somewhat unsurprising news, you're the first person to build skiboot on 32bit in... ever?

I'm going with this method, which will check to ensure it's valid flag.

```
--- a/Makefile.main
+++ b/Makefile.main
@@ -35,6 +35,7 @@ HOSTCC=gcc
 HOSTEND=$(shell uname -m | sed -e 's/^i.*86$$/LITTLE/' -e 's/^x86.*/LITTLE/' -e 's/^ppc64le/LITTLE/' -e 's/^ppc.*/BIG/')
 HOSTCFLAGS=-O1 $(CWARNS) -DHAVE_$(HOSTEND)_ENDIAN -MMD
 HOSTCFLAGS += $(call try-cflag,$(HOSTCC),-std=gnu11)
+HOSTCFLAGS += $(call try-cflag,$(HOSTCC),-m64)
 HOSTCFLAGS += $(call try-cflag,$(HOSTCC),-Wjump-misses-init) \
          $(call try-cflag,$(HOSTCC),-Wsuggest-attribute=const) \
          $(call try-cflag,$(HOSTCC),-Wsuggest-attribute=noreturn) \
```

closed 2016-10-25 by stewart-ibm

andreiw, 2016-10-25:
Thank you, Stewart!