open-power/skiboot #44: gard tool is not listing hardware gard entries.

pridhiviraj, 2016-10-27:
Install a OpenPower habanero system with latest skiboot 5.4 rc2 and clone latest skiboot in host OS.
And try to list hardware gard entries. It is failing with below error.

skiboot/external/gard# ./gard list all
[  516.062048471,3] ECC data length must be 8 byte aligned length:2d
[  423.849745] mtd mtd0: opal_flash_async_op(op=0) failed (rc -6)
[  423.849745] mtd mtd0: opal_flash_async_op(op=0) failed (rc -6)
Does not appear to be sane gard data
A libflash/libffs error has occurred 15

 opal-gard list all
[  330.123050527,3] ECC data length must be 8 byte aligned length:2d
[  237.968901] mtd mtd0: opal_flash_async_op(op=0) failed (rc -6)
[  237.968901] mtd mtd0: opal_flash_async_op(op=0) failed (rc -6)
Does not appear to be sane gard data
A libflash/libffs error has occurred 15

[   47.065304451,5] SkiBoot skiboot-5.4.0-rc2 starting...

pridhiviraj, 2016-10-27:
gard clear also failing.
opal-gard  clr all
[ 2222.311452230,3] ECC data length must be 8 byte aligned length:2d
[  225.569401] mtd mtd0: opal_flash_async_op(op=0) failed (rc -6)
[  225.569401] mtd mtd0: opal_flash_async_op(op=0) failed (rc -6)
Does not appear to be sane gard data
A libflash/libffs error has occurred 15

stewart-ibm, 2016-10-27:
Paidipeddi Pridhiviraj <notifications@github.com> writes:
> gard clear also failing.
> opal-gard  clr all
> [ 2222.311452230,3] ECC data length must be 8 byte aligned length:2d
> [  225.569401] mtd mtd0: opal_flash_async_op(op=0) failed (rc -6)
> [  225.569401] mtd [...]

Do you have skiboot log for this?

We appear to have flash issues on Habanero (or some of them)

-- 
Stewart Smith
OPAL Architect, IBM.

pridhiviraj, 2016-10-28:
[opal_msglog_gard_failure.txt](https://github.com/open-power/skiboot/files/557585/opal_msglog_gard_failure.txt)

stewart-ibm, 2016-10-30:
```
[  174.049927033,3] ECC data length must be 8 byte aligned length:2d
```

@cyrilbur-ibm - thoughts? Any idea why this just seems to be Habanero?

cyrilbur-ibm, 2016-10-30:
I'm out today. Nothing springs to mind.

pridhiviraj, 2016-10-31:
@stewart-ibm looks like it is the same case in all OpenPower platforms.
Initially i have tested it on habanero only. But now it is tested in firestone system also.
# cat /sys/firmware/opal/msglog | grep -i aligned
[244837.218979592,3] ECC data length must be 8 byte aligned length:2d
# cat /sys/firmware/opal/msglog | grep -i firestone
[   66.000386016,5] PLAT: Detected Firestone platform
# cat /sys/firmware/opal/msglog | grep -i skiboot
[   64.345580918,5] SkiBoot skiboot-5.4.0-rc1-14-gad6c67f-shilpa-dirty-0621315 starting...

But it works fine with below skiboot version:
./gard list
No GARD entries to display
root@fir01:~/skiboot/external/gard# cat /sys/firmware/opal/msglog | grep -i skiboot
[33143767067,5] SkiBoot skiboot-5.3.7 starting...

So may be some recent changes making it to failure.

cyrilbur-ibm, 2016-10-31:
This makes more sense, I found it odd that this kind of message would be
specific to habaneros.

I agree.

cyrilbur-ibm, 2016-10-31:
It looks like we've changed where the ECC decoding boundary is. It definitely used to be that userspace tools (pflash, gard...) would be given raw flash data through /dev/mtd, that is, with ecc bytes. This was mostly because skiboot didn't really know which regions had ecc.
This has changed, skiboot does know and I think it might be removing them as well, obviously this isn't working since gard expects to have to remove/add the ecc its self.

stewart-ibm, 2016-11-02:
Confirmed fixed bi https://patchwork.ozlabs.org/patch/690270/ - will be fixed in 5.4.0-rc3

closed 2016-11-02 by stewart-ibm