open-power/skiboot #47: seeing device tree warnings on latest skiboot code

pridhiviraj, 2016-11-08:
Host: 
dtc -I fs -O dts /proc/device-tree > foo.dts
Warning (reg_format): "reg" property in /ibm,opal/flash@0 has invalid length (8 bytes) (#address-cells == 0, #size-cells == 0)
Warning (unit_address_vs_reg): Node /vpd/dimm@30000 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30002 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30004 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30006 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30008 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@3000a has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@3000c has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@3000e has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30010 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30012 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30014 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30016 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@30018 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@3001a has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@3001c has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/dimm@3001e has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/processor@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/processor@0/cpu@8 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/processor@0/cpu@10 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/processor@0/cpu@18 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /vpd/processor@0/cpu@20 has a unit name, but no reg property

Petitboot
/ # 
/ # dtc -I fs -O dts /proc/device-tree > foo.dts
Warning (reg_format): "reg" property in /ibm,opal/flash@0 has invalid length (8 bytes) (#address-cells == 0, #size-cells == 0)
/ # 
/ # 

reg_format warning is known issue, But unit_address_vs_reg is looks like a new one. And this is happening only on rhel and not on ubuntu or petitboot kernel.

pridhiviraj, 2016-11-08:
Not sure this one is a OPAL or kernel issue. Just raised here to inform, and some one can confirm whether these warnings will result in other problems, where the issue is?

stewart-ibm, 2016-11-09:
Hrm... what machine?

pridhiviraj, 2016-11-09:
habanero

gery-toulouse, 2018-09-04:
Could you please retest with a recent version (and platform), and close if not reproduced?

hegdevasant, 2018-09-04:
Most of device tree warnings are fixed ... AFAIK one or two issues are pending on P9 system.

I don't know if we still have any issues on P8 system.

-Vasant

AlexanderAmelkin, 2019-03-15:
> I don't know if we still have any issues on P8 system.

@hegdevasant, yes we do. VESNIN is P8-based and we've just fixed a part of these warnings in YADRO-KNS/op-build#16

AlexanderAmelkin, 2019-03-15:
This one still remains for us:
> ```
> Warning (reg_format): "reg" property in /ibm,opal/flash@0 has invalid length (8 bytes) (#address-cells == 0, #size-cells == 0)
> ```

stewart-ibm, 2019-03-18:
IIRC this may be one we have to live with.... @jk-ozlabs I think looked into it ages ago and IIRC we have a weird thing where we'd break compatibility if we fixed it. JK, can you remember?

hegdevasant, 2019-03-19:
Unfortunately kernel code assumes `reg` property is 8 bytes and uses to get flash size.

powernv_flash_set_driver_info(...)
 205         rc = of_property_read_u64(dev->of_node, "reg", &size);
 206         if (rc) {
207                 dev_err(dev, "couldn't get resource size information\n");
208                 return rc;
209         }

So yes, I think we have to live with this :-( 

-Vasant