open-power/skiboot #49: occ's are not initialized properly in fast-reboot path, for the disabled OCC's

pridhiviraj, 2016-11-14:
On a fast-reboot enabled system if OCC's are disabled, then next reboot is not making up disabled OCC's

Disable OCC's by doing occ reset for more than 3 times
[root@localhost ~]# opal-prd occ reset
[root@localhost ~]# opal-prd occ reset
[root@localhost ~]# opal-prd occ reset
[root@localhost ~]# opal-prd occ reset
[root@localhost ~]# opal-prd occ reset
=feeling-luckyt ~]# nvram -p ibm,skiboot --update-config experimental-fast-reset 
[root@localhost ~]# reboot
[  OK  ] Started Show Plymouth Reboot Screen.

[root@localhost ~]# dmesg | grep -i powernv
[    0.000000] Using PowerNV machine description
[    0.750966] EEH: PowerNV platform initialized
[    4.971226] powernv-rng: Registering arch random hook.
[    6.938154] powernv-cpufreq: cpufreq pstate min -43 nominal -17 max 0
[    6.954201] powernv-cpufreq: Pstate set to safe frequency
[    6.968361] powernv-cpufreq: Frequency Control disabled from OS
[    6.986228] powernv-cpufreq: PMSR = d5d5807fc8000000
[    7.003352] powernv-cpufreq: CPU Frequency could be throttled
[    7.135968] powernv_idle_driver registered
[    7.136065] nx_compress_powernv: coprocessor found on chip 0, CT 3 CI 1
[   19.451572] powernv_rng: Registered powernv hwrng.
[   19.476582] ipmi-powernv ibm,opal:ipmi: Found new BMC (man_id: 0x000000, prod_id: 0x0000, dev_id: 0x00)
[root@localhost ~]# ipmitool sdr elist | grep -i occ
OCC Active       | 08h | ok  | 210.0 | Device Disabled

After fast-reboot, still OCC's are disabled that means there is no power management, user need to do a full IPL

shilpasri, 2016-11-15:
Hi Pridhiviraj,

OCC is loaded by Hostboot in OpenPOWER sytems. After a full IPL, OCC can be reset/started for a max three times. AFAIUC, this max_occ_reset_count is maintained by HBRT ot HTMGT and this needs to be cleared during the fast-reboot path inorder to get OCC reloaded. 

Thanks and Regards,
Shilpa

pridhiviraj, 2016-11-15:
yes, we need to clear this count, and also currently in fast-reboot path, occ_pstates_init is just checking for only previous init path occ_pstates_initialized set to true.

gery-toulouse, 2018-09-04:
Could you please retest with a recent version, and close if not reproduced?