open-power/skiboot #51: NVRAM cleanups

shenki, 2016-11-22:
The prlogs look a bit strange these days. See `skiboot-5.4.0` booting for the first time on a Palmetto below.

We should create a pre-initialised NVRMA partition in the PNOR to save us doing this at boot.

The downside of that would be we don't test this code path as often. Do we have some tests that make sure skiboot's NVRAM always works?

```
    2.356072633,5] NVRAM: Size is 576 KB
[    3.032408733,3] NVRAM: Partition at offset 0x0 has incorrect 0 length
[    3.032505627,3] NVRAM: Re-initializing (size: 0x00090000)
[    3.032590744,7] NVRAM: Created 'ibm,skiboot' partition at 0x00000000 for size 0x00000100 with cksum 0xb5
[    3.032593479,7] NVRAM: Created 'common' partition at 0x00001000 for size 0x00001000 with cksum 0x0c
[    3.032595759,7] NVRAM: Created 'wwwwwwwwwwww' partition at 0x00011000 for size 0x00007f00 with cksum 0x98
[    3.032598677,7] LIBFLASH: Smart writing to 0x1961000..019f1000...
[    3.032600250,7] LIBFLASH:   reading page 0x01961000..0x01962000...[    3.033912391,7]  need erase !
[    3.033913150,7] LIBFLASH: Erasing 0x1961000..01962000...
[    3.046019920,7] LIBFLASH: Writing to 0x01961000..001962000...
[    3.056795837,7] LIBFLASH: Verifying...
[    3.058110804,7] LIBFLASH:   reading page 0x01962000..0x01963000...[    3.059415278,7]  need erase !
[    3.059416015,7] LIBFLASH: Erasing 0x1962000..01963000...
[    3.071446756,7] LIBFLASH: Writing to 0x01962000..001963000...
[    3.082213923,7] LIBFLASH: Verifying...
[    3.083533976,7] LIBFLASH:   reading page 0x01963000..0x01964000...[    3.084846885,7]  same !
[    3.084847533,7] LIBFLASH:   reading page 0x01964000..0x01965000...[    3.086164489,7]  same !
[    3.086165132,7] LIBFLASH:   reading page 0x01965000..0x01966000...[    3.087483695,7]  same !
[    3.087484334,7] LIBFLASH:   reading page 0x01966000..0x01967000...[    3.088808559,7]  same !
[    3.088809217,7] LIBFLASH:   reading page 0x01967000..0x01968000...[    3.090125179,7]  same !
[    3.090125828,7] LIBFLASH:   reading page 0x01968000..0x01969000...[    3.091441530,7]  same !
[    3.091442191,7] LIBFLASH:   reading page 0x01969000..0x0196a000...[    3.092759873,7]  same !
[    3.092760514,7] LIBFLASH:   reading page 0x0196a000..0x0196b000...[    3.094077946,7]  same !
[    3.094078587,7] LIBFLASH:   reading page 0x0196b000..0x0196c000...[    3.095399679,7]  same !
[    3.095400316,7] LIBFLASH:   reading page 0x0196c000..0x0196d000...[    3.096721227,7]  same !
[    3.096721884,7] LIBFLASH:   reading page 0x0196d000..0x0196e000...[    3.098038551,7]  same !
[    3.098039225,7] LIBFLASH:   reading page 0x0196e000..0x0196f000...[    3.099352147,7]  same !
[    3.099352803,7] LIBFLASH:   reading page 0x0196f000..0x01970000...[    3.100677126,7]  same !
[    3.100677823,7] LIBFLASH:   reading page 0x01970000..0x01971000...[    3.101992381,7]  same !
[    3.101993061,7] LIBFLASH:   reading page 0x01971000..0x01972000...[    3.103308025,7]  same !
[    3.103308751,7] LIBFLASH:   reading page 0x01972000..0x01973000...[    3.104619847,7]  need erase !
[    3.104620655,7] LIBFLASH: Erasing 0x1972000..01973000...
[    3.118555259,7] LIBFLASH: Writing to 0x01972000..001973000...
[    3.129326977,7] LIBFLASH: Verifying...
[    3.130639846,7] LIBFLASH:   reading page 0x01973000..0x01974000...[    3.131960441,7]  same !
[    3.131961125,7] LIBFLASH:   reading page 0x01974000..0x01975000...[    3.133282046,7]  same !
[    3.133282726,7] LIBFLASH:   reading page 0x01975000..0x01976000...[    3.134602066,7]  same !
[    3.134602735,7] LIBFLASH:   reading page 0x01976000..0x01977000...[    3.135929113,7]  same !
[    3.135929802,7] LIBFLASH:   reading page 0x01977000..0x01978000...[    3.137245456,7]  same !
[    3.137246127,7] LIBFLASH:   reading page 0x01978000..0x01979000...[    3.138569489,7]  same !
[    3.138570182,7] LIBFLASH:   reading page 0x01979000..0x0197a000...[    3.139889260,7]  same !
[    3.139889939,7] LIBFLASH:   reading page 0x0197a000..0x0197b000...[    3.141210302,7]  same !
[    3.141210994,7] LIBFLASH:   reading page 0x0197b000..0x0197c000...[    3.142529233,7]  same !
[    3.142529909,7] LIBFLASH:   reading page 0x0197c000..0x0197d000...[    3.143852148,7]  same !
[    3.143852830,7] LIBFLASH:   reading page 0x0197d000..0x0197e000...[    3.145173647,7]  same !
[    3.145174324,7] LIBFLASH:   reading page 0x0197e000..0x0197f000...[    3.146497744,7]  same !
[    3.146498418,7] LIBFLASH:   reading page 0x0197f000..0x01980000...[    3.147816563,7]  same !
[    3.147817233,7] LIBFLASH:   reading page 0x01980000..0x01981000...[    3.149141047,7]  same !
[    3.149141711,7] LIBFLASH:   reading page 0x01981000..0x01982000...[    3.150458250,7]  same !
[    3.150458929,7] LIBFLASH:   reading page 0x01982000..0x01983000...[    3.151775017,7]  same !
[    3.151775689,7] LIBFLASH:   reading page 0x01983000..0x01984000...[    3.153096193,7]  same !
[    3.153096871,7] LIBFLASH:   reading page 0x01984000..0x01985000...[    3.154416767,7]  same !
[    3.154417464,7] LIBFLASH:   reading page 0x01985000..0x01986000...[    3.155739500,7]  same !
[    3.155740180,7] LIBFLASH:   reading page 0x01986000..0x01987000...[    3.157056575,7]  same !
[    3.157057247,7] LIBFLASH:   reading page 0x01987000..0x01988000...[    3.158374501,7]  same !
[    3.158375191,7] LIBFLASH:   reading page 0x01988000..0x01989000...[    3.159695165,7]  same !
[    3.159695843,7] LIBFLASH:   reading page 0x01989000..0x0198a000...[    3.161016244,7]  same !
[    3.161016915,7] LIBFLASH:   reading page 0x0198a000..0x0198b000...[    3.162334426,7]  same !
[    3.162335119,7] LIBFLASH:   reading page 0x0198b000..0x0198c000...[    3.163649226,7]  same !
[    3.163649903,7] LIBFLASH:   reading page 0x0198c000..0x0198d000...[    3.164969647,7]  same !
[    3.164970335,7] LIBFLASH:   reading page 0x0198d000..0x0198e000...[    3.166286834,7]  same !
[    3.166287515,7] LIBFLASH:   reading page 0x0198e000..0x0198f000...[    3.167609847,7]  same !
[    3.167610537,7] LIBFLASH:   reading page 0x0198f000..0x01990000...[    3.168927878,7]  same !
[... 105 lines cut ...]
[    3.383021352,7] LIBFLASH:   reading page 0x019e9000..0x019ea000...[    3.384343519,7]  need write !
[    3.384344294,7] LIBFLASH: Writing to 0x019e9000..0019ea000...
[    3.395124199,7] LIBFLASH: Verifying...
[    3.396433951,7] LIBFLASH:   reading page 0x019ea000..0x019eb000...[    3.397753294,7]  need write !
[    3.397754047,7] LIBFLASH: Writing to 0x019ea000..0019eb000...
[    3.408529949,7] LIBFLASH: Verifying...
[    3.409846911,7] LIBFLASH:   reading page 0x019eb000..0x019ec000...[    3.411164337,7]  need write !
[    3.411165151,7] LIBFLASH: Writing to 0x019eb000..0019ec000...
[    3.421941022,7] LIBFLASH: Verifying...
[    3.423254212,7] LIBFLASH:   reading page 0x019ec000..0x019ed000...[    3.424577255,7]  need write !
[    3.424578057,7] LIBFLASH: Writing to 0x019ec000..0019ed000...
[    3.435353823,7] LIBFLASH: Verifying...
[    3.436668102,7] LIBFLASH:   reading page 0x019ed000..0x019ee000...[    3.437983951,7]  need write !
[    3.437984763,7] LIBFLASH: Writing to 0x019ed000..0019ee000...
[    3.448756850,7] LIBFLASH: Verifying...
[    3.450069369,7] LIBFLASH:   reading page 0x019ee000..0x019ef000...[    3.451394713,7]  need write !
[    3.451395519,7] LIBFLASH: Writing to 0x019ee000..0019ef000...
[    3.462165246,7] LIBFLASH: Verifying...
[    3.463479391,7] LIBFLASH:   reading page 0x019ef000..0x019f0000...[    3.464798597,7]  need write !
[    3.464799368,7] LIBFLASH: Writing to 0x019ef000..0019f0000...
[    3.475577535,7] LIBFLASH: Verifying...
[    3.476885948,7] LIBFLASH:   reading page 0x019f0000..0x019f1000...[    3.478202881,7]  need write !
[    3.478203646,7] LIBFLASH: Writing to 0x019f0000..0019f1000...
[    3.488981066,7] LIBFLASH: Verifying...
```

cyrilbur-ibm, 2016-11-22:
On Mon, 2016-11-21 at 17:41 -0800, Joel Stanley wrote:
> The prlogs look a bit strange these days. See skiboot-5.4.0 booting
> for the first time on a Palmetto below.
> We should create a pre-initialised NVRMA partition in the PNOR to
> save us doing this at [...]

Alternatively, we could change the loglevels of the really repetitive
ones. Keeping in mind they are already be PR_DEBUG, so they shouldn't
be appearing in production systems. After the first boot seeing all
that with PR_DEBUG on is actually a sign things have gone bad and we
probably want to see it all.

In case you are wondering, they exist because they did help in
debugging an NVRAM initializing issue.

> 2.356072633,5] NVRAM: Size is 576 KB
> [    3.032408733,3] NVRAM: Partition at offset 0x0 has incorrect 0
> length
> [    3.032505627,3] NVRAM: Re-initializing (size: 0x00090000)
> [    3.032590744,7] [...]

[snip]

shenki, 2016-11-22:
On Tue, Nov 22, 2016 at 1:57 PM, Cyril Bur <notifications@github.com> wrote:
> On Mon, 2016-11-21 at 17:41 -0800, Joel Stanley wrote:
>
> Alternatively, we could change the loglevels of the really repetitive
> ones. Keeping in mind they are already be PR_DEBUG, so they shouldn't
> be [...]

Yep, I'm not saying we should removing anything. Just fix them up now
that we have timestamp prefixes.

You're correct that they don't appear on the console. They do appear
in the skiboot log.

oohal, 2019-07-15:
Looks like e50e239b75c38efc084aec21969d7060bf6a2e29 moved all the libflash output to PR_TRACE. I've sent a patch to fix the line continuation problem, but I think we can close this.

closed 2019-07-15 by oohal