open-power/skiboot #55: OPAL Console issue on fsp platforms.

pridhiviraj, 2017-01-16:
Below test steps causes the console messages are not coming on fsp ipmi console.
1. Connect ipmi sol console
2. Power OFF/ON the system
3. IPL messages will be coming to ipmi console
4. Trigger system dump from FSP
5. Wait for system dump to finish & immediately IPL starts automatically.
6. System IPL happening & boots to host OS, but all this IPL messages are not coming into active ipmi
console which was activated at the start of test.
Issue is coming with latest skiboot code.

sammj, 2017-01-17:
Do you see any output over IPMI (eg, Petitboot, host OS login prompt), or nothing at all?

Please post kernel and skiboot logs, either from Petitboot or the host OS depending on if you have Petitboot output or not.

williamspatrick, 2017-01-17:
Are you sure that the FSP's `ipmi sol` is intended for this use?  Does the content come over the normal virtual TTY protocol just fine and just not over `ipmi sol`?  If it is only a problem with SOL it is likely a problem with the FSP and not with skiboot (or any other open-source firmware).

pridhiviraj, 2017-01-17:
@sammj yes, nothing over ipmi console, But system boots fine, In order to access the system, we need to deactivate existing console and connect again.
Okay, will attach the required logs
Thanks

pridhiviraj, 2017-01-17:
[opal_msg_log_55.txt](https://github.com/open-power/skiboot/files/710093/opal_msg_log_55.txt)

pridhiviraj, 2017-01-17:
[kernel_log_55.txt](https://github.com/open-power/skiboot/files/710094/kernel_log_55.txt)

pridhiviraj, 2017-01-17:
@williamspatrick 
i know skiboot output won,t come on FSP's ipmi sol console, But even petitboot kernel and host kernel boot messages are also not coming into ipmi sol console

williamspatrick, 2017-01-17:
I'm surprised that anything comes over the ipmi console.

Just to be clear, are you suggesting you see the kernel messages sometimes but not always on the ipmi console?

@tomjoseph83 - Is this something you are aware of the behavior from an FSP perspective?

pridhiviraj, 2017-01-17:
@williamspatrick 
To be clear i am not seeing any IPL or kernel messages all the times in-case of a system dump followed by immediate IPL. In all remaining situations Host firmware able to use FSP ipmi sol console except in the above case.

tomjoseph83, 2017-01-17:
@pridhiviraj 

I think i first step we can do to isolate the problem, is to see whether the console data is available on the vtty session.

Secondly was the IPMI SOL session was active throughout and did it disconnect.

pridhiviraj, 2017-01-17:
@tomjoseph83 
yes, ipmi sol session is active throughout.
How can i check whether the data is available on the vtty session?

Thanks

sammj, 2017-01-18:
> I'm surprised that anything comes over the ipmi console.

Unless we're talking about different things here, we should definitely be seeing output from the Petitboot kernel over IPMI. On FSP systems this is the primary interface to Petitboot.

williamspatrick, 2017-01-19:
I assumed, perhaps incorrectly, that the Vtty is the primary path like it is for PowerVM.

pridhiviraj, 2017-01-19:
This issue is not reproducible if i try manually. the testcase automated here https://github.com/open-power/op-test-framework/pull/99/commits/f0d56376410c01d8c6d1915b3f6da5a9ac0af436 which is able to reproduce. To reproduce this issue we need to run this automated system dump testcase & connect ipmi sol console in parallel in a different terminal when test starts. And i am going to raise an internal bugzilla to track this issue as i am able reproduce with official firmware release build also. So closing the issue here.

closed 2017-01-19 by pridhiviraj

pridhiviraj, 2017-01-19:
Raised an internal BugZilla ID 150765

ozbenh, 2017-01-21:
On Tue, 2017-01-17 at 07:26 -0800, Paidipeddi Pridhiviraj wrote:
> @tomjoseph83
> yes, ipmi sol session is active throughout.
> How can i check whether the data is available on the vtty session?
> Thanks

Also check in the OPAL log if you see VTY disconnection messages from
the FSP...

Cheers,
Ben.