open-power/skiboot #57: Boot not stopping at petitboot as requested

geissonator, 2017-02-19:
See issue https://github.com/openbmc/openbmc/issues/1150

OpenBMC team analyzed the IPMI data during boot and did not see OPAL firmware requesting the boot flags on a P9 Witherspoon system.  This would most likely be why the system always tries to boot the OS.

This function seems to work fine on barreleye, and I assume the code is mostly common?  We're just bringing up the host IPMI interface on P9 so lots of opportunities for bugs on OpenBMC side here.  If nothing obvious on OPAL side then we'll continue looking on OBMC side.

We're running latest op-build code from master-next as of this week.
```
	buildroot-6b9fd80
	skiboot-8c51fda
	hostboot-6782606
	linux-4.9.8-4863d7c
	petitboot-v1.4.1-fe60185
	witherspoon-xml-3fd670a
	occ-4141b5f
	hostboot-binaries-541a2bd
	capp-ucode-1bb7503
```

williamspatrick, 2017-02-20:
I suspect this should be in the petitboot repo and not here. @sammj assigned to you for investigation.

sammj, 2017-02-21:
To see if Petitboot is trying to retrieve the boot flags please
- Set `nvram --update-config "petitboot,debug?=true"`
- Reboot to Petitboot
- Run `pb-sos` and save the resulting tar file

gkeishin, 2017-02-24:
```
ubuntu@w37l:~$ sudo nvram --update-config "petitboot,debug?=true"
[sudo] password for ubuntu:
ubuntu@w37l:~$
```
**Rebooted to petitboot**

```
Exiting petitboot. Type 'exit' to return.
You may run 'pb-sos' to gather diagnostic data
/ # pb-sos
Complete, tarfile location pb-sos.tar
/ # ls -larth pb-sos.tar
-rw-r--r--    1 root     root      231.5K Feb 24 06:33 pb-sos.tar
/ #

```
- [x]  petitboot log 
[pb-sos.tar.gz](https://github.com/open-power/skiboot/files/798526/pb-sos.tar.gz)

**boot_flags --> default**

```
GET  https://xx.xx.xx.xx/org/openbmc/settings/host0
{
  "data": {
    "auto_reboot": "yes", 
    "boot_flags": "default", 
    "boot_policy": "ONETIME", 
    "network_config": "ipaddress=,prefix=,gateway=,mac=,addr_type=", 
    "power_cap": 0, 
    "power_policy": "RESTORE_LAST_STATE", 
    "restricted_mode": 0, 
    "system_state": "HOST_POWERED_ON", 
    "time_mode": "NTP", 
    "time_owner": "BMC", 
    "use_dhcp_ntp": "yes"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

sammj, 2017-02-27:
In the Petitboot logs we have:
```platform: no IPMI parameter support```

This boils down to two checks,
a) Are we running on a 'BMC' platform by checking `/proc/device-tree/bmc`, and
b) Is IPMI support present by checking to see if `/dev/ipmi0` exists.

If one of these is false Petitboot won't try to use inband IPMI to find boot overrides.
Additionally in the kernel log we have
```
[    7.953001] ipmi message handler version 39.2
[    7.953008] ipmi device interface
[    7.953064] ipmi-powernv ibm,opal:ipmi: Unable to map irq from device tree
[    7.974038] ipmi-powernv ibm,opal:ipmi: Found new BMC (man_id: 0x000000, prod_id: 0x0000, dev_id: 0x00)
```

Which sounds like IPMI might not be configured correctly. Is IPMI otherwise working, and do those two nodes exist? Sounds like it might be a problem with how the system has been set up.

williamspatrick, 2017-02-27:
@stewart-ibm - Is there someone that can help debug this from a skiboot perspective?  Every Witherspoon acts this way as best as we can tell. Seems like something is missing for that machine to make Petitboot detect IPMI path. Could be broken on Zaius also.

shenki, 2017-03-01:
I took a look at Zaius. It was running a slightly newer setup, but the result was the same:
```
open-power-zaius-v1.14-62-g22ca71d
	buildroot-2016.08-1427-g6b9fd80
	skiboot-ca4a1ee
	hostboot-c3170c0
	linux-4.10-rc7-rc7-ea56772
	petitboot-v1.4.1-335166a
	zaius-xml-82f072d
	occ-4141b5f
	hostboot-binaries-f2b74a2
	capp-ucode-1bb7503
```

```
/var/log/petitboot # head -n 2 /var/log/petitboot/pb-discover.log 
--- pb-discover ---
platform: no IPMI parameter support
```

petitboot checks to paths are present to confirm it should use IPMI:
1. `/dev/ipmi0`
1. `/proc/device-tree/bmc`

`/dev/ipmi0` is present, but `/proc/device-tree/bmc` is not. AIUI this is supposed to be added by hostboot.

vsainath, 2017-03-03:
/proc/device-tree/bmc - I think this should be created by skiboot based on HDAT SPSUBSYS structure. In P8, Hostboot used to create this. In P9, I think the understanding is that skiboot will create the dev-tree based on HDAT...

ananthmg, 2017-03-03:
Patch [posted](https://lists.ozlabs.org/pipermail/skiboot/2017-March/006574.html) for review to OPAL mailing list.

stewart-ibm, 2017-08-15:
```commit 0f28e38bbeaa11bda20b81f2247c82682d93d787
Author: Ananth N Mavinakayanahalli <ananth@linux.vnet.ibm.com>
Date:   Fri Mar 3 17:18:18 2017 +0530

    hdata: Add BMC device-tree node for P9 OpenPOWER systems
    
    On P8 Hostboot creates the node. On P9, OPAL needs to create the node. BMC
    information available in HDAT's SPINFO section.
    
    Signed-off-by: Ananth N Mavinakayanahalli <ananth@linux.vnet.ibm.com>
    Acked-by: Joel Stanley <joel@jms.id.au>
    Signed-off-by: Stewart Smith <stewart@linux.vnet.ibm.com>
```

Fixed in 0f28e38bbeaa11bda20b81f2247c82682d93d787

closed 2017-08-15 by stewart-ibm