open-power/skiboot #58: pflash support for 4K erase block on mx66l51235f

rlippert, 2017-02-22:
Attempting to clear the HBEL partition on Zaius via the BMC fails complaining about the erase block size mismatch:

root@zaius:~# pflash -f -c -P HBEL  
About to erase and set ECC bits in region 0x000a0000 to 0x000c4000
About to erase 0x000a0000..0x000c4000 !
Erasing...
[============================================      ] 88%blocklevel_erase: len (0x4096) is not erase block (0x00010000) aligned
Error 4 erasing 0x000c0000

This chip does support 4KB erase blocks but I'm not sure if the pflash support for it has been tested yet.

williamspatrick, 2017-02-23:
@shenki - FYI.

I thought the pflash code gets its information from the kernel MTD on the BMC.  There might be something incorrect in the kernel for this chip.

shenki, 2017-02-23:
@williamspatrick Our MTD device exposes the chip with a 64K block size. @legoater can explain why this is the case.

@ozbenh @stewart-ibm I thought the fix here was to make sure all the partitions are aligned?

legoater, 2017-02-23:
when the spi-nor device table was created in the linux kernel, SECT_4K was not specified and so, 64K block size is used by default for these devices. If we change this size, we will break jff2 filesystems created with older kernels. The sector size is used as the erase size in the fs. 

However, new devices should specifiy SECT_4K when defined in the spi-nor framework.

williamspatrick, 2017-02-23:
@legoater - Does this mean that upstream kernel would not accept a patch to add SECT_4K because of backwards compatibility concerns?  If we are only concerned about our kernel, I think we could add it because we would just tell people to reflash the module.  We don't have any field support concerns.

For my education, do you know if UBI also has the same problem?  I would have thought that the filesystems would have had the erase block size in their own superblock to avoid issues like this.

legoater, 2017-02-23:
@williamspatrick: yes, reflashing works. I tried that sometime ago. The jffs2 superblock is created with : ```c->sector_size = c->mtd->erasesize;``` which is the problem. 

As for UBI, a quick glance showed that ```ubi->peb_size   = ubi->mtd->erasesize; ``` is used to create the UBI device and ```peb_size``` stands for  ```physical erase size```. So we should have the  same problem.

gery-toulouse, 2018-09-04:
Should be closed as not commented for nearly 2 years. Please retest with a recent version and reopen if needed

sammj, 2018-09-04:
Sounds like this was fixed via https://github.com/open-power/zaius-openpower/issues/20

closed 2018-09-04 by sammj