open-power/skiboot #62: Host wont boot after injecting checktsop [Barreleye]

SrideviRamesh, 2017-03-07:
Driver Info

VERSION_ID="v1.99.3-42-ge8c604b" 
Build version- 20170306191603

Test result description:

OS won't boot and state in HOST_BOOTING after injecting checktsop on OS.

Expectation:

OS should auto reboot after injecting check stop through OS.

Procedure:

Enable auto_reboot as shown below
   curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"yes\"}" -X   PUT   https://XX.XX.XX.XX/org/openbmc/settings/host0/attr/auto_reboot
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
On HOST OS inject below checkstop using opal-utils as shown below
	putscom -c 0x0 0x1d01080D 0x0A00000000000000
Petitboot console logs:

 9.74178|  Hostboot Build ID: hostboot-fe5c556/hbicore.bin
  9.74179|================================================
  3.81009|ISTEP  6. 3
  4.18342|ISTEP  6. 4
  4.18396|ISTEP  6. 5
  9.40124|HWAS|PRESENT> DIMM[03]=A0A0A0A0A0A0A0A0
  9.40125|HWAS|PRESENT> Membuf[04]=CCCC000000000000
  9.40125|HWAS|PRESENT> Proc[05]=C000000000000000
  9.58177|ISTEP  6. 6
  9.71680|System shutting down with error status 0x90001C50
  9.74545|================================================
  9.74545|Error reported by hwas (0x0C00)
  9.75497|  HWAS host_gard: no masterCore found
  9.75497|  ModuleId   0x81 MOD_HOST_GARD
  9.75497|  ReasonCode 0x0c84 RC_MASTER_CORE_NULL
  9.75498|  UserData1  0 : 0x0000000000000000
  9.75498|  UserData2  0 : 0x0000000000000000
  9.75498|User Data Section 0, type UD
  9.75498|  Subsection type 0x06
  9.75499|  ComponentId errl (0x0100)
  9.75499|  CALLOUT
  9.75499|  PROCEDURE ERROR
  9.75499|  Procedure: 85
  9.75499|User Data Section 1, type UD
  9.75500|  Subsection type 0x01
  9.75500|  ComponentId errl (0x0100)
  9.75500|  STRING
  9.75500|  host_gard
  9.75501|User Data Section 2, type UD
  9.75501|  Subsection type 0x03
  9.75501|  ComponentId errl (0x0100)
  9.75501|User Data Section 3, type UD
  9.75502|  Subsection type 0x01
  9.75502|  ComponentId errl (0x0100)
  9.75502|  STRING
  9.75502|  Hostboot Build ID: hostboot-fe5c556/hbicore.bin
  9.75502|================================================

Please find dmesg, serial_port, petitboot, FFDC logs in the attachments
[20170307040841664165_MyFFDCLogs.zip](https://github.com/open-power/skiboot/files/824498/20170307040841664165_MyFFDCLogs.zip)
[dmesg_barreleye_logs.txt](https://github.com/open-power/skiboot/files/824499/dmesg_barreleye_logs.txt)
[journal_bmc_logs.txt](https://github.com/open-power/skiboot/files/824500/journal_bmc_logs.txt)
[petitboot_logs.txt](https://github.com/open-power/skiboot/files/824502/petitboot_logs.txt)

sivassrr, 2017-03-20:
Any update here. Any one taking a look. Let know if you need any info. from us.

mamatha4, 2017-03-21:
What is the error you are trying to inject? is this is the supported error you can inject on the system.

Can you please try recreating this issue with the following steps

run the following commands on Host OS
1) for i in /sys/devices/system/cpu/cpu*/cpuidle/state1/disable; do echo 1 > $i; done

2) for i in /sys/devices/system/cpu/cpu*/cpuidle/state2/disable; do echo 1 > $i; done

3) ./getscom -c 0x0  0x1d01080D

4) putscom -c 0x0 0x1d01080D 0x0A00000000000000

SrideviRamesh, 2017-03-30:
hi, I have retried the scenario with latest driver on barreleye with VERSION_ID="v1.99.3-153-g217b5ac"..
Host OS rebooted after error injection. Hence closing the issue

closed 2017-03-30 by SrideviRamesh

sivassrr, 2017-06-06:
Looks like we dont have label as observation just like openbmc-repository here. I am marking this issue as "observation"