open-power/skiboot #63: OS failed to boot after injecting L2 Directory Read UE system checkstop through OS [barreleye]

SrideviRamesh, 2017-03-13:
Issue:

OS failed to boot after injecting L2 Directory Read UE system checkstop through OS

Expectation:

After error injection power off and on HOST.
OS should boot and gard records should be created

Procedure:

1. Disable auto_reboot setting

2. Disable cpu state through OS as shown below
```
sudo sh -c 'for i in /sys/devices/system/cpu/cpu*/cpuidle/state1/disable; do echo 1 > $i; done'
sudo sh -c 'for i in /sys/devices/system/cpu/cpu*/cpuidle/state2/disable; do echo 1 > $i; done'
```

3. Inject L2 Directory Read UE System Checkstop through OS as shown below
```
   putscom -c 0x00000000 0x11012800 0x0100000000000000
```

4. Check if HOST has reached quiesced state
5. Power off & on the HOST
6. check if OS is booted and gard records are created

System Info:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-68"
VERSION_ID="v1.99.3-68-g35f91b9"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-68"
BUILD_ID="v1.99.3"

Obmc console log info:

```
~# putscom -c 0x00000000 0x11012800 0x0100000000000000
  3.80115|ISTEP  6. 3
  4.17419|ISTEP  6. 4
  4.17507|ISTEP  6. 5
  9.39173|HWAS|PRESENT> DIMM[03]=A0A0A0A0A0A0A0A0
  9.39174|HWAS|PRESENT> Membuf[04]=CCCC000000000000
  9.39174|HWAS|PRESENT> Proc[05]=C000000000000000
  9.57253|ISTEP  6. 6
 10.68340|================================================
 10.68340|Error reported by unknown (0xE500)
 10.68340|  <none>
 10.68341|  ModuleId   0x0b unknown
 10.68341|  ReasonCode 0xe504 unknown
 10.68341|  UserData1  unknown : 0x0006000000000101
 10.68342|  UserData2  unknown : 0xe89b000700000000
 10.68342|User Data Section 0, type UD
 10.68342|  Subsection type 0x04
 10.68343|  ComponentId errl (0x0100)
 10.68343|User Data Section 1, type UD
 10.68343|  Subsection type 0x06
 10.68344|  ComponentId errl (0x0100)
 10.68344|  CALLOUT
 10.68344|  HW CALLOUT
 10.68344|  Reporting CPU ID: 12
 10.68345|  Called out entity:
 10.68345|User Data Section 2, type UD
 10.68345|  Subsection type 0x33
 10.68346|  ComponentId unknown (0xe500)
 10.68346|User Data Section 3, type UD
 10.68346|  Subsection type 0x01
 10.68347|  ComponentId unknown (0xe500)
 10.68347|  STRING
 10.68347|
 10.68347|User Data Section 4, type UD
 10.68348|  Subsection type 0x15
 10.68348|  ComponentId hb-trace (0x3100)
 10.68348|User Data Section 5, type UD
 10.68349|  Subsection type 0x01
 10.68349|  ComponentId unknown (0xe500)
 10.68349|  STRING
 10.68349|
 10.68350|User Data Section 6, type UD
 10.68350|  Subsection type 0x03
 10.68350|  ComponentId errl (0x0100)
 10.68350|User Data Section 7, type UD
 10.68351|  Subsection type 0x01
 10.68351|  ComponentId errl (0x0100)
 10.68351|  STRING
 10.68352|  Hostboot Build ID: hostboot-fe5c556/hbicore.bin
 10.68352|User Data Section 8, type UD
 10.68352|  Subsection type 0x04
 10.68352|  ComponentId errl (0x0100)
 10.68353|================================================
 31.74248|System Shutting Down To Perform Reconfiguration
 31.74249|IPMI: Initiate power cycle
 31.74340|Stopping istep dispatcher
 47.37376|IPMI: shutdown complete

```
Find the attached journal, dmesg, serial port, FFDC logs
[barreleye_serial_port_logs.txt](https://github.com/open-power/skiboot/files/838138/barreleye_serial_port_logs.txt)
[20170313064435575811_MyFFDCLogs.zip](https://github.com/open-power/skiboot/files/838132/20170313064435575811_MyFFDCLogs.zip)
[barreleye_dmseg.txt](https://github.com/open-power/skiboot/files/838134/barreleye_dmseg.txt)
[barreleye_journal.txt](https://github.com/open-power/skiboot/files/838135/barreleye_journal.txt)

sivassrr, 2017-03-20:
Any update here.

SrideviRamesh, 2017-04-04:
Hi, Recreated the failure scenario with latest build on barreleye
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-201"
VERSION_ID="v1.99.3-201-ga0fc7ea"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-201"
BUILD_ID="v1.99.3"

please check latest logs
[dmesg.txt](https://github.com/open-power/skiboot/files/892737/dmesg.txt)
[barreleye_journal.txt](https://github.com/open-power/skiboot/files/892738/barreleye_journal.txt)
[20170404051242116707_MyFFDCLogs.zip](https://github.com/open-power/skiboot/files/892742/20170404051242116707_MyFFDCLogs.zip)

sivassrr, 2017-06-06:
Hi Team, any update will appreciate. Please move towards closing the issue in right manner. Thank you.

gery-toulouse, 2018-09-04:
Should be closed as not updated for nearly 18 months. Please reopen if needed.

closed 2018-09-04 by sammj