open-power/skiboot #67: GARD tool on non open-power systems output is not clear

pridhiviraj, 2017-03-23:
I ran gard tool on fsp system where the /proc/device-tree/chosen/ibm,system-flash node doesn't exist. So output should be more clear and straight forward. Even i ran under root user, still it says "are you running under root"
[root@..... gard]# ./gard list
Couldn't open '/proc/device-tree/chosen/ibm,system-flash' FDT attribute to determine which flash device to use
Is your skiboot new enough to expose the flash through the device tree?
Do you have permission? Are you root?
Can't open MTD Device. Are you root?
[root@...... gard]#

cyrilbur-ibm, 2017-03-24:
What would you like to see here? I can absolutely add a massive "You're on an FSP system".

Let the bike shedding of the message begin!

stewart-ibm, 2017-03-24:
I wonder if there's a way to get GARD information out of an FSP?

Arguably we should just have OPAL calls to deal with GARD information, distancing userspace from FFS and the gard record layout

cyrilbur-ibm, 2017-03-24:
If someone provides me with all the documentation required I'll do it.

stewart-ibm, 2017-03-24:
I'd look at the FSP MBOX documentation as a first port of call.

It's entirely possible we only get the state at boot time through HDAT though... I *think* we get enabled/disabled state there....

cyrilbur-ibm, 2017-03-24:
I've added a simple error message: https://lists.ozlabs.org/pipermail/skiboot/2017-March/006781.html

I'm not sure I know of this FSP MBOX docs...

stewart-ibm, 2017-10-02:
```
commit c524d6b022e12e3e347d308d16d2bd51acec17f4
Author: Cyril Bur <cyril.bur@au1.ibm.com>
Date:   Thu Aug 3 17:08:52 2017 +1000

    external/gard: Print an error if run on an FSP system
    
    Signed-off-by: Cyril Bur <cyril.bur@au1.ibm.com>
    Signed-off-by: Stewart Smith <stewart@linux.vnet.ibm.com>
```

`v5.8-43-gc524d6b022e1` - will hit v5.9

closed 2017-10-02 by stewart-ibm