open-power/skiboot #69: Fix GNU_HASH warnings in pflash

saqibkh, 2017-03-23:
While creating a build using yocto2.2 we found the following error:
Error: pflash-5.4.0-r0 do_package_qa: QA Issue: No GNU_HASH in the elf binary

As a work around we have added LDFLAGS to TARGET_CC_ARCH

Github openBMC issue: https://github.com/openbmc/openbmc/issues/1361

stewart-ibm, 2017-03-23:
How exactly are you building pflash? Cross compiling? What CFLAGS/LDFLAGS apart from the default? Can you paste output from a failed build, and/or provide instructions on how to replicate?

saqibkh, 2017-03-24:
The errors started showing up while updating yocto to version 2.2.
The code is not yet merged in openbmc. When the code is merged you can remove the "TARGET_CC_ARCH" from openbmc/openbmc/meta-openbmc-machines/meta-openpower/common/recipes-bsp/skiboot/pflash.bb.
Then try building an image to recreate this issue.
 
I'll update you with the actual error message and when the yocto changes are merged.

stewart-ibm, 2017-03-26:
That doesn't answer my question.

Please provide clear and explicit instructions for reproducing.

closed 2017-08-15 by stewart-ibm

gtmills, 2018-07-18:
@stewart-ibm 
Steps to reproduce: 

comment out or remove `TARGET_CC_ARCH += "${LDFLAGS}"` here:
https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/common/recipes-bsp/skiboot/pflash.bb#L8

Build OpenBMC with an openpower target (my target was Witherspoon). 

```
ERROR: pflash-v6.0.4-r0 do_package_qa: QA Issue: No GNU_HASH in the elf binary: '/tmp/gmills/build/tmp/work/armv6-openbmc-linux-gnueabi/pflash/v6.0.4-r0/packages-split/pflash/usr/lib/libflash.so.5.3.0' [ldflags]
ERROR: pflash-v6.0.4-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: pflash-v6.0.4-r0 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /tmp/gmills/build/tmp/work/armv6-openbmc-linux-gnueabi/pflash/v6.0.4-r0/temp/log.do_package_qa.9117
ERROR: Task (/esw/san5/gmills/obmc/openbmc/meta-openbmc-machines/meta-openpower/common/recipes-bsp/skiboot/pflash.bb:do_package_qa) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4767 tasks of which 4719 didn't need to be rerun and 1 failed.
```

gtmills, 2018-07-18:
We are now on Yocto 2.5. We are using skiboot version v6.0.4. 
https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/common/recipes-bsp/skiboot/skiboot.inc
Please reopen this issue.