open-power/skiboot #70: FLASH: Invalid ELF header part ROOTFS

pridhiviraj, 2017-03-27:
[   12.365257552,3] FLASH: Invalid ELF header part ROOTFS

Seeing above error message of invalid elf header for rootfs partition with latest upstream skiboot code.

Skiboot level:
[   44.260480324,5] SkiBoot 461d410 starting...

stewart-ibm, 2017-03-27:
hah! What platform?

pridhiviraj, 2017-03-27:
Not sure this issue is due to latest ffs changes or due to running of below testcase.

I have an habanero system with 1S chip and having 8 cores. Then tried making one by one all CPU's gard expect one master CPU(By using injecting HMI Core checkstop). Then when the system is booting with single core i have observed this. Later on even if i clear the gard entries still i am seeing it. Even BMC Cold reset also did not solve the issue

stewart-ibm, 2017-03-27:
I wonder if this is solved in 6a3a09c27315d858647a1c978fa7032591a8e6c5 and it's a symptom of that libflash bug.

Able to test with 6a3a09c27315d858647a1c978fa7032591a8e6c5 ? A full one commit further along :)

pridhiviraj, 2017-03-27:
Not yet tested with https://github.com/open-power/skiboot/commit/6a3a09c27315d858647a1c978fa7032591a8e6c5.

Will test and update it.

pridhiviraj, 2017-03-27:
This time with flashed with https://github.com/open-power/skiboot/commit/6a3a09c27315d858647a1c978fa7032591a8e6c5 , But we didn't get any ROOTFS part info.

 [   13.498249658,3] FLASH: Failed to get ROOTFS partition info

 cat /sys/firmware/opal/msglog | grep -i skiboot
[   43.359007972,5] SkiBoot 6a3a09c starting...

stewart-ibm, 2017-03-27:
Ahh yep, that's back to normal.

Well, where "normal" = "we should really reduce the severity of that message on systems where it's not an actual error"

closed 2017-03-27 by stewart-ibm