open-power/skiboot #72: Seeing some I2C errors on a P8 Firestone when running I2C tests

pridhiviraj, 2017-04-04:
[ 4160.021985317,3] CENTAUR 80000000: Remote FSI SCOM error, status=0x010090c4
[ 4160.022188021,3] I2C: Failed to read the STAT_REG
[ 4160.022241809,6] IPMI: dropping non severe PEL event
[ 4160.022248375,3] CENTAUR 80000000: Remote FSI SCOM error, status=0x010090c4
[ 4160.024529721,3] I2C: Failed to read the STAT_REG
[ 4160.026307113,6] IPMI: dropping non severe PEL event
[ 4160.026324344,3] I2C: Request complete with residual data req=1 done=0
[ 4160.027932714,6] IPMI: dropping non severe PEL event

Test ran from the op-test-framework I2C driver test.

gery-toulouse, 2018-09-04:
Could you please retest on a recent version (and platform) and close if not reproduced?

mikey, 2019-07-24:
Untouched forever, closing.

closed 2019-07-24 by mikey