open-power/skiboot #75: SLW: timer stuck, falling back to OPAL pollers.

pridhiviraj, 2017-05-03:
cat /sys/firmware/opal/msglog | grep -i ,3
[    8.268169672,3] PHB#0000: Base location code not found !
[   10.418389612,3] PHB#0001: Base location code not found !
[   12.088868483,3] PHB#0002: Base location code not found !
[   29.206767108,3] FLASH: Failed to get ROOTFS partition info
[   91.310974336,3] SLW: timer stuck, falling back to OPAL pollers. You will likely have slower I2C and may have experienced increased jitter.
[  165.209460486,3] I2C: Request timeout !
[  165.238525669,3] I2C: Chip 00000000 Eng. 0 Port 0--
[  165.261066287,3] I2C: Request info--
[  165.281729487,3] I2C:  start_time=00000013b5e609bd end_time=00000013b7e29c3b (duration=0000000001fc927e)
[  165.304989460,3] I2C: Register dump--
[  165.425589969,3] I2C: Request timeout !
[  165.432880682,3] I2C: Chip 00000000 Eng. 0 Port 0--
[  165.465325701,3] I2C: Request info--
[  165.490019567,3] I2C:  start_time=00000013c39e2aee end_time=00000013c4c47c1b (duration=000000000126512d)
[  166.018350096,3] I2C: Register dump--
[  166.162670522,3] I2C: Request timeout !
[  166.170084492,3] I2C: Chip 00000000 Eng. 0 Port 1--
[  166.201816590,3] I2C: Request info--
[  166.226526292,3] I2C:  start_time=00000013d185e6df end_time=00000013d39d26fa (duration=000000000217401b)
[  166.243711775,3] I2C: Register dump--
[  166.402080871,3] I2C: Request timeout !
[  166.403912696,3] I2C: Chip 00000000 Eng. 0 Port 1--
[  166.428552413,3] I2C: Request info--
[  166.460633071,3] I2C:  start_time=00000013dff90395 end_time=00000013e1e243af (duration=0000000001e9401a)
[  166.479450495,3] I2C: Register dump--
[  167.117441903,3] I2C: Request timeout !
[  167.126971107,3] I2C: Chip 00000000 Eng. 1 Port 2--
[  167.160159349,3] I2C: Request info--
[  167.176851336,3] I2C:  start_time=00000013ee606e2f end_time=00000013ef6f84bf (duration=00000000010f1690)
[  167.201653146,3] I2C: Register dump--

Seeing these SLW timer stuck and I2C request timeouts on normal IPL when console level to memory is set to 9.

Steps to reproduce:
1. Set the log-level-memory to 9
nvram -p ibm,skiboot --update-config log-level-memory=9
2. Reset the system
ipmitool power reset

During IPL we can observe those errors.
[   22.003797675,3] SLW: timer stuck, falling back to OPAL pollers. You will likely have slower I2C and may have experienced increased jitter.
[   22.003799745,7] SLW: Stuck with odd generation !
[   22.042837993,7] SLW: Register state:
[   22.054085496,7] SLW:  e0000 = 0e01000800000acc
[   22.096442516,7] SLW:  e0001 = 0000ffffffffffff
[   22.104860670,7] SLW:  e0002 = 0000000000000000
[   22.104913236,7] SLW:  e0003 = 00bff00000000000
[   22.112877545,7] SLW:  e0004 = 0000000000000000
[   22.141005314,7] SLW:  e0005 = 0000000000000000
[   22.165425951,7] SLW:  e0006 = 00000000007755c8
[   22.165478905,7] SLW:  e0007 = 000000000000000a
[   22.171263002,7] SLW:  e0008 = 0000000800000000
[   22.208497567,7] SLW:  e0009 = 0000000000000000
[   22.233927958,7] SLW:  e000a = 0000000000000000
[   22.244247119,7] SLW:  e000b = 0000000000000000
[   22.244300495,7] SLW:  e000c = 0000000000000000
[   22.255552519,7] SLW:  e000d = 0000000000000000
[   22.270298566,7] SLW:  e000e = 0000000000000000
[   22.278513725,7] SLW:  e000f = 0000000000000000
[   22.314022318,7] SLW:  e0010 = 000800000acc0000
[   22.324235034,7] SLW:  e0011 = 0000000000000000
[   22.324284438,7] SLW:  e0012 = 0000000000000000
[   22.336113103,7] SLW:  e0013 = 0000000000000000
[   22.348713049,7] SLW:  e0014 = 0000000000000904
[   22.375086273,7] SLW:  e0015 = 0000000300000000
[   22.375139631,7] SLW:  e0016 = 0000000280002000
[   22.387205465,7] SLW:  e0017 = c200510500000000
[   22.405030534,7] SLW:  e0018 = b400000300000000
[   22.426082742,7] SLW:  e0019 = 0000000000000000
[   22.445821089,7] SLW:  5001c = a800fb0100000000
[   22.445873909,7] SLW:  50038 = 0000000100000000
[   22.474706268,7] SLW:  50039 = 4004002600000000
[   22.485307813,7] SLW:  5003a = 0000000000000000
[   22.488540290,7] SLW:  5003b = 8069000000000000

Skiboot level:
SkiBoot 44790c6 starting...

pridhiviraj, 2017-05-03:
[opal_log_75_pr.txt](https://github.com/open-power/skiboot/files/973157/opal_log_75_pr.txt)

oohal, 2017-05-12:
SLW getting stuck is interesting and shouldn't be happening. I think the I2C timeouts here are only a side effect of the SLW problem since the I2C driver uses the OPAL timer API to implement the timeout detection.

gery-toulouse, 2018-09-04:
Could you please retest with a recent version and close if not reproduced?

closed 2018-09-05 by pridhiviraj