open-power/skiboot #77: console driver loglevel 5 is broken.

pridhiviraj, 2017-05-15:
42.51711|ISTEP 21. 3
[   43.001463146,5] SkiBoot 6916c2a starting...
[   43.001466225,5] initial console log level: memory 7, driver 5
[   43.001469168,6] CPU: P8 generation processor(max 8 threads/core)
[   43.001471742,7] CPU: Boot CPU PIR is 0x0068 PVR is 0x004d0200
[   43.001474588,7] CPU: Initial max PIR set to 0x1fff
[   43.001961041,5] OPAL table: 0x300d6840 .. 0x300d6d00, branch table: 0x30002000
[   43.001965830,5] Assigning physical memory map table for unused
[   43.001969452,5] FDT: Parsing fdt @0xff00000
[   43.006070254,5] XSCOM: chip 0x0 at 0x3fc0000000000 [P8 DD2.0]
[   43.006079319,6] XSTOP: XSCOM addr = 0x2010c82, FIR bit = 31
[   43.006082563,6] MFSI 0:0: Initialized
[   43.006084668,6] MFSI 0:2: Initialized
[   43.006086747,6] MFSI 0:1: Initialized
[   43.006440425,5] LPC: LPC[000]: Initialized, access via XSCOM @0xb0020
[   43.006447804,5] LPC: LPC: Default bus on chip 0x0
[   43.006564365,6] MEM: parsing reserved memory from node /ibm,hostboot/reserved-memory
[   43.006573687,7] HOMER: Init chip 0
[   43.006576531,7]   PBA BAR0 : 0x0000001ffd800000
[   43.006579227,7]   PBA MASK0: 0x0000000000300000
[   43.006581836,7]   HOMER Image at 0x1ffd800000 size 4MB
[   43.006585340,7]   PBA BAR2 : 0x4000001ffda00000
[   43.006588002,7]   PBA MASK2: 0x0000000000000000
[   43.006590396,7]   SLW Image at 0x1ffda00000 size 1MB
[   43.006593691,7]   PBA BAR3 : 0x0000001fff800000
[   43.006596200,7]   PBA MASK3: 0x0000000000700000
[   43.006598682,7]   OCC Common Area at 0x1fff800000 size 8MB
[   43.006622658,7] CPU idle state device tree init
[   43.006626086,5] SLW: HB-provided idle states property found
[   43.006630135,5] SLW: Enabling: nap
[   43.006632249,5] SLW: Enabling: fastsleep_
[   43.006634754,5] SLW: Enabling: winkle
[   43.006847499,5] PLAT: Fixing up PSI BAR on chip 0 BAR=3fffe80000001
[   43.006877027,7] AST: PNOR LPC offset: 0x0c000000
[   43.006945141,5] PLAT: Using virtual UART
[   43.007246750,7] UART: Using LPC IRQ 4
[   43.311158472,5] PLAT: Detected Habanero platform
[   43.322590643,5] PLAT: Detected BMC platform AMI
[   43.323613620,5] CENTAUR: Found centaur for chip 0x0 channel 0
[   43.334138327,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 3
[   43.350026745,5] CENTAUR: Found centaur for chip 0x0 channel 1
[   43.366102333,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 2
[   43.383896580,5] CENTAUR: Found centaur for chip 0x0 channel 4
[   43.393383788,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 7
[   43.404720509,5] CENTAUR: Found centaur for chip 0x0 channel 5
[   43.416367459,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 6
[   43.425851621,5] PSI[0x000]: Found PSI bridge [active=0]
[   43.447681608,5] CPU: All 80 processors called in...
[    2.218214723,5] FLASH: Found system flash: Macronix MXxxL51235F id:0
[    2.218358437,5] BT: Interface initialized, IO 0x00e4
[    3.167440144,5] NVRAM: Size is 576 KB
[    3.354493685,5] STB: secure mode off
[    3.354542095,5] STB: trusted mode off
[    3.354608990,5] OPAL: Using OPAL UART console
[    3.354814492,5] CAPI: Preloading ucode 200ea
[    3.364143038,5] FLASH: Queueing preload of 2/200ea
[    3.375586452,5] FLASH: Queueing preload of 0/0
[    3.392341159,5] FLASH: Queueing preload of 1/0
[    3.392357392,7] FFS: Partition map size: 0x1000
[    3.402830344,7] FLASH: CAPP partition has ECC
[    3.404365099,7] FLASH: CAPP partition isn't signed
[    3.404366411,7] FLASH: flash subpartition eyecatcher CAPP
[    3.404367589,7] FLASH: Computed CAPP partition size: 73728 (subpart 131306 size 36432 offset 4096)
[    3.439382606,5] STB: sb_verify skipped resource 2, secure_mode=0
[    3.439384269,5] STB: tb_measure skipped resource 2, trusted_mode=0
[    3.465655309,7] FFS: Partition map size: 0x1000
[    3.488395337,7] FLASH: BOOTKERNEL partition doesn't have ECC
[    3.498746743,7] FLASH: BOOTKERNEL partition isn't signed
[    3.498747971,7] FLASH: computed BOOTKERNEL size 15640424
[    4.097233036,5] Chip 0 Found PBCQ0 at /xscom@3fc0000000000/pbcq@2012000
[    4.139139162,5] Chip 0 Found PBCQ1 at /xscom@3fc0000000000/pbcq@2012400
[    4.223097454,5] Chip 0 Found PBCQ2 at /xscom@3fc0000000000/pbcq@2012800
[    4.453931311,3] PHB#0000: Base location code not found !
[    7.252082527,3] PHB#0001: Base location code not found !
[    9.089793625,3] PHB#0002: Base location code not found !
[   10.355795914,5] PCI: Resetting PHBs...
[   13.258329459,5] PCI: Probing slots...
[   13.419102110,5] STB: sb_verify skipped resource 0, secure_mode=0
[   13.419122521,7] BT: seq 0x09 netfn 0x06 cmd 0x35: IPMI MSG done
[   13.419123905,7] IPMI read event 35 complete: 16 bytes. cc: 00
[   13.419125232,6] IPMI: dropping System Event Record SEL
[   13.421756971,7] PCI: Checking slot table against detected devices
[   13.421761105,7] PCI Summary:
[   13.421769283,5] PHB#0000:00:00.0 [ROOT] 1014 03dc R:00 C:060400 B:01..01 SLOT=Slot3 
[   13.421777952,5] PHB#0000:01:00.0 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[   13.421785001,5] PHB#0000:01:00.1 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[   13.421791793,5] PHB#0000:01:00.2 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[   13.421798572,5] PHB#0000:01:00.3 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[   13.421807807,5] PHB#0001:00:00.0 [ROOT] 1014 03dc R:00 C:060400 B:01..15 SLOT=Backplane PLX 
[   13.421816188,5] PHB#0001:01:00.0 [SWUP] 10b5 8748 R:ca C:060400 B:02..15 LOC_CODE=Backplane PLX
[   13.421832784,5] PHB#0001:02:01.0 [SWDN] 10b5 8748 R:ca C:060400 B:03..07 SLOT=Network Mezz 
[   13.421845487,5] PHB#0001:02:08.0 [SWDN] 10b5 8748 R:ca C:060400 B:08..08 SLOT=Storage Mezz 
[   13.421854847,5] PHB#0001:08:00.0 [LGCY] 1b4b 9235 R:11 C:010601 (          sata) LOC_CODE=Storage Mezz
[   13.421865713,5] PHB#0001:02:09.0 [SWDN] 10b5 8748 R:ca C:060400 B:09..09 SLOT=Backplane USB 
[   13.421875637,5] PHB#0001:09:00.0 [EP  ] 104c 8241 R:02 C:0c0330 (      usb-xhci) LOC_CODE=Backplane USB
[   13.421886671,5] PHB#0001:02:0a.0 [SWDN] 10b5 8748 R:ca C:060400 B:0a..0b SLOT=Backplane BMC 
[   13.421896873,5] PHB#0001:0a:00.0 [ETOX] 1a03 1150 R:03 C:060400 B:0b..0b LOC_CODE=Backplane BMC
[   13.433540961,5] PHB#0001:0b:00.0 [PCID] 1a03 2000 R:30 C:030000 (           vga) LOC_CODE=Backplane BMC
[   13.433552110,5] PHB#0001:02:10.0 [SWDN] 10b5 8748 R:ca C:060400 B:0c..10 SLOT=Slot2 
[   13.433564759,5] PHB#0001:02:11.0 [SWDN] 10b5 8748 R:ca C:060400 B:11..15 SLOT=Slot1 
[   13.433575597,5] PHB#0002:00:00.0 [ROOT] 1014 03dc R:00 C:060400 B:01..ff SLOT=Slot4 
[   13.433582460,6] IPMI: setting fw progress sensor 05 to 07
[   14.194616142,7] BT: seq 0x0a netfn 0x04 cmd 0x30: Message sent to host
[   14.273925460,5] irq 10 name: psi:fsp (7/0)

[   43.001466225,5] initial console log level: memory 7, driver 5
so we have console log level for driver we set to 5, But i am seeing loglevel 6 and 7 also logged to the driver. But all 6 and 7 loglevels are logged to memory as expected. But for driver some INFO and DEBUG log level messages are logging(Not all messages). is it expected?

mikey, 2017-05-15:
FWIW, looks like this patch (and associated series) from @oohal attempts to fix it https://patchwork.ozlabs.org/patch/707655/

gery-toulouse, 2018-09-04:
Should be closed as not updated for nearly 18 months. Please reopen if needed.

sammj, 2018-09-04:
Looks like this fix went in

closed 2018-09-04 by sammj