open-power/skiboot #78: ECC: uncorrectable error: ffffffffffffffff ff shown even after clearing gard entries

PavamanSubramaniyam, 2017-05-16:
I have installed the Pegas1.0 nightly build on a P9 Zaius Hardware.
Then installed the upstream skiboot on the OS: git clone https://github.com/open-power/skiboot.git
Then compiled the xscom and gard utilities coming within the upstream skiboot git repostiory.

Happened to see a gard entry on the system after upgrading the OpenBMC and PNOR Firmware images from upstream and powered on the HOST.

[root@zaiuslp3 xscom-utils]# ../gard/gard list
|    ID    |   Error  | Type            |
+---------------------------------------+
| 00000001 | 90000004 | physical        |
+=======================================+

[root@zaiuslp3 xscom-utils]# ../gard/gard show 00000001
Record ID:    0x00000001
========================
Error ID:     0x90000004
Error Type:         0xe2
Res Recovery:       0xff
Path Type: physical
>System, Instance #0
 >Node, Instance #0
  >Processor, Instance #0
   >Unknown, Instance #1
    >MSC, Instance #0
     >Last, Instance #1

I wanted to clear the gard entries and used the below command.

[root@zaiuslp3 xscom-utils]# ../gard/gard clear all
Clearing the entire gard partition...
done

Then again tried to verify if the gard entries were cleared.

[root@zaiuslp3 xscom-utils]# ../gard/gard list
ECC: uncorrectable error: ffffffffffffffff ff
The data at the GUARD partition does not appear to be valid gard data
Clear the entire GUARD partition? [y/N]
y
No GARD entries to display

But there is a ECC: uncorrectable error: ffffffffffffffff ff message getting displayed.

PavamanSubramaniyam, 2017-05-16:
The machine is installed with the below mentioned OpenBMC and PNOR Firmware levels:

root@zaius:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%open-power-zaius-v1.16-66-g69aaa76
        buildroot-2017.02.2-7-g23118ce
        skiboot-f0ebbc8
        hostboot-43ccc17
        linux-4.11-openpower1-p5456eff
        petitboot-v1.4.2-p2462045
        zaius-xml-6bcffcf
        occ-456d9a4
        hostboot-binaries-d8f4ab7
        capp-ucode-9c73e9f
        sbe-d770027
[==================================================] 100%

root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-40"
VERSION_ID="v1.99.5-40-gbaf766c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-40"
BUILD_ID="v1.99.5"

stewart-ibm, 2017-10-02:
@cyrilbur-ibm any thoughts here?

cyrilbur-ibm, 2017-10-03:
Firstly, in the original post I'd like to clarify something.
The steps performed were:
1. `../gard/gard list`
```
| ID | Error | Type |
+---------------------------------------+
| 00000001 | 90000004 | physical |
+=======================================+
```
2. `../gard/gard show 00000001`
```
Record ID: 0x00000001

Error ID: 0x90000004
Error Type: 0xe2
Res Recovery: 0xff
Path Type: physical

>System, Instance #0
>Node, Instance #0
>Processor, Instance #0
>Unknown, Instance #1
>>MSC, Instance #0
>>Last, Instance #1
```
3. `../gard/gard clear all`
```
Clearing the entire gard partition...
done
```
4. `../gard/gard list`
```
ECC: uncorrectable error: ffffffffffffffff ff
The data at the GUARD partition does not appear to be valid gard data
Clear the entire GUARD partition? [y/N]
y
No GARD entries to display
```

Is this correct?

---

What happens if step 5 was `../gard/gard list`?
What is the output of `../gard/gard --help`?

oohal, 2019-07-15:
I've done a bunch of work with `opal-guard` since this was filed and never seen this particular issue. Considering it seems to be erasing the partition without writing ECC i'd say this was a libflash bug that's since been fixed. Closing.

closed 2019-07-15 by oohal