open-power/skiboot #79: gard tool is not working in OpenBMC

mine260309, 2017-05-16:
Build gard tool for OpenBMC, but it's not working on both P8 and P9.

On P8 Palmetto:
```
# ./opal-gard list
No GARD entries to display
*** Error in `./opal-gard-p8': munmap_chunk(): invalid pointer: 0x015e9074 ***
Aborted (core dumped)
```

On P9 Romulus:
```
# ./opal-gard list
LIBFLASH: Flash identification failed
Failed to open flash chip
Can't open MTD Device. Are you root?
```
Enabling `FL_DBG` shows `LIBFLASH: Flash ID: 50.41.52 (504152)`, which dose not match the flash device ids.

mine260309, 2017-05-16:
The tool is using `/dev/mem` to access the flash, which seems broken.

Following `pflash`, if changing the code to access the flash via MTD, then everything works fine:
```
diff --git a/external/gard/gard.c b/external/gard/gard.c
index 4b26a3b..7015062 100644
--- a/external/gard/gard.c
+++ b/external/gard/gard.c
@@ -638,6 +638,10 @@ int main(int argc, char **argv)
        argv += optind;
        action = argv[0];

+    if (arch_flash_access(NULL, PNOR_MTD) == ACCESS_INVAL) {
+        fprintf(stderr, "Can't access PNOR flash through MTD on this system\n");
+        exit(1);
+    }
        if (arch_flash_init(&(ctx->bl), filename, true)) {
                /* Can fail for a few ways, most likely couldn't open MTD device */
                fprintf(stderr, "Can't open %s\n", filename ? filename : "MTD Device. Are you root?");
```

With the above change, it works on both P8 and P9.

stewart-ibm, 2017-05-17:
Lei YU <notifications@github.com> writes:
> The tool is using `/dev/mem` to access the flash, which seems broken.
>
> Following `pflash`, if changing the code to access the flash via MTD, then everything works fine:
> ```
> diff --git [...]

Can you please send a patch to the skiboot list?

I suspect we may want to solve it slightly differently, but I'm not
entirely sure off the top of my head what order we try to access flash in.

-- 
Stewart Smith
OPAL Architect, IBM.

mine260309, 2017-05-18:
I'm not convinced by the change, since this tool can be used on both host and BMC, I'm not sure if it will break the functionality on host.

And it can be workarounded by specifying the mtd device by `-f` option.

So on OpenBMC `opal-gard -f /dev/mtd6 list` works just fine.

closed 2017-05-18 by mine260309