open-power/skiboot #80: Concurrent read of PNOR flash chip fails for second read operation.

pridhiviraj, 2017-05-17:
Parallel read of PNOR pflash chip fails for second read.

1st read:
# pflash -r /tmp/pnor
Reading to "/tmp/pnor" from 0x00000000..0x04000000 !
[================                                  ] 31% ETA:20s     [  397.107078] mtd mtd0: opal_flash_async_op(op=0) failed (rc -2)
[==================================================] 100% ETA:0s     

2nd read:
# pflash -r /tmp/pnor1
Reading to "/tmp/pnor1" from 0x00000000..0x04000000 !
[                                                  ] 0%Flash read error 15 for chunk at 0x00000000

So first read of PNOR is success but second read of PNOR fails(Which is run in parallel to first one).

Skiboot level:
 cat /sys/firmware/opal/msglog | grep -i skiboot
[   43.376158626,5] SkiBoot 30df6f0 starting...
pflash -v
Open-Power Flash tool pflash-5.5.0-43-g44790c6

Seen the failure with latest pflash level as well.
pflash  -v
Open-Power Flash tool pflash-5.6.0-rc1

mikey, 2017-05-17:
This on P8 or P9?

oohal, 2017-05-31:
the code printing that error message is at drivers/mtd/devices/powernv_flash.c:85
```
        if (rc != OPAL_ASYNC_COMPLETION) {
                dev_err(dev, "opal_flash_async_op(op=%d) failed (rc %d)\n",
                                op, rc);
                opal_async_release_token(token);
                return -EIO;
        }
```

At a guess that should be translating OPAL_BUSY into EAGAIN rather than EIO.

cyrilbur-ibm, 2017-05-31:
It appears to reproduce doing:
`dd if=/dev/mtd0 of=/tmp/flash`
the second one fails.

I also see `mtd mtd0: opal_flash_async_op(op=0) failed (rc -2)` spat out on my terminal.

Its probably in the powernv_flash driver. I'll have a look.

pridhiviraj, 2018-02-02:
Any updates on this bug?

hegdevasant, 2020-11-18:
I believe this is addressed in upstream via commit 6f469b67f (mtd: powernv_flash: Use opal_async_wait_response_interruptible()).

-Vasant