open-power/skiboot #81: OS failed to shutdown - hang's in the shutdown path

gkeishin, 2017-05-24:
**Driver Info:**
```
- [x] BUILD ID   : https://openpower.xyz/job/openbmc-build/755/
- [x] VERSION_ID : v1.99.5-74-g60189dc
- [x] Kernel     : Linux witherspoon 4.7.10-fb92f38f80699b2089c3467b1344ac107f797d3c #1 Wed May 24 09:21:11 UTC 2017 armv6l GNU/Linux

PNOR Version:
IBM-witherspoon-ibm-OP9_v1.16_1.24
        op-build-v1.16-83-ge67b53c-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-rc2
        hostboot-d02273f
        linux-4.11.1-openpower1-pc11ec03
        petitboot-v1.4.2-pcb6424c
        machine-xml-e536539
        occ-8a7df03
        hostboot-binaries-d8f4ab7
        capp-ucode-9c73e9f
        sbe-d770027

```

```
reboot: Power down
[  116.172830672,5] OPAL: Shutdown request type 0x0...
[  117.393481232,5] IPMI: Soft shutdown requested
NMI watchdog: BUG: soft lockup - CPU#7 stuck for 23s! [kworker/7:1:1750]
INFO: rcu_sched self-detected stall on CPU
        7-...: (2101 ticks this GP) idle=16b/140000000000001/0 softirq=878/878 fqs=1048
         (t=2101 jiffies g=1819 c=1818 q=134)
INFO: rcu_sched detected stalls on CPUs/tasks:
        0-...: (1 GPs behind) idle=c83/140000000000000/0 softirq=2088/2088 fqs=3343
```

- [x] OS SOL log  
[OS_Console_shutdown.txt](https://github.com/open-power/skiboot/files/1025520/OS_Console_shutdown.txt)

geissonator, 2017-05-24:
I see a similar fail on w37 when doing the soft power off command when the OS is at the ubuntu login prompt:
```
[  OK  ] Stopped Apply Kernel Variables.
[  OK  ] Stopped Load Kernel Modules.
[  OK  ] Reached target Shutdown.
[  660.171504] INFO: rcu_sched detected stalls on CPUs/tasks:
[  660.171581] 	0-...: (1 ticks this GP) idle=329/140000000000000/0 softirq=1184/1184 fqs=65461 
[  660.171725] 	(detected by 19, t=131292 jiffies, g=4917, c=4916, q=70)
[  683.816288] ata1.00: exception Emask 0x0 SAct 0x80 SErr 0x0 action 0x6 frozen
[  683.816389] ata1.00: failed command: READ FPDMA QUEUED
[  683.816482] ata1.00: cmd 60/08:38:00:41:81/00:00:11:00:00/40 tag 7 ncq dma 4096 in
[  683.816482]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
[  683.816706] ata1.00: status: { DRDY }
[  693.817032] ata1: softreset failed (1st FIS failed)
[  703.817393] ata1: softreset failed (1st FIS failed)
[  723.193505] INFO: rcu_sched detected stalls on CPUs/tasks:
[  723.193580] 	0-...: (1 ticks this GP) idle=329/140000000000000/0 softirq=1184/1184 fqs=73339 
[  723.193714] 	(detected by 12, t=147047 jiffies, g=4917, c=4916, q=70)
[  738.818025] ata1: softreset failed (1st FIS failed)
[  743.818705] ata1: softreset failed (1st FIS failed)
[  743.818763] ata1: reset failed, giving up
[  743.818869] blk_update_request: I/O error, dev sda, sector 293683456
[  743.819034] EXT4-fs error (device sda2): ext4_find_entry:1463: inode #9175041: comm plymouthd: reading directory lblock 0
[  743.819233] blk_update_request: I/O error, dev sda, sector 16384
[  743.819346] blk_update_request: I/O error, dev sda, sector 16384
[  743.819455] Buffer I/O error on dev sda2, logical block 0, lost sync page write
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped Forward Password Requests to Plymouth Directory Watch.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed UUID daemon activation socket.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Stopped target System Initialization.
         Stopping Update UTMP about System Boot/Shutdown...
[  OK  ] Stopped target Swap.
         Deactivating swap /swapfile...
[  OK  ] Stopped target Encrypted Volumes.
[  OK  ] Stopped Forward Password Requests to Wall Directory Watch.
         Stopping Network Time Synchronization...
         Stopping Load/Save Random Seed...
[  OK  ] Closed Syslog Socket.
[  OK  ] Deactivated swap /swapfile.
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped ifup for enP5p1s0f0.
[  OK  ] Stopped Raise network interfaces.
[  OK  ] Stopped target Local File Systems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Stopped target Network (Pre).
[  OK  ] Stopped Apply Kernel Variables.
[  OK  ] Stopped Load Kernel Modules.
[  OK  ] Reached target Shutdown.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped Forward Password Requests to Plymouth Directory Watch.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Sockets.
...
Host console seems to just keep looping on the above output
```

geissonator, 2017-05-24:
This one was my fault.  BMC issued soft power off to Host, Host shuts down and issued power off to BMC, BMC would then send SMS up to host and tell it to power off again...rinse and repeat.  Not sure if we need a check on the host side to just ignore the attention once shutdown but I need to change the bmc code to not re-send it.

sivassrr, 2017-05-30:
@gkeishin do we need to move it back to openbmc? if equivalent issue not there.

geissonator, 2017-07-05:
I hit another flavor of this in our testing today with the following on wsbmc017:
```
Reading to "/dev/stdout" from 0x02820000..0x02821000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.17_1.46
	op-build-v1.17-101-g1c57f18-dirty
	buildroot-2017.02.2-7-g23118ce
	skiboot-5.6.0-158-ga1e0a047b2a0
	hostboot-0ffbb55
	linux-4.11.6-openpower1-p57dd16e
	petitboot-v1.4.3-pfac92ff
	machine-xml-5b59a1d
	occ-1dc97a6
	hostboot-binaries-711147e
	capp-ucode-9c73e9f
	sbe-0df81aa
```

The system was booted up to the OS, here was the SOL output:
```
[24;10HUbuntu          [24;16H
 Loaded kernel image from file:///var/petitboot/mnt/dev/sda2/boot/vmlinux-4.10.[?7l0[?7h[24;9Hinitrd from file:///var/petitboot/mnt/dev/sda2/boot/initrd.img-4.10.0-1[?7l4[?7h
 Running boot hooks[K
 Performing kexec loadreboot[   47.503357088,5] IPMI: Soft shutdown requested

The system is going down NOW!

Sent SIGTERM to all processes

Sent SIGKILL to all processes
[   33.771463] kexec_core: Starting new kernel
[   51.400289776,5] OPAL: Switch to big-endian OS
[   51.501274960,5] OPAL: Switch to little-endian OS
 -> smp_release_cpus()
spinning_secondaries = 15
 <- smp_release_cpus()
[   52.401987008,3] XIVE[ IC 00  ] ISN 1 lead to invalid IVE !
[   52.402123200,3] XIVE[ IC 00  ] ISN 1 lead to invalid IVE !
[   52.402233296,3] XIVE[ IC 00  ] ISN 1 lead to invalid IVE !
[   52.403588368,3] XIVE[ IC 00  ] ISN 1 lead to invalid IVE !
[   52.403701376,3] XIVE[ IC 00  ] ISN 1 lead to invalid IVE !
[   52.403807376,3] XIVE[ IC 00  ] ISN 1 lead to invalid IVE !
[    2.425345] Unable to open file: /etc/keys/x509_ima.der (-2)
[    2.425350] Unable to open file: /etc/keys/x509_evm.der (-2)
[40m[2J[-1;-1f[?25l[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[37m.  [37m.  [37m.  [37m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[33m.  [37m.  [37m.  [37m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[33m.  [33m.  [37m.  [37m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[33m.  [33m.  [33m.  [37m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[33m.  [33m.  [33m.  [33m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[37m.  [33m.  [33m.  [33m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[37m.  [37m.  [33m.  [33m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[37m.  [37m.  [37m.  [33m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[37m.  [37m.  [37m.  [37m.[-1;-1f[40m[37mUbuntu 17.04[-1;-1f[33m.  [37m.  [37m.  [37m.[49m[2J[-1;-1f[?25h
Ubuntu Zesty Zapus (development branch) ws017p1 hvc0
ws017p1 login: 
```

There was a lot of unusual traffic sent over host IPMI to the BMC during the shutdown:
```
Jul 05 18:00:20 witherspoon systemd[1]: Starting Soft power off of the host...
Jul 05 18:00:26 witherspoon phosphor-ledmanager[944]: De-Asserting LEDs
Jul 05 18:00:26 witherspoon phosphor-ledmanager[944]: No service providers for physical LED
Jul 05 18:00:26 witherspoon systemd[1]: Started Assert power_on LED.
Jul 05 18:00:26 witherspoon ipmid[1040]: Failed to call the method: Input/output error
Jul 05 18:00:26 witherspoon ipmid[1040]: Dest: :1.11, Path: /org/openbmc/HostIpmi/1
Jul 05 18:00:26 witherspoon ipmid[1040]: Pushing cmd on to queue
Jul 05 18:00:26 witherspoon ipmid[1040]: Asserting SMS Attention
Jul 05 18:00:26 witherspoon ipmid[1040]: SMS Attention asserted
Jul 05 18:00:26 witherspoon ipmid[1040]: sscanf Failed in extracting mac address.
Jul 05 18:00:26 witherspoon ipmid[1040]: getHostNetworkData failed for get_sys_boot_options.
Jul 05 18:00:26 witherspoon ipmid[1040]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Jul 05 18:00:26 witherspoon phosphor-softpoweroff[1488]: Host control signal values
Jul 05 18:00:26 witherspoon phosphor-softpoweroff[1488]: Timer started waiting for host to shutdown
Jul 05 18:00:41 witherspoon ipmid[1040]: WATCHDOG RESET
Jul 05 18:00:41 witherspoon ipmid[1040]: IPMI SET_SENSOR [0x03]
Jul 05 18:00:41 witherspoon ipmid[1040]: WATCHDOG RESET
Jul 05 18:00:41 witherspoon ipmid[1040]: IPMI DCMI_GET_POWER_LEVEL
Jul 05 18:00:41 witherspoon ipmid[1040]: IPMI GET_SENSOR_READING [0x09]
Jul 05 18:00:41 witherspoon ipmid[1040]: IPMI SET_SENSOR [0x08]
[... 73 lines cut ...]
Jul 05 18:00:42 witherspoon ipmid[1040]: IPMI Handling GET-SEL-TIME
Jul 05 18:00:42 witherspoon ipmid[1040]: Host Time read:[Wed Jul  5 18:00:31 2017 GMT] :: [1499277631]
Jul 05 18:00:42 witherspoon ipmid[1040]: IPMI Handling GET-SEL-TIME
Jul 05 18:00:42 witherspoon ipmid[1040]: Host Time read:[Wed Jul  5 18:00:33 2017 GMT] :: [1499277633]
Jul 05 18:00:42 witherspoon ipmid[1040]: IPMI Handling GET-SEL-TIME
Jul 05 18:00:42 witherspoon ipmid[1040]: Host Time read:[Wed Jul  5 18:00:33 2017 GMT] :: [1499277633]
Jul 05 18:00:42 witherspoon hwmon.py[1012]: Removing: /sys/class/hwmon/hwmon5/
Jul 05 18:00:42 witherspoon hwmon.py[1012]: Removing: /sys/class/hwmon/hwmon8/
Jul 05 18:00:42 witherspoon hwmon.py[1012]: Removing: /sys/class/hwmon/hwmon6/
Jul 05 18:00:42 witherspoon hwmon.py[1012]: Removing: /sys/class/hwmon/hwmon7/
Jul 05 18:01:13 witherspoon ipmid[1040]: IPMI GET DEVICEERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
Jul 05 18:03:20 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.94:54024).
Jul 05 18:03:22 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.94:54026).
Jul 05 18:03:24 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.94:54028).
Jul 05 18:03:25 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.94:54030).
Jul 05 18:03:30 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.94:54050).
[... cut ...]
Jul 05 18:05:27 witherspoon phosphor-softpoweroff[1488]: Timer expired
Jul 05 18:05:27 witherspoon phosphor-softpoweroff[1488]: Host did not shutdown within configured time.
Jul 05 18:05:27 witherspoon phosphor-log-manager[945]: Failed to find metadata
Jul 05 18:05:27 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Main process exited, code=exited, status=255/n/a
Jul 05 18:05:27 witherspoon systemd[1]: Failed to start Soft power off of the host.
```

The 5 minute shutdown timer expired and the bmc just forcefully shut down power.

geissonator, 2017-07-05:
Those last error messages look related to https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1656908

geissonator, 2017-07-06:
Another flavor seen today:
```
reboot
The system is going down NOW!

Sent SIGTERM to all processes
[   50.001946624,5] IPMI: Soft shutdown requested

Sent SIGKILL to all processes
[   34.683853] kexec_core: Starting new kernel
[   52.466085328,5] OPAL: Switch to big-endian OS
[   53.037871536,5] OPAL: Switch to little-endian OS
 -> smp_release_cpus()
spinning_secondaries = 15
 <- smp_release_cpus()
[    0.633015] Severe Machine check interrupt [[Not recovered]
[    0.633076]   I[   54.176099488,0] OPAL: Reboot requested due to Platform error.
[   54.176147712,3] OPAL: Reboot requested due to Platform error.nitiator: CPU
[    0.633130]   Error type: UE [Load/Store]
[    0.633173]     Effective address: 0000000003186b01
[    0.633451] opal: Reboot type 1 not supported
[    0.633534] Kernel panic - not syncing: PowerNV Unrecovered Machine Check
[    0.633654] CPU: 1 PID: 1 Comm: swapper/0 Tainted: G   M            [   54.279841024,4] IPMI: Dropped eSEL: BMC code is buggy/missing
4.10.0-26-generic #30-Ubuntu
[    0.633812] Call Trace:
[    2.044141] ---[ end Kernel panic - not syncing: PowerNV Unrecov
```

geissonator, 2017-07-10:
Another example of us failing to shutdown.  You can see the "31.054979616,5] IPMI: Soft shutdown requested" come in during boot but we seem to just keep going.

```
0.250870] EEH: PCI Enhanced I/O Error Handling Enabled
[    0.263609] vgaarb: loaded
[    0.264244] SCSI subsystem initialized
[    0.264528] usbcore: registered new interface driver usbfs
[    0.264662] usbcore: registered new interface driver hub
[    0.265393] usbcore: registered new device driver usb
[    0.265550] pps_core: LinuxPPS API ver. 1 registered
[    0.265651] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.265823] PTP clock support registered
[    0.266572] clocksource: Switched to clocksource timebase
[    0.272097] NET: Registered protocol family 2
[    0.272608] TCP established hash table entries: 524288 (order: 6, 4194304 bytes)
[    0.277276] TCP bind hash table entries: 65536 (order: 4, 1048576 bytes)
[    0.277821] TCP: Hash tables configured (established 524288 bind 65536)
[    0.278074] UDP hash table entries: 32768 (order: 4, 1048576 bytes)
[    0.278826] UDP-Lite hash table entries: 32768 (order: 4, 1048576 bytes)
[    0.279993] NET: Registered protocol family 1
[    6.612797] Initialise system trusted keyrings
[    6.613005] workingset: timestamp_bits=54 max_order=20 bucket_order=0
[    6.621577] Key type asymmetric registered
[    6.621652] Asymmetric key parser 'x509' registered
[    6.621769] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 250)
[    6.621924] io scheduler noop registered
[    6.622788] io scheduler cfq registered (default)
[    6.622874] io scheduler mq-deadline registered
[    6.624375] ipmi message handler version 39.2
[    6.624461] ipmi device interface
[    6.624630] ipmi-powernv ibm,opal:ipmi: Unable to map irq from device tree
[   11.695848] ipmi-powernv ibm,opal:ipmi: Found new BMC (man_id: 0x000000, prod_id: 0x0000, dev_id: 0x00)
[   11.720948] hvc0: raw protocol on /ibm,opal/consoles/serial@0 (boot console)
[   11.721068] hvc0: No interrupts property, using OPAL event
[   11.721695] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[   11.722557] [drm] radeon kernel modesetting enabled.
[   11.732708] brd: module loaded
[   11.743281] loop: module loaded
[   11.743720] ahci 0004:03:00.0: Using 64-bit DMA iommu bypass
[   11.753939] ahci 0004:03:00.0: AHCI 0001.0000 32 slots 4 ports 6 Gbps 0xf impl SATA mode
[   11.754073] ahci 0004:03:00.0: flags: 64bit ncq sntf led only pmp fbs pio slum part sxs 
[   11.755224] scsi host0: ahci
[   11.755611] scsi host1: ahci
[   11.755995] scsi host2: ahci
[   11.756395] scsi host3: ahci
[   11.756592] ata1: SATA max UDMA/133 abar m2048@0x600c200010000 port 0x600c200010100 irq 88
[   11.756760] ata2: SATA max UDMA/133 abar m2048@0x600c200010000 port 0x600c200010180 irq 88
[   11.756911] ata3: SATA max UDMA/133 abar m2048@0x600c200010000 port 0x600c200010200 irq 88
[   11.757078] ata4: SATA max UDMA/133 abar m2048@0x600c200010000 port 0x600c200010280 irq 88
[   11.757422] libphy: Fixed MDIO Bus: probed
[   11.757545] tg3.c:v3.137 (May 11, 2014)
[   11.757644] tg3 0005:01:00.0: enabling device (0140 -> 0142)
[   11.792840] tg3 0005:01:00.0: Using 64-bit DMA iommu bypass
[   11.793548] tg3 0005:01:00.0 eth0: Tigon3 [partno(BCM95719) rev 5719001] (PCI Express) MAC address 70:e2:84:14:24:21
[   11.793749] tg3 0005:01:00.0 eth0: attached PHY is 5719C (10/100/1000Base-T Ethernet) (WireSpeed[1], EEE[1])
[   11.793951] tg3 0005:01:00.0 eth0: RXcsums[1] LinkChgREG[0] MIirq[0] ASF[1] TSOcap[1]
[   11.794116] tg3 0005:01:00.0 eth0: dma_rwctrl[00000000] dma_mask[64-bit]
[   11.794357] tg3 0005:01:00.1: enabling device (0140 -> 0142)
[   11.833584] tg3 0005:01:00.1: Using 64-bit DMA iommu bypass
[   11.834248] tg3 0005:01:00.1 eth1: Tigon3 [partno(BCM95719) rev 5719001] (PCI Express) MAC address 70:e2:84:14:24:22
[   11.834443] tg3 0005:01:00.1 eth1: attached PHY is 5719C (10/100/1000Base-T Ethernet) (WireSpeed[1], EEE[1])
[   11.834644] tg3 0005:01:00.1 eth1: RXcsums[1] LinkChgREG[0] MIirq[0] ASF[1] TSOcap[1]
[   11.834811] tg3 0005:01:00.1 eth1: dma_rwctrl[00000000] dma_mask[64-bit]
[   11.835232] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[   11.835350] ehci-pci: EHCI PCI platform driver
[   11.835493] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[   11.835599] ohci-pci: OHCI PCI platform driver
[   11.835844] usbcore: registered new interface driver usb-storage
[   12.087878] ata4: SATA link down (SStatus 0 SControl 300)
[   12.088040] ata3: SATA link down (SStatus 0 SControl 300)
[... 126 lines cut ...]
numa:     NODE_DATA(250) on node 8
numa: Initmem setup node 251
numa:   NODE_DATA [mem 0x2007ff71c600-0x2007ff7262ff]
numa:     NODE_DATA(251) on node 8
numa: Initmem setup node 252
numa:   NODE_DATA [mem 0x2007ff712900-0x2007ff71c5ff]
numa:     NODE_DATA(252) on node 8
numa: Initmem setup node 253
numa:   NODE_DATA [mem 0x2007ff708c00-0x2007ff7128ff]
numa:     NODE_DATA(253) on node 8
numa: Initmem setup node 254
numa:   NODE_DATA [mem 0x2007ff6fef00-0x2007ff708bff]
numa:     NODE_DATA(254) on node 8
numa: Initmem setup node 255
numa:   NODE_DATA [mem 0x2007ff6f5200-0x2007ff6feeff]
numa:     NODE_DATA(255) on node 8
Initializing IODA2 PHB (/pciex@600c3c0000000)
...
[[0;32m  OK  [0m] Started Stop ureadahead data collection 45s after completed startup.
[[0;32m  OK  [0m] Started Update UTMP about System Runlevel Changes.

Ubuntu Zesty Zapus (development branch) 21762 hvc0

21762 login: 
```

jk-ozlabs, 2017-07-14:
OK, here's what I'm confused about:

```
root@spoon2:~# ipmitool power soft
Set Chassis Power Control to Soft failed: Invalid command
```

[This sends a chassis control message, with command = 0x5, which appears to not be supported]

@jms tells me that (currently), soft off is actually implemented by using hard power off (IPMI chassis control, with command = 0x0)

However, that's the message that the host firmware uses at the *end* of the graceful power down sequence, to tell the BMC to actually (and immediately) cut power to the host. So, with that difference in behaviour, correct soft power off behaviour would be impossible to implement.

How does the BMC distinguish between those two messages?

Can we instead initiate a soft power off with the IPMI Chassis Control, with command = 0x5? Anything else is going to cause problems.

geissonator, 2017-07-14:
The ipmi path for these different power off's is still being worked.  To recreate this problem, just issue a "obmcutil poweroff" to cause the soft off path on the BMC.

jk-ozlabs, 2017-07-17:
Right, but my point is: the actual power off (once the OS has been shut down) is implemented as a hard power off (Chassis power control, command = 0x0). If that doesn't work as specified (ie, cutting the power immediately), then the soft off will not work.

jk-ozlabs, 2017-07-18:
So it seems to be working okay here; I've given it a few retries (initiating the soft off via obmcutil, at various stages once OPAL/petitboot/Ubuntu has started), and I've consistently seen the following IPMI traffic:

```
method call time=1500349474.223211 sender=:1.72 -> destination=:1.10 serial=3070 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=setAttention
signal time=1500349474.230826 sender=:1.10 -> destination=(null destination) serial=2206 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 47
   byte 6
   byte 0
   byte 49
   array [
   ]
method call time=1500349474.870111 sender=:1.72 -> destination=:1.10 serial=3072 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 47
   byte 7
   byte 0
   byte 49
   byte 0
   array of bytes [
      02
   ]
signal time=1500349474.879971 sender=:1.10 -> destination=(null destination) serial=2208 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 48
   byte 6
   byte 0
   byte 53
   array [
   ]
method call time=1500349474.898242 sender=:1.72 -> destination=:1.10 serial=3074 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 48
   byte 7
   byte 0
   byte 53
   byte 0
   array of bytes [
      55 55 c0 41 a7 00 00 00 00 00 3a 04 00 ff ff ff
   ]
signal time=1500349479.778618 sender=:1.10 -> destination=(null destination) serial=2210 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 49
   byte 0
   byte 0
   byte 2
   array of bytes "" + \0
method call time=1500349480.105049 sender=:1.72 -> destination=:1.10 serial=3076 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 49
   byte 1
   byte 0
   byte 2
   byte 0
   array [
   ]
```

\- showing the SMS attention, the Get Message Flags, Read Event, and the subsequent Chassis Power Control (command = 0x0).

@geissonator : do you have any hints on how I might be able to reproduce this? Do you have any specific points during boot where this might be more likely to be occuring?

Also, are you able to recreate while capturing dbus IPMI traffic?

[I used the following command to do that:

```
dbus-monitor --system path=/org/openbmc/HostIpmi/1 | tee /tmp/dbus-ipmi.log
```
]

geissonator, 2017-07-19:
@gkeishin - any chance you could help us out with a recreate?   If you go look at our internal runs, any job over 8 minutes is one that hit this issue.  For example, one I picked at random has this.

You can see the soft off request at "[   33.474903952,5] IPMI: Soft shutdown requested"

```
[   14.533995] powernv-cpufreq: cpufreq pstate min 60 nominal 57 max 0
[   14.534137] powernv-cpufreq: Workload Optimized Frequency is enabled in the platform
[   14.536480] usbcore: registered new interface driver usbhid
[   14.536565] usbhid: USB HID core driver
[   14.537008] ipip: IPv4 and MPLS over IPv4 tunneling driver
[   14.537397] NET: Registered protocol family 17
[   14.537501] Key type dns_resolver registered
[   14.537935] registered taskstats version 1
[   14.538021] Loading compiled-in X.509 certificates
[   14.548566] Loaded X.509 cert 'Build time autogenerated kernel key: 08339e0962c807d7c679a6046cefc474da6b7c12'
[   14.565386] Key type encrypted registered
[   14.565484] ima: No TPM chip found, activating TPM-bypass! (rc=-19)
[   14.565687] evm: HMAC attrs: 0x1
[   33.474903952,5] IPMI: Soft shutdown requested
[   14.692610] opal-power: Poweroff requested
[   14.704267] rtc-opal opal-rtc: setting system clock to 2017-04-20 17:39:01 UTC (1492709941)
[   14.707915] Freeing unused kernel memory: 11328K
[   14.708029] This architecture does not have kernel memory protection.
[   14.751817] udevd[1641]: starting version 3.2.2
[   14.759690] udevd[1642]: starting eudev-3.2.2
[   14.833890] tg3 0005:01:00.0 enP5p1s0f0: renamed from eth0
[   14.837885] scsi 0:0:0:0: Attached scsi generic sg0 type 0
[   14.838197] scsi 1:0:0:0: Attached scsi generic sg1 type 0
[   14.843619] sd 1:0:0:0: [sdb] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[   14.843709] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[   14.843714] sd 0:0:0:0: [sda] 4096-byte physical blocks
[   14.843806] sd 0:0:0:0: [sda] Write Protect is off
[   14.843856] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   14.844348] sd 1:0:0:0: [sdb] 4096-byte physical blocks
[   14.844498] sd 1:0:0:0: [sdb] Write Protect is off
[   14.844659] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA

(B)0[1;24r[m[?7h[?1h=[H[J[H[J Petitboot (v1.4.3-p84d353a)
[1B [0mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
[2B  [0mSystem information
[1B  System configuration
[1B  System status log
[1B  Language
[1B  Rescan devices
[1B  Retrieve config from URL
[1B *[0;7mExit to shell           [12B[25D[0mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
[1B [0mEnter=accept, e=edit, n=new, x=exit, l=language, g=log, h=help
[1B Welcome to Petitboot
[J[1;47H0000000000000000 0000000000000000
[23B [3;21r[3;1HMM[1;24r[3;3H[Disk: sda2 / 0566767e-5331-44ea-b067-c056fa071384][1B[49DUbuntu[12;27H[0;7m                           [12B[52D[m[4;21r[4;1HM[1;24r[4;5HUbuntu, with Linux 4.10.0-gd40f5c5
[20B [4;21r[4;1HM[1;24r[4;5HUbuntu, with Linux 4.10.0-gd40f5c5 (recovery mode)[14;54H[0;7m [10B[53D[m[enP5p1s0f0] Configuring with DHCP
 Booting in 9 sec: Ubuntu          [24;26H[24;13H8[24;26H[24;13H7[24;26H[24;13H6[24;26H[24;13H5[24;26H[24;13H4[24;26H[24;13H3[24;26H[24;13H2[24;26H[24;13H1[24;26H[24;10HUbuntu          [24;16H
 Loaded kernel image from file:///var/petitboot/mnt/dev/sda2/vmlinux-4.10.0-gd4[?7l0[?7h[24;9Hinitrd from file:///var/petitboot/mnt/dev/sda2/initrd.img-4.10.0-gd40f5[?7lc[?7h
 Running boot hooks[K
 Performing kexec loadreboot
The system is going down NOW!

Sent SIGTERM to all processes

Sent SIGKILL to all processes
[   29.704209] kexec_core: Starting new kernel
[   48.498569040,5] OPAL: Switch to big-endian OS
[   49.208626688,5] OPAL: Switch to little-endian OS
opal: OPAL detected !
Page sizes from device-tree:
Page size shift = 12 AP=0x0
Page size shift = 16 AP=0x5
Page size shift = 21 AP=0x1
Page size shift = 30 AP=0x2
Initializing Radix MMU
[... 37 lines cut ...]
numa:   NODE_DATA [mem 0x2007ff712900-0x2007ff71c5ff]
numa:     NODE_DATA(252) on node 8
numa: Initmem setup node 253
numa:   NODE_DATA [mem 0x2007ff708c00-0x2007ff7128ff]
numa:     NODE_DATA(253) on node 8
numa: Initmem setup node 254
numa:   NODE_DATA [mem 0x2007ff6fef00-0x2007ff708bff]
numa:     NODE_DATA(254) on node 8
numa: Initmem setup node 255
numa:   NODE_DATA [mem 0x2007ff6f5200-0x2007ff6feeff]
numa:     NODE_DATA(255) on node 8
Initializing IODA2 PHB (/pciex@600c3c0000000)
PCI host bridge /pciex@600c3c0000000 (primary) ranges:
 MEM 0x000600c000000000..0x000600c07ffeffff -> 0x0000000080000000 
 MEM 0x0006000000000000..0x0006003fffffffff -> 0x0006000000000000 (M64 #1..31)
 Using M64 #31 as default window
  256 (255) PE's M32: 0x80000000 [segment=0x800000]
                 M64: 0x4000000000 [segment=0x40000000]
  Allocated bitmap for 4088 MSIs (base IRQ 0xfe000)
Initializing IODA2 PHB (/pciex@600c3c0100000)
PCI host bridge /pciex@600c3c0100000  ranges:
 MEM 0x000600c080000000..0x000600c0fffeffff -> 0x0000000080000000 
 MEM 0x0006004000000000..0x0006007fffffffff -> 0x0006004000000000 (M64 #1..15)
 Using M64 #15 as default window
```

gkeishin, 2017-07-19:
Currently, we landed into Host SOL crashing out at petitbot with newer PNOR

```
IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.75
        op-build-v1.17-183-gd7635e7-dirty
        buildroot-2017.05-93-g9b91216
        skiboot-v5.7-rc2
        hostboot-9ecbdfd
        linux-4.12.1-openpower1-p3a89ea5
        petitboot-v1.5.0-pac16519
        machine-xml-5575e7e
        occ-2f3397d
        hostboot-binaries-c24fccd
        capp-ucode-9c73e9f
        sbe-02021c6-pfc4c21
```

if I can boot to OS, then I can try those OS shutdown test to recreate.

geissonator, 2017-07-19:
I know @jk-ozlabs has been digging into this one.

jk-ozlabs, 2017-07-20:
OK, I think I've managed to reproduce, by issuing the soft off just before petitboot userspace starts. The IPMI log I get from that:

```
method call time=1500439993.948617 sender=:1.72 -> destination=:1.10 serial=14043 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=setAttention
method call time=1500439994.041700 sender=:1.72 -> destination=:1.10 serial=14047 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 12
   byte 7
   byte 0
   byte 8
   byte 0
   array of bytes [
      3b 68 f0 08 eb 9a 55 9c b9 48 b3 8c 54 14 da d0
   ]
signal time=1500439994.054166 sender=:1.10 -> destination=(null destination) serial=9757 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 13
   byte 6
   byte 0
   byte 49
   array [
   ]
method call time=1500439994.062950 sender=:1.72 -> destination=:1.10 serial=14048 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 13
   byte 7
   byte 0
   byte 49
   byte 0
   array of bytes [
      02
   ]
signal time=1500439994.099882 sender=:1.10 -> destination=(null destination) serial=9759 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 14
   byte 6
   byte 0
   byte 53
   array [
   ]
method call time=1500439994.111609 sender=:1.72 -> destination=:1.10 serial=14050 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 14
   byte 7
   byte 0
   byte 53
   byte 0
   array of bytes [
      55 55 c0 41 a7 00 00 00 00 00 3a 04 00 ff ff ff
   ]
signal time=1500439994.422809 sender=:1.10 -> destination=(null destination) serial=9761 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 15
   byte 10
   byte 0
   byte 72
   array [
   ]
method call time=1500439994.488920 sender=:1.72 -> destination=:1.10 serial=14053 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 15
   byte 11
   byte 0
   byte 72
   byte 0
   array of bytes [
      ba e5 6e 59
   ]
signal time=1500439994.910915 sender=:1.10 -> destination=(null destination) serial=9763 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 16
   byte 10
   byte 0
   byte 72
   array [
   ]
method call time=1500439994.934928 sender=:1.72 -> destination=:1.10 serial=14056 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 16
[... 175 lines cut ...]
   byte 1
   byte 0
   array of bytes [
      00 80 01 99 02 8d 41 a7 00 4f 42 00 08 01 00
   ]
signal time=1500439995.758092 sender=:1.10 -> destination=(null destination) serial=9785 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 27
   byte 58
   byte 0
   byte 26
   array [
   ]
method call time=1500439995.784787 sender=:1.72 -> destination=:1.10 serial=14095 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 27
   byte 59
   byte 0
   byte 26
   byte 193
   array [
   ]
```

It looks like the kernel has received the soft off notification, but we don't see (over dbus) an IPMI request to issue the hard power off. I assume that this means that the btbridge daemon isn't seeing any traffic on the BT interface either, but we may need to confirm that too.

I'll do some kernel tracking to see if we're failing to invoke the graceful shutdown.

jk-ozlabs, 2017-10-10:
Just to keep all informed: @debmc has been doing quite a bit of analysis on this, and it looks like we have two potential races of the poweroff path during boot:

 1. OPAL message handlers in Linux: it's possible that the IPMI SEL message (to request the graceful poweroff) is received and processed by OPAL before the kernel registers the `OPAL_MSG_SHUTDOWN` handler. This means that the opal message handler for that message will end up calling an empty notifier chain, so no action is taken. We could potentially fix this by modifying the order of registration of these "core" message handlers (wrt the first message dequeues) in the opal platform code. Or possibly keep events pending until the notifier chain has at least one handler... @stewart: any other thoughts?

 1. busybox: after that point (where the kernel has a message handler registered for `OPAL_MSG_SHUTDOWN`), it's also possible that the orderly poweroff handler (`/sbin/poweroff` in our case) is called during very early stages of PID 1's init. Busybox's `poweroff` is implemented simply as sending a `SIGUSR2` to PID 1; it's possible that this is happening before PID 1 has installed a signal handler for `SIGUSR2`.

The former should be tracked here, but (2) is a busybox issue. I'll discuss this with the busybox folks, but keep updating here for completeness.

Massive thanks to @debmc on getting to the root causes of this one.

gkeishin, 2018-01-16:
```
[    4.192145] ata3: SATA link down (SStatus 0 SControl 300)
[    4.358003] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    4.358073] ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    4.359835] ata1.00: ATA-10: ST1000NX0313         00LY266 00LY265IBM, BE35, max UDMA/133
[    4.359901] ata1.00: 1953525168 sectors, multi 0: LBA48 NCQ (depth 31/32), AA
[    4.360276] ata2.00: ATA-10: ST1000NX0313         00LY266 00LY265IBM, BE35, max UDMA/133
[    4.360370] ata2.00: 1953525168 sectors, multi 0: LBA48 NCQ (depth 31/32), AA
[    4.361489] ata1.00: configured for UDMA/133
[    4.362655] ata2.00: configured for UDMA/133
[    4.363085] scsi 0:0:0:0: Direct-Access     ATA      ST1000NX0313     BE35 PQ: 0 ANSI: 5
[    4.480637] scsi 1:0:0:0: Direct-Access     ATA      ST1000NX0313     BE35 PQ: 0 ANSI: 5
[    4.610072] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    4.610131] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    4.610197] sd 0:0:0:0: [sda] Write Protect is off
[    4.610270] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.610597] sd 1:0:0:0: [sdb] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    4.610687] sd 1:0:0:0: [sdb] 4096-byte physical blocks
[    4.610769] sd 1:0:0:0: [sdb] Write Protect is off
[    4.610865] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.626809]  sda: sda1 sda2 sda3
[    4.628535] sd 0:0:0:0: [sda] Attached SCSI removable disk
[    4.656670]  sdb: sdb1 sdb2 sdb3
[    4.658873] sd 1:0:0:0: [sdb] Attached SCSI removable disk
[**    ] A start job is running for dev-mapp...oot.device (20min 6s / no limit )

```
Seeing shutdown taking time here . any fixes coming up for this one.

debmc, 2018-01-16:
Two patches have been submitted:

Busybox
http://lists.busybox.net/pipermail/busybox/2017-November/085980.html

linuxpcc-dev
https://lists.ozlabs.org/pipermail/linuxppc-dev/2017-November/166390.html

gkeishin, 2018-01-16:
Thanks @debmc  for the info.

oohal, 2019-07-15:
@debmc I guess this has been resolved?

debmc, 2019-07-15:
> has been resolved

The two patches submitted addressed boundary cases which had to be programmatically induced to recreate, so these two patches are still applicable, but since this particular symptom has not been seen since the other issues in this flow have been addressed, I think this issue can be closed.

hegdevasant, 2020-11-18:
We did address few other corner case issues in OPAL as well. Hence I guess its safe to close this issue now.

-Vasant

closed 2020-11-18 by hegdevasant