open-power/skiboot #82: /sys/firmware/devicetree/base/lid-type is showing phyp

PavamanSubramaniyam, 2017-05-29:
I have installed a P9 8375-42A Hardware with Ubuntu 16.04.3 OS.
Then I happened to cat the contents of the lid-type file and found as follows:

root@zz376p1:/sys/firmware/devicetree/base# cat lid-type
phyp
root@zz376p1:/sys/firmware/devicetree/base#

So, for all the P9 Hardware's we get the output of lid-type as phyp even though the machine is booted in OPAL mode.

The same output is even seen in Petitboot Shell prompt, when we cat for the output of lid-type.

pridhiviraj, 2017-05-29:
On P8 OPAL systems also showing same phyp as lid-type 
machine		: PowerNV 8284-22A
/sys/firmware/devicetree/base # cat lid-type 
phyp/sys/firmware/devicetree/base #

hegdevasant, 2017-05-29:
IIRC long back (when OPAL started) we were using PHYP LID ID to load OPAL LID and we are printing it as PHYP. 

Given that we have separate LID ID for OPAL, I think we can make it as OPAL now. Its minor issue. Will look into the code later this week.

-Vasant

pridhiviraj, 2017-08-08:
Seeing same issue in P9 boston system as well.
/ {
	compatible = "ibm,powernv", "supermicro,p9dsu", "supermicro,p9dsu";
	vendor = "IBM";
	nest-frequency = <0x0 0x5f5e1000>;
	model-name = "supermicro,p9dsu";
	lid-type = "phyp";

stewart-ibm, 2017-08-08:
Paidipeddi Pridhiviraj <notifications@github.com> writes:
> Seeing same issue in P9 boston system as well.
> / {
> 	compatible = "ibm,powernv", "supermicro,p9dsu", "supermicro,p9dsu";
> 	vendor = "IBM";
> 	nest-frequency = <0x0 0x5f5e1000>;
> 	model-name = "supermicro,p9dsu";
> 	lid-type = "phyp";

This is all from fairly ancient hackery and I'm thinking the correct
behaviour is to only add the property on systems prior to P9. Patch incoming.

-- 
Stewart Smith
OPAL Architect, IBM.

closed 2017-08-22 by stewart-ibm