open-power/skiboot #84: Clean up boot-time errors/complaints on Zaius P9 platform

rlippert, 2017-06-15:
Latest Skiboot on Zaius platform currently outputs all of these messages during the boot which sound like problems but are probably OK:

```
[  111.320573612,3] I2C: HDAT device array has no version! Assuming v1
[  111.320847779,3] I2C: HDAT device array has no version! Assuming v1
[  111.321167137,4] VPD: Model name ZAIUS_FX_08      not known
[  111.322621939,4] XIVE[ IC 00  ] IC BAR invalid, reconfiguring
[  111.325499306,4] XIVE[ IC 08  ] IC BAR invalid, reconfiguring
[  111.343384492,5] PSI[0x000]: ESB MMIO invalid, reconfiguring...
[  111.343547286,5] PSI[0x008]: ESB MMIO invalid, reconfiguring...
[    4.433959072,5] CAPI: WARNING: no compat thing found
[    4.441167872,3] PHB#0000[0:0]: Base location code not found !
[    4.442220800,3] PHB#0001[0:1]: Base location code not found !
[    4.443125888,3] PHB#0002[0:2]: Base location code not found !
[    4.444028384,3] PHB#0003[0:3]: Base location code not found !
[    4.445052928,3] PHB#0030[8:0]: Base location code not found !
[    4.446224960,3] PHB#0031[8:1]: Base location code not found !
[    4.447202832,3] PHB#0032[8:2]: Base location code not found !
[    4.448179216,3] PHB#0033[8:3]: Base location code not found !
[    4.449352528,3] PHB#0034[8:4]: Base location code not found !
[    4.450327568,3] PHB#0035[8:5]: Base location code not found !
[   16.385913728,3] FLASH: Failed to get ROOTFS partition info
```

We need to clean up the underlying causes of these issues or decrease the loglevel so they are not output by default.

ozbenh, 2017-06-15:
Good idea, I've been wanting to do some spring cleaning on our messages for a while too.

I'll see what I can do today.

stewart-ibm, 2017-06-15:
Yeah, I've been somewhat loosely tracking this with a test in op-test-framework: https://github.com/open-power/op-test-framework/blob/master/testcases/OpalMsglog.py#L36 (which is literally grepping for log messages with a higher priority than 5 after boot).

ozbenh, 2017-06-16:
So the I2C one: we should fix HostBoot. I'm taking care of the rest

closed 2017-06-16 by ozbenh