open-power/skiboot #86: System fails to boot when flashing upstream skiboot with upstream pflash.

pridhiviraj, 2017-06-22:


pridhiviraj, 2017-06-22:
Pflash Version:
root@ltc-test-hab02:~/skiboot/external/pflash# pflash  --version
Open-Power Flash tool pflash-5.6.0-123-g4df5312
root@ltc-test-hab02:~/skiboot/external/pflash# pflash -p skiboot.lid.xz -e -P PAYLOAD
WARNING: Erase region adjusted to 0x00ca9000..0x00cf5000
About to erase 0x00ca9000..0x00cf5000 !
WARNroot@ltc-test-hab02:~/skiboot/external/pflash# pflash  --version
Open-Power Flash tool pflash-5.6.0-123-g4df5312
root@ltc-test-hab02:~/skiboot/external/pflash# pflash -p skiboot.lid.xz -e -P PAYLOAD
WARNING: Erase region adjusted to 0x00ca9000..0x00cf5000
About to erase 0x00ca9000..0x00cf5000 !
WARNING ! This will modify your HOST flash chip content !
Enter "yes" to confirm:yes
Erasing...
About to program "skiboot.lid.xz" at 0x00ca9000..0x00cf4b80 !
Programming & Verifying...
[==================================================] 100%
Updating actual size in partition header...
ING ! This will modify your HOST flash chip content !
Enter "yes" to confirm:yes
Erasing...
About to program "skiboot.lid.xz" at 0x00ca9000..0x00cf4b80 !
Programming & Verifying...
[==================================================] 100%
Updating actual size in partition header...

pridhiviraj, 2017-06-22:
As system is down, i try to get the info of PNOR partition using some stable pflash on BMC busy box.
# ./pflash --info
Flash info:
-----------
Name          = Macronix MXxxL51235F
Total size    = 64MB 
Erase granule = 4KB 

Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000)
ID=01            HBEL 00008000..0002c000 (actual=00024000)
ID=02           GUARD 0002c000..00031000 (actual=00005000)
ID=03             HBD 00031000..00091000 (actual=00060000)
ID=04           DJVPD 00091000..000d9000 (actual=00048000)
ID=05            MVPD 000d9000..00169000 (actual=00090000)
ID=06            CVPD 00169000..001b1000 (actual=00048000)
ID=07             HBI 001b1000..00751000 (actual=005a0000)
ID=08            SBEC 00751000..007e1000 (actual=00090000)
ID=09             SBE 007e1000..00829000 (actual=00048000)
ID=10            WINK 00829000..00949000 (actual=00120000)
ID=11            HBRT 00949000..00ca9000 (actual=00360000)
FFS: Bad entry 12 in partition map
Error 102 scanning partitions
# 
# ./pflash -v    
Palmetto Flash tool 0.8.5
# 

By this tool in busybox, i am able to flash all the lids/ pnor images earlier without any issues. Looks like some thing got corrupted in PNOR. Which leads to fail the boot.

pridhiviraj, 2017-06-22:
I reverted pflash version to one commit lower, it flashes and boots fine.
Looks like this is the commit causing the issues https://github.com/open-power/skiboot/commit/73e1e8a727a9e7179719eb7844bd4248d9890114

pridhiviraj, 2017-06-22:
pflash  --version
Open-Power Flash tool pflash-5.6.0-119-g396b940

Nope, some thing went wrong with this level also. When i look at the booting side it boots from golden side instead of primary side.
# ipmitool -I lanplus -H <BMC-IP>  -U ADMIN -P admin sensor list | grep -i golden
BIOS Golden Side | 0x0        | discrete   | 0x0180| na        | na        | na        | na        | na        | na        
BMC Golden Side  | 0x0        | discrete   | 0x0080| na        | na        | na        | na        | na        | na        
#

cyrilbur-ibm, 2017-06-23:
Hi Pridiviraj,

Was this running pflash always on the BMC or were some commands run from the host?

pridhiviraj, 2017-06-23:
The failed case is where i tried flashing skiboot using upstream pflash in host.
To recover back used some stable pflash in BMC in the passed case.

cyrilbur-ibm, 2017-06-23:
So I have a fix. I don't understand how it fixes it. I'll do some more testing Monday.

pridhiviraj, 2017-06-23:
@cyrilbur-ibm Cool. Thanks.

cyrilbur-ibm, 2017-06-26:
The reason memset() the struct to zero works is that we memset the checksum word.
I've posted the fix to skiboot list: http://patchwork.ozlabs.org/patch/780576/

closed 2017-06-26 by cyrilbur-ibm