open-power/skiboot #87: pflash memory leaks

shenki, 2017-06-26:
I did an x86 build of pflash like this:
```
$ make CFLAGS=-g
```

And valgrind reports some (probably harmless) leaks:
```
$ valgrind --leak-check=full --show-leak-kinds=all ./pflash  -F /srv/tftp/firestone.pnor -P PAYLOAD -r foo 
==28213== Memcheck, a memory error detector
==28213== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==28213== Using Valgrind-3.12.0 and LibVEX; rerun with -h for copyright info
==28213== Command: ./pflash -F /srv/tftp/firestone.pnor -P PAYLOAD -r foo
==28213== 
Reading to "foo" from 0x00ca9000..0x00da9000 !
[==================================================] 100%
==28213== 
==28213== HEAP SUMMARY:
==28213==     in use at exit: 5,980 bytes in 32 blocks
==28213==   total heap usage: 38 allocs, 6 frees, 7,322 bytes allocated
==28213== 
==28213== 4 bytes in 1 blocks are definitely lost in loss record 1 of 5
==28213==    at 0x4C2DB2F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28213==    by 0x50CF799: strdup (strdup.c:42)
==28213==    by 0x109CF5: main (pflash.c:572)
==28213== 
==28213== 8 bytes in 1 blocks are definitely lost in loss record 2 of 5
==28213==    at 0x4C2DB2F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28213==    by 0x50CF799: strdup (strdup.c:42)
==28213==    by 0x109DE6: main (pflash.c:562)
==28213== 
==28213== 80 bytes in 1 blocks are still reachable in loss record 3 of 5
==28213==    at 0x4C2FB45: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28213==    by 0x10E827: ffs_init (libffs.c:252)
==28213==    by 0x10ADAA: open_partition (pflash.c:156)
==28213==    by 0x10997C: lookup_partition (pflash.c:207)
==28213==    by 0x10997C: main (pflash.c:788)
==28213== 
==28213== 1,792 bytes in 28 blocks are possibly lost in loss record 4 of 5
==28213==    at 0x4C2FB45: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28213==    by 0x10E9F4: ffs_init (libffs.c:303)
==28213==    by 0x10ADAA: open_partition (pflash.c:156)
==28213==    by 0x10997C: lookup_partition (pflash.c:207)
==28213==    by 0x10997C: main (pflash.c:788)
==28213== 
==28213== 4,096 bytes in 1 blocks are still reachable in loss record 5 of 5
==28213==    at 0x4C2DB2F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28213==    by 0x10E982: ffs_init (libffs.c:289)
==28213==    by 0x10ADAA: open_partition (pflash.c:156)
==28213==    by 0x10997C: lookup_partition (pflash.c:207)
==28213==    by 0x10997C: main (pflash.c:788)
==28213== 
==28213== LEAK SUMMARY:
==28213==    definitely lost: 12 bytes in 2 blocks
==28213==    indirectly lost: 0 bytes in 0 blocks
==28213==      possibly lost: 1,792 bytes in 28 blocks
==28213==    still reachable: 4,176 bytes in 2 blocks
==28213==         suppressed: 0 bytes in 0 blocks
==28213== 
==28213== For counts of detected and suppressed errors, rerun with: -v
==28213== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)
```

shenki, 2017-06-26:
My tree is at skiboot-5.6.0-123-g4df53124cb34 (4df53124cb34)

cyrilbur-ibm, 2017-07-11:
Patches are underway. there is some churn around pflash and this has prompted a bit of a rewrite.

I'm trying to include tests too.

For what its worth these are harmless and only exist for the userspace task. I haven't found any to be integral to libflash or libffs which skiboot uses, where leaks would be very bad.

cyrilbur-ibm, 2017-08-03:
Problems should be fixed by e1cf130df4da549e618b9657884d81b2dbd676da

closed 2017-08-03 by cyrilbur-ibm