open-power/skiboot #89: Easy to run out of async tokens with opal-sensor driver

rlippert, 2017-06-28:
Parallel sensor reads seem to make it easy to run out of async tokens because `opal_get_sensor_data()` in the kernel grabs a token for every sensor read before waiting behind a single mutex.

stewart-ibm, 2017-06-29:
Huh, yep - you're right there.

I'm trying to work out what that mutex is actually protecting.... and I think the answer is "nothing at all", and the correct patch will be in the kernel to just remove the mutex.

After reading through the kernel and skiboot code, I'm rather convinced that's the fix.

(kernel) patch incoming, as well as some documentation clarification for skiboot.

oohal, 2017-06-29:
@stewart-ibm you might want to check that this doesn't conflict with what @cyrilbur-ibm is working on

cyrilbur-ibm, 2017-06-29:
Yeah looks like it won't interfere - at the very least if you want to be super paranoid you could swap the mutex and the token acquisition?

stewart-ibm, 2017-06-29:
I have a patch, going to blend it with @cyrilbur-ibm kernel side async rework part of #90 as it turns out what the mutex in the sensors code was protecting from was subtly broken async token code.

stewart-ibm, 2017-07-07:
Quick update, @cyrilbur-ibm is looking at sending an updated series next week, either incorporating my patch to fix the sensors issue or NAKing it :)

stewart-ibm, 2017-08-15:
https://patchwork.ozlabs.org/patch/786988/ and associated series fixes this

rlippert, 2017-09-22:
Seems this is still pending review for the kernel?  Are you waiting for me to test it out or something like that? :)

stewart-ibm, 2017-10-02:
Might be?

@mpe anything to add? It looks like may be waiting another spin from @cyrilbur-ibm ?

cyrilbur-ibm, 2017-10-03:
Yes, we're waiting on me. My apologies. There is a spin in the works, unfortunately I'm trying to get this right and not just push a minimal style fix.

stewart-ibm, 2018-09-18:
Patches were merged here: https://patchwork.ozlabs.org/project/linuxppc-dev/list/?series=11638&state=* closing this issue.

closed 2018-09-18 by stewart-ibm

mpe, 2018-09-20:
Ending with this commit:

https://git.kernel.org/torvalds/c/6f469b67ff8a