open-power/skiboot #92: I2C: HDAT version 2 not supported!

rlippert, 2017-07-20:
Seems hostboot as of open-power/hostboot@9ecbdfd70311bd7074ce341201779791a14e381c has updated to a new version of I2C HDAT data but skiboot doesn't know how to parse it yet:

```
[   71.098741507,3] I2C: HDAT version 2 not supported! THIS IS A BUG
[   71.098917207,3] I2C: HDAT version 2 not supported! THIS IS A BUG
```

This breaks all I2C functionality from linux...

wak-google, 2017-07-20:
http://patchwork.ozlabs.org/patch/791337/

rlippert, 2017-07-20:
Yea already tried that patch, it only fixes the version number check to report it as version 2 instead of 32.  

Skiboot still needs to parse the new version :)

oohal, 2017-07-21:
@rlippert They're reporting v2 without actually populating any of the extra fields that were added in v2.  I asked the hostboot team put the version number back to v1 a few days ago, but I'm not sure when they'll have a fix out. 

For now you can work around it with this in skiboot:

```
From c1590ad05e6a9f4659d2784b3ec819cc272e8e2b Mon Sep 17 00:00:00 2001
From: Oliver O'Halloran <oohall@gmail.com>
Date: Fri, 21 Jul 2017 14:15:13 +1000
Subject: [PATCH] HACK: workaround broken HDAT version

Hostboot commit 3fbd50ab46e402867da8a5ae6b041aa9330fd6ea reworked the
Host I2C device array generation. As a side effect of this work around
it's now porting array version 2 without actually populating any of the
V2 data.

---
 hdata/i2c.c | 5 ++---
 1 file changed, 2 insertions(+), 3 deletions(-)

diff --git a/hdata/i2c.c b/hdata/i2c.c
index 5e9084e7e3e2..bf4d88ff7d2b 100644
--- a/hdata/i2c.c
+++ b/hdata/i2c.c
@@ -192,9 +192,8 @@ int parse_i2c_devs(const struct HDIF_common_hdr *hdr, int idata_index,
 	}
 
 	if (version != 1) {
-		prerror("I2C: HDAT version %d not supported! THIS IS A BUG\n",
-			version);
-		return -1;
+		prerror("I2C: Got HDAT version %d, forcing v1\n", version);
+		version = 1;
 	}
 
 	HDIF_iarray_for_each(&ahdr->hdr, i, dev) {
-- 
2.9.4
```

Or this in hostboot:

```
commit 6177a2cb37badce250c954b43eb4a3967baacd2d
Author: Oliver O'Halloran <oohall@gmail.com>
Date:   Fri Jul 21 14:20:02 2017 +1000

    fixup! Support new i2c device link ID format

diff --git a/src/usr/hdat/hdatmsarea.C b/src/usr/hdat/hdatmsarea.C
index ceb469828ecb..a9f3cc39d07b 100755
--- a/src/usr/hdat/hdatmsarea.C
+++ b/src/usr/hdat/hdatmsarea.C
@@ -390,7 +390,7 @@ void HdatMsArea::setMsaI2cInfo(
     iv_msaI2cHdr.hdatArrayCnt = i_I2cDevEntries.size();
     iv_msaI2cHdr.hdatAllocSize = sizeof(hdatI2cData_t);
     iv_msaI2cHdr.hdatActSize = sizeof(hdatI2cData_t);
-    iv_msaI2cHdr.hdatVersion = HOST_I2C_DEV_INFO_VERSION::V2;
+    iv_msaI2cHdr.hdatVersion = HOST_I2C_DEV_INFO_VERSION::V1;
     iv_msaHostI2cCnt = i_I2cDevEntries.size();
     iv_msaHostI2cSize = sizeof(iv_msaI2cHdr) +
         (sizeof(hdatI2cData_t) * iv_msaHostI2cCnt);
diff --git a/src/usr/hdat/hdatpcrd.C b/src/usr/hdat/hdatpcrd.C
index 1ecda256b426..06d99f3c8862 100644
--- a/src/usr/hdat/hdatpcrd.C
+++ b/src/usr/hdat/hdatpcrd.C
@@ -473,7 +473,7 @@ errlHndl_t HdatPcrd::hdatLoadPcrd(uint32_t &o_size, uint32_t &o_count)
                 l_hostI2cFullPcrdHdrPtr->hdatActSize =
                     sizeof(hdatI2cData_t);
                 l_hostI2cFullPcrdHdrPtr->hdatVersion =
-                    HOST_I2C_DEV_INFO_VERSION::V2;
+                    HOST_I2C_DEV_INFO_VERSION::V1;
 
                 hdatI2cData_t *l_hostI2cFullPcrdDataPtr =
                     reinterpret_cast<hdatI2cData_t *>
diff --git a/src/usr/hdat/hdatutil.H b/src/usr/hdat/hdatutil.H
index 1e730aebaaaf..289d4560f239 100755
--- a/src/usr/hdat/hdatutil.H
+++ b/src/usr/hdat/hdatutil.H
@@ -103,6 +103,7 @@ extern trace_desc_t* g_trac_hdat;
 
 enum HOST_I2C_DEV_INFO_VERSION : uint32_t
 {
+    V1 = 0x00000001,
     V2 = 0x00000002,
 };
 
```

stewart-ibm, 2017-08-15:
```
commit 0818df625e3539a6aaad025ee6f1d1a441dee4a5
Author: Oliver O'Halloran <oohall@gmail.com>
Date:   Wed Aug 2 22:58:46 2017 +1000

    hdata/i2c: Work around broken I2C array version
    
    Work around a bug in the I2C devices array that shows the
    array version as being v2 when only the v1 data is populated.
    
    Signed-off-by: Oliver O'Halloran <oohall@gmail.com>
    [stewart@linux.vnet.ibm.com: only parse V2 as V1, still pass on >v2]
    Signed-off-by: Stewart Smith <stewart@linux.vnet.ibm.com>
```

With 0818df625e3539a6aaad025ee6f1d1a441dee4a5 upstream, we have the workaround for this in skiboot.

Hilarity may ensue when Hostboot starts populating V2 information though.

closed 2017-08-15 by stewart-ibm