open-power/skiboot #136: fast-reset: NMI watchdog: BUG: soft lockup - CPU#113 stuck for 22s! [kopald:854]

stewart-ibm, 2017-10-12:
(bringing over from internal bugzilla)

 Pridhiviraj Paidipeddi 2016-12-24 00:15:00 AEDT:

Install a 8284-22A PowerNV system with FW860.20 latest firmware build, and install a Ubuntu 16.10.
Then below test steps caused the NMI Watchdog: Bug soft lockup issue in petitboot kernel.

Test-steps:
1. Boot the system to runtime(i,e into petitboot)

2. Enable the fast-reboot by using below command
/ # nvram -p ibm,skiboot --update-config experimental-fast-reset=feeling-lucky
/ # nvram --print-config=experimental-fast-reset -p ibm,skiboot
feeling-lucky

3. Then do a FSP ResetReload test
smgr resetReload

4. Once FSP comes UP, do reboot in petitboot.

5. After system reaching to peitiboot, check dmesg for softlockup's

```
[   32.107804] NMI watchdog: BUG: soft lockup - CPU#113 stuck for 22s! [kopald:854]
[   32.107894] Modules linked in: sr_mod sd_mod cdrom lpfc ipr crc_t10dif crct10dif_generic crct10dif_common
[   32.107903] CPU: 113 PID: 854 Comm: kopald Not tainted 4.4.27-openpower1 #1
[   32.107905] task: c000002fc61a5ef0 ti: c000002fc621c000 task.ti: c000002fc621c000
[   32.107907] NIP: c00000000000d8dc LR: c00000000000d8dc CTR: c000000000015d00
[   32.107908] REGS: c000002fc621f630 TRAP: 0901   Not tainted  (4.4.27-openpower1)
[   32.107909] MSR: 9000000000009033 <SF,HV,EE,ME,IR,DR,RI,LE>  CR: 42000024  XER: 00000000
[   32.107915] CFAR: c0000000000a5448 SOFTE: 1 
[   32.107915] GPR00: c000000000059534 c000002fc621f8b0 c00000000089a500 0000000000000900 
[   32.107915] GPR04: 0000000000000000 0000000022b2980d 0000000004898d98 0000000000000001 
[   32.107915] GPR08: 0000000000000000 0000000000000002 000000e371050eb6 0000000000000003 
[   32.107915] GPR12: c000000000038d10 c00000000fe95300 
[   32.107925] NIP [c00000000000d8dc] arch_local_irq_restore+0x5c/0x80
[   32.107927] LR [c00000000000d8dc] arch_local_irq_restore+0x5c/0x80
[   32.107928] Call Trace:
[   32.107929] [c000002fc621f8b0] [c00000000007e7e0] vtime_common_account_irq_enter+0x48/0x5c (unreliable)
[   32.107932] [c000002fc621f8d0] [c000000000059534] __do_softirq+0xe0/0x2b4
[   32.107934] [c000002fc621f9c0] [c0000000000599a0] irq_exit+0x74/0xcc
[   32.107935] [c000002fc621f9e0] [c00000000001646c] timer_interrupt+0xb0/0xc4
[   32.107937] [c000002fc621fa10] [c00000000000995c] restore_check_irq_replay+0x54/0x70
[   32.107941] --- interrupt: 901 at opal_handle_events+0xb0/0xb8
[   32.107941]     LR = opal_handle_events+0x5c/0xb8
[   32.107943] [c000002fc621fd50] [c00000000003b6ac] kopald+0x2c/0x64
[   32.107945] [c000002fc621fd80] [c0000000000717a0] kthread+0xfc/0x108
[   32.107947] [c000002fc621fe30] [c0000000000094b4] ret_from_kernel_thread+0x5c/0xa8
[   32.107948] Instruction dump:
[   32.107949] f821ffe1 419e000c e92d0020 7d210164 39200000 992d0252 4bffff11 39400001 
[   32.107952] 994d0252 2fa30000 419e0010 4bff4591 <60000000> 48000010 e92d0020 61298000 
[   34.138212] device-mapper: ioctl: 4.34.0-ioctl (2015-10-28) initialised: dm-devel@redhat.com
[   34.187971] tg3 0001:03:00.3 enP1p3s0f3: renamed from eth3
[   34.220791] EXT4-fs (dm-2): mounted filesystem with ordered data mode. Opts: (null)
[   34.260346] EXT4-fs (dm-5): INFO: recovery required on readonly filesystem
[   34.260351] EXT4-fs (dm-5): write access will be enabled during recovery
[   34.272708] EXT4-fs (dm-5): recovery complete
[   34.272743] EXT4-fs (dm-5): mounted filesystem with ordered data mode. Opts: (null)
[   36.479095]  rport-3:0-4: blocked FC remote port time out: removing rport
[   55.007967] tg3 0003:05:00.1 enP3p5s0f1: renamed from eth5
[   78.017790] INFO: rcu_sched self-detected stall on CPU
[   78.017795] 	100-...: (1 GPs behind) idle=167/140000000000002/0 softirq=176/176 fqs=2082 
[   78.017796] 	 (t=2100 jiffies g=-220 c=-221 q=97)
[   78.017817] Task dump for CPU 100:
[   78.017818] kopald          R  running task        0   854      2 0x00000804
[   78.017821] Call Trace:
[   78.017826] [c000002fc621f1e0] [c00000000007c6f0] sched_show_task+0x10c/0x118 (unreliable)
[   78.017830] [c000002fc621f250] [c00000000009eda8] rcu_dump_cpu_stacks+0xa0/0xd0
[   78.017832] [c000002fc621f2a0] [c0000000000a2428] rcu_check_callbacks+0x294/0x734
[   78.017834] [c000002fc621f3c0] [c0000000000a44e0] update_process_times+0x3c/0x74
[... 141 lines cut ...]
[  277.167842]     LR = arch_local_irq_restore+0x5c/0x80
[  277.167846] [c000002fc621f8b0] [c00000000007e7e0] vtime_common_account_irq_enter+0x48/0x5c (unreliable)
[  277.167849] [c000002fc621f8d0] [c000000000059534] __do_softirq+0xe0/0x2b4
[  277.167852] [c000002fc621f9c0] [c0000000000599a0] irq_exit+0x74/0xcc
[  277.167854] [c000002fc621f9e0] [c00000000001646c] timer_interrupt+0xb0/0xc4
[  277.167857] [c000002fc621fa10] [c00000000000995c] restore_check_irq_replay+0x54/0x70
[  277.167860] --- interrupt: 901 at irq_work_queue+0xa0/0xa4
[  277.167860]     LR = irq_work_queue+0x78/0xa4
[  277.167864] [c000002fc621fd00] [c00000000003f528] opal_handle_events+0x5c/0xb8 (unreliable)
[  277.167867] [c000002fc621fd50] [c00000000003b6ac] kopald+0x2c/0x64
[  277.167870] [c000002fc621fd80] [c0000000000717a0] kthread+0xfc/0x108
[  277.167873] [c000002fc621fe30] [c0000000000094b4] ret_from_kernel_thread+0x5c/0xa8

/ # 
/ # update_flash -d
Current firwmare version :
  P side    : FW860.12 (SV860_066)
  T side    : FW860.20 (SV860_067)
  Boot side : FW860.20 (SV860_067)
/ # 

/ # cat /sys/firmware/opal/msglog | grep -i skiboot
[   81.163066898,5] SkiBoot f38ed26 starting...
/ #81 
```

hegdevasant, 2017-10-12:
Interesting. Something similar to other bug we saw internally. kopald makes opal call to get outstanding events and then calls our handler in kernel. 

-Vasant

hegdevasant, 2017-10-26:
Can you please retry with upstream kernel? 

-Vasant

npiggin, 2017-10-26:
I proposed a patch to process opal events directly by kopald, and to enable and disable interrupts and also cond_resched between processing each event. It might improve this situation.

https://marc.info/?l=linuxppc-embedded&m=150902327030882&w=2

pridhiviraj, 2017-10-27:
@hegdevasant Reproduced with both official build and upstream kernel.
```
<4>[    0.000000] Could not find start_pfn for node 17
<4>[    1.918993] This architecture does not have kernel memory protection.
<4>[    1.997483] Emulex LightPulse Fibre Channel SCSI driver 11.4.0.1
<4>[    1.997487] Copyright (C) 2017 Broadcom. All Rights Reserved. The term "Broadcom" refers to Broadcom Limited and/or its subsidiaries.
<3>[    2.001923] EEH: Frozen PE#fd on PHB#1 detected
<3>[    2.001929] EEH: PE location: U78CB.001.WZS0030-P1-C14, PHB location: N/A
<4>[    2.002752] EEH: This PCI device has failed 1 times in the last hour
<4>[    2.002793] EEH: of node=0001:04:00.0
<4>[    2.002796] EEH: PCI device/vendor: 034a1014
<4>[    2.002799] EEH: PCI cmd/status register: 00100142
<4>[    2.002800] EEH: PCI-E capabilities and status follow:
<4>[    2.002853] EEH: PCI-E 00: 0002a410 10008fc2 0000283e 0003f503 
<4>[    2.002862] EEH: PCI-E 10: 10830000 00000000 00000000 00000000 
<4>[    2.002863] EEH: PCI-E 20: 00000000 
<4>[    2.002864] EEH: PCI-E AER capability register set follows:
<4>[    2.002875] EEH: PCI-E AER 00: 14820001 00000000 00400000 00462030 
<4>[    2.002885] EEH: PCI-E AER 10: 00000000 0000e000 000001e0 00000000 
<4>[    2.002894] EEH: PCI-E AER 20: 00000000 00000000 00000000 00000000 
<4>[    2.002897] EEH: PCI-E AER 30: 00000000 00000000 
<3>[    2.211328] ELOG: OPAL log info read failed
<3>[    2.348038] lpfc 0004:01:00.0: 0:6101 Disabling NVME support: Not supported by firmware: 0 0
<3>[    2.348043] lpfc 0004:01:00.0: 0:2574 IO channels: irqs 4 fcp 4 nvme 0 MRQ: 0
<3>[    4.578036] lpfc 0004:01:00.1: 1:6101 Disabling NVME support: Not supported by firmware: 0 0
<3>[    4.578041] lpfc 0004:01:00.1: 1:2574 IO channels: irqs 4 fcp 4 nvme 0 MRQ: 0
<3>[   10.254238] lpfc 0005:0b:00.0: 2:1303 Link Up Event x1 received Data: x1 x1 x10 x0 x0 x0 0
<3>[   13.159540] lpfc 0005:0b:00.1: 3:1303 Link Up Event x1 received Data: x1 x1 x10 x0 x0 x0 0
<3>[   26.918024] INFO: rcu_sched self-detected stall on CPU
<3>[   26.918033] 	35-...: (2 GPs behind) idle=1a2/140000000000002/0 softirq=56/58 fqs=900 
<3>[   26.918033] 	 (t=2100 jiffies g=-265 c=-266 q=730)
<1>[   26.918048] Call Trace:
<1>[   26.918053] [c000000ff4a0b1c0] [c0000000000926cc] sched_show_task+0xe8/0xf4 (unreliable)
<1>[   26.918057] [c000000ff4a0b230] [c0000000000c0500] rcu_dump_cpu_stacks+0xd0/0xe0
<1>[   26.918060] [c000000ff4a0b280] [c0000000000bfa80] rcu_check_callbacks+0x304/0x788
<1>[   26.918062] [c000000ff4a0b3b0] [c0000000000c3054] update_process_times+0x3c/0x74
<1>[   26.918065] [c000000ff4a0b3e0] [c0000000000d2330] tick_sched_handle.isra.3+0x50/0x64
<1>[   26.918068] [c000000ff4a0b400] [c0000000000d2390] tick_sched_timer+0x4c/0x8c
<1>[   26.918069] [c000000ff4a0b440] [c0000000000c3b1c] __hrtimer_run_queues+0xe0/0x16c
<1>[   26.918071] [c000000ff4a0b4c0] [c0000000000c3dbc] hrtimer_interrupt+0xac/0x1f0
<1>[   26.918074] [c000000ff4a0b580] [c000000000019228] __timer_interrupt+0x7c/0xdc
<1>[   26.918076] [c000000ff4a0b5c0] [c000000000019330] timer_interrupt+0xa8/0xc0
<1>[   26.918079] [c000000ff4a0b5f0] [c00000000000b864] fast_exception_return+0x128/0x150
<1>[   26.918082] --- interrupt: 901 at .L144+0x0/0x4
<1>[   26.918082]     LR = arch_local_irq_restore+0x5c/0x80
<1>[   26.918084] [c000000ff4a0b8e0] [c000000000093588] vtime_account_irq_enter+0x54/0x5c (unreliable)
<1>[   26.918086] [c000000ff4a0b900] [c00000000006eb80] __do_softirq+0xd8/0x290
<1>[   26.918089] [c000000ff4a0b9f0] [c00000000006efe0] irq_exit+0x88/0xe0
<1>[   26.918091] [c000000ff4a0ba10] [c000000000019334] timer_interrupt+0xac/0xc0
<1>[   26.918093] [c000000ff4a0ba40] [c00000000000b864] fast_exception_return+0x128/0x150
<1>[   26.918097] --- interrupt: 901 at _restgpr0_31+0x0/0xba38
<1>[   26.918097]     LR = irq_work_queue+0x74/0xa4
<1>[   26.918099] [c000000ff4a0bd30] [c00000000004b204] opal_handle_events+0x54/0xa8 (unreliable)
<1>[   26.918101] [c000000ff4a0bd80] [c00000000004755c] kopald+0x44/0x70
<1>[   26.918103] [c000000ff4a0bdc0] [c0000000000880b4] kthread+0x150/0x158
<1>[   26.918106] [c000000ff4a0be30] [c00000000000b464] ret_from_kernel_thread+0x5c/0x78
<1>[   90.728029] .
<1>[   90.728073] ready
<1>[   90.808024] .
<1>[   90.808025] .
<1>[   90.808027] .
<1>[   90.808029] .
<1>[   90.808105] ready
<1>[   90.808110] ready
<1>[   90.808117] ready
<1>[   90.808128] ready
<1>[   90.888025] .
<1>[   90.888025] .
<1>[   90.888026] .
<1>[   90.888034] .
[... 237 lines cut ...]
<1>[  335.728050] [c000000ff4a0b230] [c0000000000c0500] rcu_dump_cpu_stacks+0xd0/0xe0
<1>[  335.728053] [c000000ff4a0b280] [c0000000000bfa80] rcu_check_callbacks+0x304/0x788
<1>[  335.728055] [c000000ff4a0b3b0] [c0000000000c3054] update_process_times+0x3c/0x74
<1>[  335.728057] [c000000ff4a0b3e0] [c0000000000d2330] tick_sched_handle.isra.3+0x50/0x64
<1>[  335.728059] [c000000ff4a0b400] [c0000000000d2390] tick_sched_timer+0x4c/0x8c
<1>[  335.728061] [c000000ff4a0b440] [c0000000000c3b1c] __hrtimer_run_queues+0xe0/0x16c
<1>[  335.728063] [c000000ff4a0b4c0] [c0000000000c3dbc] hrtimer_interrupt+0xac/0x1f0
<1>[  335.728065] [c000000ff4a0b580] [c000000000019228] __timer_interrupt+0x7c/0xdc
<1>[  335.728067] [c000000ff4a0b5c0] [c000000000019330] timer_interrupt+0xa8/0xc0
<1>[  335.728069] [c000000ff4a0b5f0] [c00000000000b864] fast_exception_return+0x128/0x150
<1>[  335.728073] --- interrupt: 901 at .L144+0x0/0x4
<1>[  335.728073]     LR = arch_local_irq_restore+0x5c/0x80
[... cut ...]
<1>[  335.728083] [c000000ff4a0ba40] [c00000000000b864] fast_exception_return+0x128/0x150
<1>[  335.728087] --- interrupt: 901 at _restgpr0_31+0x0/0xba38
<1>[  335.728087]     LR = irq_work_queue+0x74/0xa4
<1>[  335.728089] [c000000ff4a0bd30] [c00000000004b204] opal_handle_events+0x54/0xa8 (unreliable)
<1>[  335.728091] [c000000ff4a0bd80] [c00000000004755c] kopald+0x44/0x70
<1>[  335.728093] [c000000ff4a0bdc0] [c0000000000880b4] kthread+0x150/0x158
<1>[  335.728096] [c000000ff4a0be30] [c00000000000b464] ret_from_kernel_thread+0x5c/0x78
/ # 
```

pridhiviraj, 2017-10-27:
@npiggin Thanks, Have you done the tests mention in this issue against your patches?

npiggin, 2017-10-27:
No I haven't had a chance to yet, I've been caught up doing P9 work at the moment.

npiggin, 2018-04-05:
I brought the patch up to date and resubmitted it, but... I don't think it solves the root cause of the problem, there must be a long running OPAL event that triggers it AFAIKS. Maybe EEH?