open-power/skiboot #138: [Witherspoon] Unable to boot till the login prompt using RHEL

swe12345, 2017-10-12:
Unable to boot to the login prompt using RHEL on the witherspoon system.
```
[  786.463168] blocking rcu_node structures: l=1:54-107:0x800000/.[  786.463263] Task dump for CPU 77:[  786.463323] swapper/77      R  running task        0     0      1 0x00000800[  786.463423] Call Trace:[  786.463464] [c000003fd4f63b20] [c000003fd4f63e20] 0xc000003fd4f63e20 (unreliable)[  786.463554] [c000003fd4f63cf0] [0000000000000000]           (null) |  
-- | --
```
RHEL Version:
```
Red Hat Enterprise Linux Server (4.11.0-10.el7a.ppc64le) 7.4 (Pegas)
```
System Configuration:
BMC Details:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-139"
VERSION_ID="v1.99.10-139-g3202561"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-139"
BUILD_ID="v1.99.10"
```
PNOR Version:
```
root@witherspoon:~# cat /media/pnor-ro-46e65782/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.60
        op-build-v1.19-147-gc4823a9
        buildroot-2017.08-8-g5e23247
        skiboot-v5.8-150-gabb1d4e81f33
        hostboot-b01e6bb
        linux-4.13.5-openpower1-p402ce7c
        petitboot-v1.6.0-pf254bee
        machine-xml-389c56f
        occ-ba4e81e
        hostboot-binaries-d9d2910
        capp-ucode-p9-dd2-v2
        sbe-374add4
```

stewart-ibm, 2017-10-12:
Loks like something hanging in the kernel. You'll need the full kernel log to be able to work out what, as that looks to be a very mangled one. Can you get into xmon and dump out the log?

swe12345, 2017-10-25:
Call Traces after booting using RHEL :
```
[ 1000.536710] Call Trace:
[ 1000.536732] [c000003fce317390] [c000000000b5f370] sched_show_task.part.61+0x78/0x90 (unreliable)
[ 1000.536800] [c000003fce317400] [c000000000b5f9d0] rcu_dump_cpu_stacks+0xd0/0x134
[ 1000.536858] [c000003fce317450] [c0000000001b5f14] rcu_check_callbacks+0xbf4/0xcd0
[ 1000.536915] [c000003fce317570] [c0000000001be80c] update_process_times+0x7c/0xf0
[ 1000.536972] [c000003fce3175b0] [c0000000001da364] tick_sched_timer+0x84/0x190
[ 1000.537071] [c000003fce3175f0] [c0000000001bf588] __hrtimer_run_queues+0x118/0x3e0
[ 1000.537159] [c000003fce317690] [c0000000001c07c4] hrtimer_interrupt+0xf4/0x300
[ 1000.537217] [c000003fce317760] [c000000000023f70] __timer_interrupt+0x90/0x270
[ 1000.537274] [c000003fce3177b0] [c000000000024380] timer_interrupt+0xa0/0xe0
[ 1000.537324] [c000003fce3177e0] [c000000000008f84] decrementer_common+0x114/0x120
[ 1000.537383] --- interrupt: 901 at smp_call_function_single+0x180/0x2a0
[ 1000.537383]     LR = smp_call_function_single+0x208/0x2a0
[ 1000.537467] [c000003fce317b30] [c000000000296920] perf_reboot+0x100/0x1a0
[ 1000.537515] [c000003fce317bc0] [c00000000013e508] blocking_notifier_call_chain+0x98/0x100
[ 1000.537573] [c000003fce317c10] [c000000000140f6c] kernel_power_off+0x2c/0x150
[ 1000.537630] [c000003fce317c80] [c00000000012e410] process_one_work+0x1c0/0x660
[ 1000.537687] [c000003fce317d20] [c0000000001314dc] worker_thread+0xac/0x6d0
[ 1000.537736] [c000003fce317dc0] [c00000000013aa90] kthread+0x160/0x1a0
[ 1000.537786] [c000003fce317e30] [c00000000000b368] ret_from_kernel_thread+0x5c/0x74
[ 1001.506438] INFO: task swapper/0:1 blocked for more than 120 seconds.
[ 1001.506498]       Tainted: G             L ------------   4.11.0-10.el7a.ppc64le #1
[ 1001.506581] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1001.506693] swapper/0       D    0     1      0 0x00000800
[ 1001.506761] Call Trace:
[ 1001.506804] [c000003fd4c83940] [c00000000001b910] __switch_to+0x300/0x470
[ 1001.506900] [c000003fd4c839a0] [c000000000b4b688] __schedule+0x3a8/0xb70
[ 1001.506997] [c000003fd4c83a70] [c000000000b4be90] schedule+0x40/0xb0
[ 1001.507086] [c000003fd4c83aa0] [c0000000001afab4] _synchronize_rcu_expedited.constprop.58+0x444/0x4b0
[ 1001.507230] [c000003fd4c83b60] [c0000000001afec4] synchronize_sched+0xb4/0xd0
[ 1001.507354] [c000003fd4c83bd0] [c0000000006f41e0] register_sysrq_key+0xa0/0x120
[ 1001.507478] [c000003fd4c83c10] [c000000000e5d594] setup_xmon_sysrq+0x24/0x3c
[ 1001.507601] [c000003fd4c83c30] [c00000000000cca0] do_one_initcall+0x60/0x1c0
[ 1001.507705] [c000003fd4c83cf0] [c000000000e44434] kernel_init_freeable+0x28c/0x36c
[ 1001.507807] [c000003fd4c83db0] [c00000000000d4fc] kernel_init+0x2c/0x2b0
[ 1001.507911] [c000003fd4c83e30] [c00000000000b368] ret_from_kernel_thread+0x5c/0x74
[ 1001.508059] INFO: task kworker/1:1:975 blocked for more than 120 seconds.
[ 1001.508142]       Tainted: G             L ------------   4.11.0-10.el7a.ppc64le #1
[ 1001.508259] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1001.508365] kworker/1:1     D    0   975      2 0x00000800
[ 1001.508443] Workqueue: events wait_rcu_exp_gp
[ 1001.508511] Call Trace:
[ 1001.508545] [c000003fcfe87750] [c000003fcf684280] 0xc000003fcf684280 (unreliable)
[ 1001.508652] [c000003fcfe87920] [c00000000001b910] __switch_to+0x300/0x470
[ 1001.508749] [c000003fcfe87980] [c000000000b4b688] __schedule+0x3a8/0xb70
[ 1001.508846] [c000003fcfe87a50] [c000000000b4be90] schedule+0x40/0xb0
[ 1001.508936] [c000003fcfe87a80] [c000000000b52074] schedule_timeout+0x374/0x550
[ 1001.509038] [c000003fcfe87b60] [c0000000001af020] rcu_exp_wait_wake+0x120/0x720
[ 1001.509147] [c000003fcfe87c80] [c00000000012e410] process_one_work+0x1c0/0x660
[ 1001.509254] [c000003fcfe87d20] [c0000000001314dc] worker_thread+0xac/0x6d0
[ 1001.509350] [c000003fcfe87dc0] [c00000000013aa90] kthread+0x160/0x1a0
[ 1001.509444] [c000003fcfe87e30] [c00000000000b368] ret_from_kernel_thread+0x5c/0x74
[ 1016.876437] INFO: rcu_sched detected expedited stalls on CPUs/tasks: { 8-... } 61643 jiffies s: 5 root: 0x1/.
[ 1016.876544] blocking rcu_node structures: l=1:0-53:0x100/.
[ 1016.876603] Task dump for CPU 8:
[ 1016.876632] kworker/8:1     R  running task        0   985      2 0x00000804
[ 1016.876763] Workqueue: events poweroff_work_func
```

stewart-ibm, 2018-08-21:
Is this still being seen with current released firmware? IF so, let's reopen.

closed 2018-08-21 by stewart-ibm