open-power/skiboot #139: Fast-reboot: Boot crashes after a PNOR reset/reprovision.

pridhiviraj, 2017-10-17:
```
root@ltc-test-hab02:~# ipmitool  raw 0x3a 0x1d
 03
root@ltc-test-hab02:~# [147390.509426555,5] IPMI: PNOR access released

root@ltc-test-hab02:~# 
root@ltc-test-hab02:~# ipmitool  raw 0x3a 0x1d
 00
root@ltc-test-hab02:~# reboot

[-1;-1f[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[  159.046257] reboot: Restarting system
[147484.023085907,5] OPAL: Reboot request...
[147484.023135349,5] RESET: Initiating fast reboot 3...
[147484.230693375,5] PSI: Hot reset!
[147484.230693235,7] blocklevel_read: 0x0	0x31c43b40	0x30
[147484.230695125,7] blocklevel_raw_read: 0x0	0x31c43b40	0x30
[147484.230743624,6] FFS: Flash header not found. Code: 102
[147484.230744651,3] FLASH: Can't open ffs handle
[147484.230745519,7] blocklevel_read: 0x0	0x31c43b40	0x30
[147484.230746633,7] blocklevel_raw_read: 0x0	0x31c43b40	0x30
[147484.237729309,6] FFS: Flash header not found. Code: 102
[147484.237730213,3] FLASH: Can't open ffs handle
[147484.290422793,5]  Need EOI !
[147484.300706398,5] PCI: Clearing all devices...
[147489.506274365,5] PCI: Resetting PHBs and training links...
[147492.096770676,5] PCI: Probing slots...
[147492.153143007,5] PHB#0000:00:00.0 [ROOT] 1014 03dc R:00 C:060400 B:01..01 SLOT=Slot3 
[147492.153252137,5] PHB#0000:01:00.0 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[147492.155902718,5] PHB#0000:01:00.1 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[147492.170900112,5] PHB#0000:01:00.2 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[147492.174835599,5] PHB#0000:01:00.3 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[147492.190338596,5] PHB#0001:00:00.0 [ROOT] 1014 03dc R:00 C:060400 B:01..15 SLOT=Backplane PLX 
[147492.194252073,5] PHB#0001:01:00.0 [SWUP] 10b5 8748 R:ca C:060400 B:02..15 LOC_CODE=Backplane PLX
[147492.209814358,5] PHB#0001:02:01.0 [SWDN] 10b5 8748 R:ca C:060400 B:03..07 SLOT=S000103 
[147492.212619776,5] PHB#0001:02:08.0 [SWDN] 10b5 8748 R:ca C:060400 B:08..08 
[147492.221428308,5] PHB#0001:08:00.0 [LGCY] 1b4b 9235 R:11 C:010601 (          sata) LOC_CODE=Backplane PLX
[147492.232561183,5] PHB#0001:02:09.0 [SWDN] 10b5 8748 R:ca C:060400 B:09..09 SLOT=S000109 
[147492.252045702,5] PHB#0001:09:00.0 [EP  ] 104c 8241 R:02 C:0c0330 (      usb-xhci) LOC_CODE=S000109
[147492.255897717,5] PHB#0001:02:0a.0 [SWDN] 10b5 8748 R:ca C:060400 B:0a..0b 
[147492.265874120,5] PHB#0001:0a:00.0 [ETOX] 1a03 1150 R:03 C:060400 B:0b..0b LOC_CODE=Backplane PLX
[147492.286855068,5] PHB#0001:0b:00.0 [PCID] 1a03 2000 R:30 C:030000 (           vga) LOC_CODE=Backplane PLX
[147492.289646472,5] PHB#0001:02:10.0 [SWDN] 10b5 8748 R:ca C:060400 B:0c..10 
[147492.303161240,5] PHB#0001:02:11.0 [SWDN] 10b5 8748 R:ca C:060400 B:11..15 
[147492.326703249,5] PHB#0002:00:00.0 [ROOT] 1014 03dc R:00 C:060400 B:01..ff SLOT=Slot4 
[147492.340660199,5] IPMI: Resetting boot count on successful boot
[147492.423407161,5] INIT: Waiting for kernel...
[147492.445863026,5] INIT: platform wait for kernel load failed
[147492.470039219,5] INIT: Assuming kernel at 0x20000000
[147493.031249533,5] INIT: 64-bit LE kernel discovered
[147493.064987973,5] INIT: Starting kernel at 0x20002140, fdt at 0x30678fe0 87512 bytes)
[147493.121828739,3] ***********************************************
[147493.142838051,3] Unexpected exception e40 !
[147493.157587423,3] SRR0 : 0000000030002804 SRR1 : 90000000001a1000
[147493.177541240,3] HSRR0: 0000000020002140 HSRR1: 9000000000001000
[147493.214533902,3] LR   : 0000000030014598 CTR  : 0000000020002140
[147493.231729091,3] CFAR : 000000003000519c
[147493.259999339,3] CR   : 48084482  XER: 20000000
[147493.280108535,3] GPR00: 000000003001454c GPR16: 0000000000000001
[147493.299456571,3] GPR01: 0000000031c83bb0 GPR17: c000001ffb4e8080
[147493.309631897,3] GPR02: 00000000300f4d00 GPR18: c000000000fc9c28
[147493.336846278,3] GPR03: 0000000030678fe0 GPR19: 0000000000000008
[147493.357828025,3] GPR04: 0000000000000000 GPR20: 0000000000000006
[147493.388434306,3] GPR05: 0000000000000000 GPR21: 0000000000002140
[147493.397519484,3] GPR06: 0000000065504150 GPR22: 0000000000002140
[147493.408788655,3] GPR07: 0000000040000001 GPR23: 0000000000000000
[147493.419629737,3] GPR08: 0000000030000000 GPR24: 000000003031cb20
[147493.426265654,3] GPR09: 00000000300050b0 GPR25: 0000000000000001
[147493.465681919,3] GPR10: 00000000000050b0 GPR26: 0000000040000000
[147493.465841935,3] GPR11: 0000000000000000 GPR27: 0000000000000001
[147494.030663610,3] GPR12: 0000000048084484 GPR28: 0000000000000000
[147494.050312415,3] GPR13: 0000000031c80000 GPR29: 0000000000000000
[147494.080704353,3] GPR14: c00000000163ef18 GPR30: 000000003031caf0
[147494.106680686,3] GPR15: c000000000fdd698 GPR31: 000000003031cae0
[147494.116705330,0] Aborting!
CPU 0020 Backtrace:
 S: 0000000031c83a10 R: 000000003001374c   .backtrace+0x40
 S: 0000000031c83aa0 R: 0000000030019c28   ._abort+0x4c
 S: 0000000031c83b20 R: 0000000030027888   .exception_entry+0x114
 S: 0000000031c83bb0 R: 0000000000001f00 * +0x1f00
 S: 0000000031c83d80 R: 000000003001454c   .load_and_boot_kernel+0xa10
 S: 0000000031c83e60 R: 0000000030025474   .fast_reboot_entry+0x4b8
 S: 0000000031c83f00 R: 0000000030002970   fast_reset_entry+0x2c

```

Steps to reproduce:
```
1. Boot the system to runtime.
2. Enable fast-reboot.
3. Do a PNOR reset/reprovision.
4. Reboot the system.
Boot will crash at this stage.
```

stewart-ibm, 2017-10-17:
Paidipeddi Pridhiviraj <notifications@github.com> writes:
> ```
> 1. Boot the system to runtime.
> 2. Enable fast-reboot.
> 3. Do a PNOR reset/reprovision.
> 4. Reboot the system.
> Boot will crash at this stage.
> ```

Is reprovision meant to be supported while the machine is running?

I expect we're not waiting until we have control of the flash?

-- 
Stewart Smith
OPAL Architect, IBM.

pridhiviraj, 2017-10-17:
@stewart-ibm Yes, it is expected to work in system runtime as well, 
right, looks like this messages may be useful.
```
[147484.230744651,3] FLASH: Can't open ffs handle
[147484.230743624,6] FFS: Flash header not found. Code: 102
```