open-power/skiboot #143: external: Intermittant parallel make build failures

bradbishop, 2017-12-15:
Seeing intermittant parallel make build failures on pflash/external 5.9.4 after moving from 5.9:

```02:40:49 | NOTE: make -j16 CROSS_COMPILE=arm-openbmc-linux-gnueabi- SKIBOOT_VERSION=v5.9.4 V=1 PFLASH_VERSION=v5.9.4 GARD_VERSION=v5.9.4 -C external/gard all
02:40:49 | make: Entering directory '/tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/gard/v5.9.4-r0/git/external/gard'
02:40:49 | ln -sf ../../libflash ./libflash
02:40:49 | ln -sf ../../ccan ./ccan
02:40:49 | ln -sf ../common ./common
02:40:49 | arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/tmp/openbmc/sysroots/firestone  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/gard/v5.9.4-r0=/usr/src/debug/gard/v5.9.4-r0 -fdebug-prefix-map=/tmp/openbmc/sysroots/x86_64-linux= -fdebug-prefix-map=/tmp/openbmc/sysroots/firestone=  -O2 -Wall -Werror -I. -c gard.c -o gard.o
02:40:49 | gard.c:29:40: fatal error: ccan/array_size/array_size.h: No such file or directory
02:40:49 |  #include <ccan/array_size/array_size.h>
02:40:49 |                                         ^
02:40:49 | compilation terminated.
02:40:49 | rules.mk:32: recipe for target 'gard.o' failed
02:40:49 | make: *** [gard.o] Error 1
02:40:49 | make: *** Waiting for unfinished jobs....
02:40:49 | make: Leaving directory '/tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/gard/v5.9.4-r0/git/external/gard'```

stewart-ibm, 2018-02-12:
I think https://patchwork.ozlabs.org/patch/871852/ will solve it. Testing is *very* much appreciated.

stewart-ibm, 2018-02-23:
We merged the patch, marking as fixed in 5.10.

closed 2018-02-23 by stewart-ibm